fgcz/rawrr #11: functional test on output of multiple devices

cpanse, 2020-10-28:
# determine input 

last two files of each available instrument
```{bash}
PFILE=/srv/www/htdocs/Data2San/sync_LOGS/pfiles.txt ; 
cat ${PFILE} \
  | cut -f4 -d";" \
  | cut -d"/" -f3 \
  | sort \
  | uniq \
  |while read i ; 
  do 
        grep -E "/${i}/.*autoQC01.*raw$" ${PFILE} \
          | tail -n 2; 
  done \
  | cut -d";" -f4 \
  | while read raw ; 
    do 
        [[ -f /srv/www/htdocs/${raw} ]] && echo ${raw} ; 
  done
```

tobiasko, 2020-10-28:
Sorry, had to change that issue title. ;-)

tobiasko, 2020-10-28:
Could you build a webpage for the output? That would be really cool. Our own little test environment!

tobiasko, 2020-10-28:
That would also be a cool extension for our case II, far too big for the manuscript, but we could put the link in the text!

cpanse, 2020-10-30:
https://fgcz-ms.uzh.ch/~cpanse/rawR/test/functional_test.html

cpanse, 2020-10-31:
mission completed

closed 2020-10-31 by cpanse