fgcz/rawrr #14: Installation fails

Adafede, 2020-11-12:
Hi,

Very nice tool!

At the moment, sadly:

> install.packages('http://fgcz-ms.uzh.ch/~cpanse/rawR_0.1.0.tar.gz')
Warning in install.packages :
  package ‘http://fgcz-ms.uzh.ch/~cpanse/rawR_0.1.0.tar.gz’ is not available for this version of R

A version of this package for your version of R might be available elsewhere,
see the ideas at
https://cran.r-project.org/doc/manuals/r-patched/R-admin.html#Installing-packages
> 
> R.version
               _                           
platform       x86_64-apple-darwin17.0     
arch           x86_64                      
os             darwin17.0                  
system         x86_64, darwin17.0          
status                                     
major          4                           
minor          0.3                         
year           2020                        
month          10                          
day            10                          
svn rev        79318                       
language       R                           
version.string R version 4.0.3 (2020-10-10)
nickname       Bunny-Wunnies Freak Out

cpanse, 2020-11-12:
Dear @Adafede; Thank you for reporting. I have the same OS and R version running. I can only reproduce that error while not having mono available on the system. can you please run  `mono -V` in the terminal and provide us the output.

Adafede, 2020-11-12:
Dear @cpanse  thak you for your very prompt answer! 

Here it is:

```
mono -V
Mono JIT compiler version 6.4.0.198 (2019-06/fe64a4765e6 Wed Sep 18 14:14:51 EDT 2019)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           
	SIGSEGV:       altstack
	Notification:  kqueue
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug 
	Interpreter:   yes
	LLVM:          yes(610)
	Suspend:       hybrid
	GC:            sgen (concurrent by default)
```

cpanse, 2020-11-12:
@Adafede to be sure, can you please add the following arguments `type='source'` and  `repo=NULL`:

```{r}
install.packages('http://fgcz-ms.uzh.ch/~cpanse/rawR_0.1.0.tar.gz',  repo=NULL)
```

if it does **not** work:

```{r}
install.packages('http://fgcz-ms.uzh.ch/~cpanse/rawR_0.1.0.tar.gz', type='source', repo=NULL)
```
thanks
Christian

Adafede, 2020-11-12:
Yep,

> install.packages('http://fgcz-ms.uzh.ch/~cpanse/rawR_0.1.0.tar.gz',  repo=NULL)
essai de l'URL 'http://fgcz-ms.uzh.ch/~cpanse/rawR_0.1.0.tar.gz'
Content type 'application/x-gzip' length 2533329 bytes (2.4 MB)
```
==================================================
downloaded 2.4 MB

* installing *source* package ‘rawR’ ...
** using staged installation
** R
** exec
** inst
** byte-compile and prepare package for lazy loading
** help
*** installing help indices
** building package indices
** testing if installed package can be loaded from temporary location
** testing if installed package can be loaded from final location
** testing if installed package keeps a record of temporary installation path
* DONE (rawR)
```

Wonderful, thank you!

cpanse, 2020-11-12:
Thank you! We will adapt the release/install information. C

closed 2020-11-12 by cpanse