fgcz/rawrr #15: package naming

cpanse, 2020-11-13:
solve the naming conflict with 

 https://CRAN.R-project.org/package=rawr 

```
* package encoding: UTF-8
* checking CRAN incoming feasibility ... ERROR
Maintainer: 'Christian Panse <cp@fgcz.ethz.ch>'
New submission
Conflicting package names (submitted: rawR, existing: rawr [https://CRAN.R-project.org])
Conflicting package names (submitted: rawR, existing: rawr [CRAN archive])
The Title field should be in title case. Current version is:
'Access to Thermo Fisher Scientific raw files from R'
In title case that is:
```

```
R> BiocCheck("rawR_0.1.1.tar.gz")
This is BiocCheck version 1.26.0. BiocCheck is a work in
progress. Output and severity of issues may change. Installing
package...
* Checking Package Dependencies...
* Checking if other packages can import this one...
* Checking to see if we understand object initialization...
* Checking for deprecated package usage...
* Checking for remote package usage...
* Checking version number...
* Checking for version number mismatch...
* Checking version number validity...
    Package version 0.1.1; pre-release
* Checking R Version dependency...
* Checking package size...
* Checking individual file sizes...
    * WARNING: The following files are over 5MB in size:
      'rawRcolor.tif'
* Checking biocViews...
* Checking that biocViews are present...
    * ERROR: No biocViews terms found.
See http://bioconductor.org/developers/how-to/biocViews/
* Checking build system compatibility...
* Checking for blank lines in DESCRIPTION...
* Checking if DESCRIPTION is well formatted...
* Checking for proper Description: field...
* Checking for whitespace in DESCRIPTION field names...
* Checking that Package field matches directory/tarball
  name...
* Checking for Version field...
* Checking for valid maintainer...
* Checking DESCRIPTION/NAMESPACE consistency...
    * WARNING: Import grDevices, graphics, utils in
      DESCRIPTION as well as NAMESPACE.
* Checking vignette directory...
    This is an unknown type of package
    * ERROR: No 'vignettes' directory.
* Checking library calls...
* Checking for library/require of rawR...
* Checking coding practice...
    * NOTE: Avoid sapply(); use vapply()
      Found in files:
        rawR.R (line 1011, column 29)
    * NOTE: Avoid 1:...; use seq_len() or seq_along()
      Found in files:
        rawR.R (line 600, column 36)
        rawR.R (line 745, column 70)
Warning in readLines(infile) :
  incomplete final line found on '/tmp/RtmpLJg2l6/filedebd713f5408/rawR/tests/testthat/test-header.R'
    * WARNING: Avoid class() == or class() != ; use is() or
      !is()
      Found in files:
        R/rawR.R (line 68)
[... 86 lines cut ...]
vignette, available at
https://bioconductor.org/packages/3.12/bioc/vignettes/BiocCheck/inst/doc/BiocCheck.html#interpreting-bioccheck-output
BiocCheck FAILED.
$error
[1] "No biocViews terms found."                                                                                      
[2] "No 'vignettes' directory."                                                                                      
[3] "At least 80% of man pages documenting exported objects must have runnable examples. The following pages do not:"
[4] "Package must be removed from CRAN."                                                                             
$warning
[1] "The following files are over 5MB in size: 'rawRcolor.tif'"                                                                                                                                                 
[2] "Import grDevices, graphics, utils in DESCRIPTION as well as NAMESPACE."                                                                                                                                    
[3] " Avoid class() == or class() != ; use is() or !is()"                                                                                                                                                       
[4] "Add non-empty \\value sections to the following man pages: man/plot.rawRchromatogram.Rd, man/plot.rawRchromatogramSet.Rd, man/plot.rawRspectrum.Rd, man/print.rawRspectrum.Rd, man/summary.rawRspectrum.Rd"
$note
 [1] " Avoid sapply(); use vapply()"                                                                                                                                                     
 [2] " Avoid 1:...; use seq_len() or seq_along()"                                                                                                                                        
 [3] "Recommended function length <= 50 lines."                                                                                                                                          
 [4] "Usage of dontrun{} / donttest{} found in man page examples."                                                                                                                       
 [5] "Use donttest{} instead of dontrun{}."                                                                                                                                              
 [6] "Consider adding a NEWS file, so your package news will be included in Bioconductor release announcements."                                                                         
 [7] "Consider shorter lines; 32 lines (2%) are > 80 characters long."                                                                                                                   
 [8] "Consider 4 spaces instead of tabs; 5 lines (0%) contain tabs."                                                                                                                     
 [9] "Consider multiples of 4 spaces for line indents, 233 lines(14%) are not."                                                                                                          
R> 
```

tobiasko, 2020-11-13:
@cpanse Could you add the complete first command?

cpanse, 2020-11-13:
@tobiasko `R CMD check rawR_0.1.1.tar.gz`

tobiasko, 2020-11-17:
The easiest way out of this dilemma is to submit `rawR` to Bioc using a slightly "extended" package name that preserves the onomatopoetic idea. For instance `rawrrR`. But reading this:

“Package:” field: This is the name of the package. This should match the github repository name and is **case sensitive**. A package name should be descriptive and not already exist as a current package (**case-insensitive**) in Bioconductor or CRAN. Avoid names that are easily confused with existing package names, or that imply a temporal (e.g., ExistingPackage2) or qualitative (e.g., ExistingPackagePlus) relationship."

Still makes me think that the check should NOT fail.

tobiasko, 2020-11-17:
Interesting!

```
> if (!requireNamespace("BiocManager"))
+     install.packages("BiocManager")
Loading required namespace: BiocManager
> BiocManager::install("rawR")
Bioconductor version 3.10 (BiocManager 1.30.10), R 3.6.3 (2020-02-29)
Installing package(s) 'rawR'
Old packages: 'ape', 'backports', 'bit', 'bit64', 'bookdown', 'boot', 'broom', 'callr', 'car', 'carData', 'class', 'cli', 'clipr',
  'codetools', 'colorspace', 'covr', 'cowplot', 'data.table', 'dbplyr', 'DelayedArray', 'devtools', 'digest', 'doParallel', 'doSNOW',
  'dplyr', 'DT', 'e1071', 'ellipsis', 'foreach', 'Formula', 'fs', 'generics', 'ggforce', 'ggplot2', 'ggpubr', 'git2r', 'glue', 'gplots',
  'gtools', 'haven', 'HDF5Array', 'Hmisc', 'htmlTable', 'htmltools', 'htmlwidgets', 'httpuv', 'httr', 'isoband', 'iterators', 'jsonlite',
  'kableExtra', 'KernSmooth', 'knitr', 'labeling', 'later', 'lattice', 'lme4', 'lmtest', 'lubridate', 'maptools', 'MASS', 'matrixStats',
  'mgcv', 'modelr', 'MSstatsTMT', 'mvtnorm', 'nlme', 'nloptr', 'nnet', 'openssl', 'openxlsx', 'pillar', 'pkgbuild', 'pkgload', 'processx',
  'promises', 'protViz', 'ps', 'purrr', 'quantreg', 'R6', 'raster', 'RcmdrMisc', 'Rcpp', 'RcppArmadillo', 'RCurl', 'readr', 'remotes',
  'reshape2', 'rex', 'rlang', 'rmarkdown', 'roxygen2', 'rprojroot', 'RSQLite', 'rstudioapi', 'rversions', 'rvest', 'S4Vectors', 'sandwich',
  'scales', 'shiny', 'shinyjs', 'sp', 'spatial', 'statmod', 'stringi', 'survival', 'survminer', 'sys', 'testthat', 'tibble', 'tidyr',
  'tinytex', 'usethis', 'vcd', 'vctrs', 'withr', 'xfun', 'xml2', 'zip', 'zoo'
Update all/some/none? [a/s/n]: 
n
Warning messages:
1: package ‘rawR’ is not available (for R version 3.6.3) 
2: Perhaps you meant ‘rawr’ ? 
```

cpanse, 2020-11-17:
> Interesting!
>
> ```
> +     install.packages("BiocManager")
> Loading required namespace: BiocManager
> Bioconductor version 3.10 (BiocManager 1.30.10), R 3.6.3 (2020-02-29)
> Installing package(s) 'rawR'
> Old [...]

yes this is the CRAN https://CRAN.R-project.org/package=rawr package

tobiasko, 2020-12-02:
I just realized that the cross check with CRAN is done case-INsensitive! "Wer lesen kann ist klar im Vorteil"! ;-)

tobiasko, 2020-12-02:
https://support.bioconductor.org/p/p133305/#p133310

tobiasko, 2020-12-04:
updates from Martin Morgan:

https://support.bioconductor.org/p/p133305/#p133310

rRenaming the package will be necessary.

cpanse, 2020-12-04:
most intuitive would be then `rawFileReader`.

cpanse, 2020-12-04:
ah it is too boring maybe `orbitrap` reader

tobiasko, 2020-12-04:
Why not

`rawrr`

because

https://www.urbandictionary.com/define.php?term=rawrr

[image]

It is as close to `rawR` as possible and solves the naming conflict witn CRAN.

tobiasko, 2020-12-09:
Done!

closed 2020-12-09 by tobiasko