fgcz/rawrr #16: readChromatogram does not return 0 values

TaraBartolec, 2020-11-17:
Hi,

Thanks for making a great tool! I have found it quite useful so far 👍 
I have an issue for XIC values when I wish to plot a certain peptides.

Firstly, I can successfully extract and plot the XICs using your inbuilt functions, but cannot figure out how to constrain the retention times plotted/extracted. 

I did manage to access the S3 elements in the chromatogram object and plot them myself in ggplot, but then had an issue where rawR does not report the 0 values for M/Zs at certain times. This is useful to see the shape of the eluting peptide, though I acknowledge it will likely increase the object size...

Is it possible to clarify (1) how to constrain the XIC for a certain retention time range, and (2) how to access (or at least impute from RT of MS1 scans) the 0 values of XICs.

Thanks again for making this tool,
Tara

tobiasko, 2020-11-17:
Hi Tara,

thanks for your positive feedback and for asking these questions! 😀

Regarding the restriction of extraction: I think the API, or more precisely the `GetChromatogramData` method, [see](https://github.com/fgcz/rawR/blob/44d3cc08b15ea66241647d3c1df47a3649cab682/src/rawR.cs#L752) allows you to define a `startScan` and `endScan` argument to restrict the import. But currently, we call the method using `-1` and `-1` [see](https://github.com/fgcz/rawR/blob/44d3cc08b15ea66241647d3c1df47a3649cab682/src/rawR.cs#L752)
which most likely means without restriction. I think this could be changed, but would require some code changes on the C# level and we would need to think about how to pass these arguments from the R layer to C# and incorporate the arguments as attributes to the chromatogram object.

But in essence, it would be a very useful extension. Not sure why we didn't do it right away. 😜

In terms of functionality, what would be your favourite option? Would you like to restrict by scan index, or by RT?

I would need to confirm all of the above with @cpanse first and have a look at the API docu.

Best,
Tobi

cpanse, 2020-11-17:
@tobiasko https://github.com/fgcz/rawR/blob/44d3cc08b15ea66241647d3c1df47a3649cab682/src/rawR.cs#L769

tobiasko, 2020-11-17:
Aha. @cpanse,  so does this mean the API explicitly returns zeros, but our code "discards" them? They are not written to the tmp file?

cpanse, 2020-11-17:
> Aha. @cpanse, so does this mean the API explicitly returns zeros, but our code "discards" them? They are not written to the temp file?

yes; for exactly that reason @TaraBartolec mentioned ``it will likely increase the object size.'' For the moment we should deliver the libraries' output and we have to think about a `slim` or `trim` argument for removing the (0, 0) tuples.!?

tobiasko, 2020-11-17:
makes sense. or we think about a smarter way of storing the data inside the S3 object. If these vectors tend to be very sparse...well, use a sparse vector instead of a dense one. This will only store none zero values and `plot()` will behave as expected.

cpanse, 2020-11-17:
Dear @TaraBartolec @tobiasko 

I uncommented line 769. can you have a look?

```
install.packages('http://fgcz-ms.uzh.ch/~cpanse/rawR_0.1.2.tar.gz', repo=NULL)
```

C

TaraBartolec, 2020-11-17:
Perfect, it works! 
also @tobiasko I think personally it would be useful to restrict an XIC by retention time, but it would be useful to have the option for both if not too difficult.
Thanks again,
Tara

tobiasko, 2020-11-19:
Hi Tara,

one last question since you tried to plot using the gglot library: How important is this to you? Our approach was to use base R  for plotting to keep it simple from a developers perspective (no dependency, always available). But maybe this is not the most important aspect...

Greetings,
Tobi

TaraBartolec, 2020-11-19:
Hi Tobi,

It is not that important to use ggplot, I just used it as I wanted to restrain the retention times plotted and had trouble doing that with the plot() function for some reason. Being able to easily filter RT is the most important thing for the plotting for us.

Best,
Tara

tobiasko, 2020-11-20:
Ok, great! THX again for your feedback.

closed 2020-11-20 by tobiasko