fgcz/rawrr #17: CI / GitHub actions

tobiasko, 2020-11-20:
Should we start using CI services like GitHub actions?

https://ropensci.org/technotes/2020/11/19/moving-away-travis/

I think it is a nice way of making sure `rawR` works on different OS platforms and different R versions.

tobiasko, 2020-11-25:
First try done!

https://github.com/fgcz/rawR/actions/runs/383025130

with limited success 😂! Looks like the package was installed from source (without the dlls and rawR.exe and having mono installed). Maybe we change the script so that the latest release is downloaded and installed, but then only the releases are checked. Still nice to have, since we could test release candidates before putting them into the wild.

Or we integrate scripts that download the missing components separately from our ftp server, create the executable (rawR.exe) and place them were they need to be.

Is it possible to install mono on the test system somehow?

tobiasko, 2020-11-25:
Or we finally add the RawFileReader dlls and the `rawR.exe` to the repository and have these problems eliminated once and for all.

tobiasko, 2020-11-25:
The compomics [ThermoRawFileParser](https://github.com/compomics/ThermoRawFileParser) repo also contains RawFileReader libraries in the repo.

cpanse, 2021-03-10:
[image]

closed 2021-03-10 by cpanse