fgcz/rawrr #18: readChromatogram

KarambolageNed, 2020-11-23:
Hi, 

I really like the package. Thank you for that! I was just going through the vignette and when I do readChromatogramm it gives me the following error. I guess I will not be the only one with that. How do you solve that?

plot(rawR::readChromatogram(rawfile = rawfile, type = "tic"))
Error in read.table(file = file, header = header, sep = sep, quote = quote,  : 
  duplicate 'row.names' are not allowed

Cheers!

tobiasko, 2020-11-24:
Hi @KarambolageNed,

thanks for you interest in our package. The problem with your issue is that I have no clue what `rawfile` contained at runtime. Could you please execute the following code bricks:

```
sessionInfo()
rawR:::.monoInfo()
```

and

```
library(rawR)
## Example 1: not meaning full but proof-of-concept
rawfile <- file.path(path.package(package = 'rawR'), 'extdata', 'sample.raw')
TIC <- readChromatogram(rawfile, type='tic')
plot(TIC)
```

You should see this:

[image]

KarambolageNed, 2020-11-24:
Hi tobiasko,

I am a beginner with R, but I guess there is lots of people in proteomics without no clue in R. Thanks for your answer!

Here:

> sessionInfo()
R version 4.0.3 (2020-10-10)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 18363)

Matrix products: default

locale:
[1] LC_COLLATE=German_Germany.1252  LC_CTYPE=German_Germany.1252    LC_MONETARY=German_Germany.1252 LC_NUMERIC=C                   
[5] LC_TIME=German_Germany.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] rawR_0.1.0

loaded via a namespace (and not attached):
 [1] Biobase_2.50.0        vsn_3.58.0            pkgload_1.1.0         foreach_1.5.1         assertthat_0.2.1     
 [6] BiocManager_1.30.10   affy_1.68.0           stats4_4.0.3          remotes_2.2.0         impute_1.64.0        
[11] sessioninfo_1.1.1     pillar_1.4.7          lattice_0.20-41       glue_1.4.2            limma_3.46.0         
[16] digest_0.6.27         colorspace_2.0-0      preprocessCore_1.52.0 plyr_1.8.6            MALDIquant_1.19.3    
[21] XML_3.99-0.5          pkgconfig_2.0.3       devtools_2.3.2        zlibbioc_1.36.0       purrr_0.3.4          
[26] scales_1.1.1          processx_3.4.4        affyio_1.60.0         BiocParallel_1.24.1   tibble_3.0.4         
[31] generics_0.1.0        IRanges_2.24.0        ggplot2_3.3.2         usethis_1.6.3         ellipsis_0.3.1       
[36] withr_2.3.0           BiocGenerics_0.36.0   cli_2.2.0             magrittr_2.0.1        crayon_1.3.4         
[41] memoise_1.1.0         ps_1.4.0              fs_1.5.0              ncdf4_1.17            fansi_0.4.1          
[46] doParallel_1.0.16     MASS_7.3-53           pkgbuild_1.1.0        mzR_2.24.1            tools_4.0.3          
[51] prettyunits_1.1.1     lifecycle_0.2.0       MSnbase_2.15.7        S4Vectors_0.28.0      munsell_0.5.0        
[56] callr_3.5.1           pcaMethods_1.82.0     compiler_4.0.3        mzID_1.28.0           tinytex_0.27         
[61] rlang_0.4.8           grid_4.0.3            iterators_1.0.13      rstudioapi_0.13       testthat_3.0.0       
[66] gtable_0.3.0          codetools_0.2-18      R6_2.5.0              dplyr_1.0.2           rprojroot_2.0.2      
[71] ProtGenerics_1.22.0   desc_1.2.0            parallel_4.0.3        Rcpp_1.0.5            vctrs_0.3.5          
[76] tidyselect_1.1.0      xfun_0.19            
> rawR:::.monoInfo()
Error in system2("mono", "-V", stdout = TRUE) : '"mono"' not found

And for the proof of concept:

library(rawR)
> ## Example 1: not meaning full but proof-of-concept
> rawfile <- file.path(path.package(package = 'rawR'), 'extdata', 'sample.raw')
> TIC <- readChromatogram(rawfile, type='tic')
Error in read.table(file = file, header = header, sep = sep, quote = quote,  : 
  duplicate 'row.names' are not allowed
> plot(TIC)
Error in plot(TIC) : object 'TIC' not found

I do not know what "mono" is.

Thanks!

tobiasko, 2020-11-24:
Since your OS is MS Windows you don't need to have `mono` installed, but you need the .NET framework. Could you please check if you have .Net Framework 4.5.1 and greater installed?

https://docs.microsoft.com/en-us/dotnet/framework/migration-guide/how-to-determine-which-versions-are-installed

tobiasko, 2020-11-24:
Or you download and (re)install:

https://dotnet.microsoft.com/download/dotnet-framework

The .NET framework is needed for the execution of our `C#` code that calls methods of the [RawFileReader.](https://planetorbitrap.com/rawfilereader)

KarambolageNed, 2020-11-24:
Hi again,

I just tried to fix that problem, but still does not work. 

My .NET framework is 4.8. I checked it in the REGISTRY and it has the number 528040. 

The command rawR::readSpectrum works fine. With rawR::readIndex(rawfile = rawfile), I get the same issue: 

> Idx <- rawR::readIndex(rawfile = rawfile)
Error in read.table(file = file, header = header, sep = sep, quote = quote,  : 
  more columns than column names

tobiasko, 2020-11-24:
Ok...it could be that the problem is much simpler. Your `locale` indicates that your are using German language settings:

```
locale:
[1] LC_COLLATE=German_Germany.1252 LC_CTYPE=German_Germany.1252 LC_MONETARY=German_Germany.1252 LC_NUMERIC=C
[5] LC_TIME=German_Germany.1252
```

Could you please check your corresponding Windows settings, especially with respect to the formatting of numbers. What do you see? Does changing to the format English (US) solve the problem?

cpanse, 2020-11-24:
@KarambolageNed  as an alternative, you can also try to install the latest release:
```
install.packages('http://fgcz-ms.uzh.ch/~cpanse/rawR_0.1.2.tar.gz', repo=NULL)
```
this will make the `readIndex` function work for your language setting. Can you confirm?

KarambolageNed, 2020-12-03:
Hi, 

sry for the late answer. I changed the language setting, but it did not help. I solved my question differently. Somehow there is a problem with my MONO JIT compiler, but I can not solve it. Installing the 0.1.2 version did not make it work.

closed 2021-01-20 by tobiasko