fgcz/rawrr #26: display sublicense at startup of interactive R session

tobiasko, 2020-12-10:
The following should be displayed after loading the package:

```
RawFileReader reading tool. Copyright © 2016 by Thermo Fisher Scientific, Inc. All rights reserved
```

cpanse, 2020-12-10:
@tobiasko 
[image]

cpanse, 2020-12-10:
[RawFileReaderLicense.pdf](https://github.com/fgcz/rawrr/files/5675089/RawFileReaderLicense.pdf)

tobiasko, 2020-12-10:
What just came to my mind. Are their different versions of the RawFileReader? Shouldn't that be displayed after `using`? Like

```
...using
RawFileReader 1.0.0
RawFileReader reading tool. Copyright © 2016 by Thermo Fisher Scientific, Inc. All rights reserved
```

cpanse, 2020-12-10:
@

> What just came to my mind. Are their different versions of the RawFileReader? Shouldn't that be displayed after `using`? Like
> 
> ```
> ...using
> RawFileReader 1.0.0
> RawFileReader reading tool. Copyright © 2016 by Thermo Fisher Scientific, Inc. All rights reserved
> ```

we add the version number in the package information (NEWS file)

closed 2020-12-23 by cpanse