fgcz/rawrr #27: citation()

tobiasko, 2020-12-10:
We should update the package in a way that `citation("rawrr") returns the desired information. The current state is:

```
> citation(package = "rawrr")

To cite package ‘rawrr’ in publications use:

  Christian Panse and Tobias Kockmann (NA). rawrr: Access to Thermo Fisher Scientific raw
  files from R. R package version 0.1.7. https://github.com/fgcz/rawR/

A BibTeX entry for LaTeX users is

  @Manual{,
    title = {rawrr: Access to Thermo Fisher Scientific raw files from R},
    author = {Christian Panse and Tobias Kockmann},
    note = {R package version 0.1.7},
    url = {https://github.com/fgcz/rawR/},
  }

Warning messages:
1: In citation(package = "rawrr") :
  no date field in DESCRIPTION file of package ‘rawrr’
2: In citation(package = "rawrr") :
  could not determine year for ‘rawrr’ from package DESCRIPTION file
```
I would suggest to reference our bioRxiv manuscript for now.

tobiasko, 2020-12-10:
For comparison:

```
> citation("MSstats")

  M. Choi, C. Chang, T. Clough, D. Broudy, T. Killeen, B. MacLean, O. Vitek, MSstats: an R
  package for statistical analysis of quantitative mass spectrometry-based proteomic
  experiments, Bioinformatics. 30:2524-2536, 2014.

A BibTeX entry for LaTeX users is

  @Article{,
    title = {MSstats: an R package for statistical analysis of quantitative mass spectrometry-based proteomic experiments},
    author = {Meena Choi},
    journal = {Bioinformatics},
    volume = {30},
    page = {2524-2536},
    year = {2014},
  }
```

cpanse, 2020-12-10:
```
R> citation('rawrr')

To cite the package rawrr in publications use:

Kockmann T, Panse C (2020). "The rawrr R package - Direct
Access to Orbitrap Data and Beyond." _Cold Spring Harbor
Laboratory_. doi: 10.1101/2020.10.30.362533 (URL:
https://doi.org/10.1101/2020.10.30.362533).

A BibTeX entry for LaTeX users is

  @Article{,
    title = {The rawrr R package - Direct Access to Orbitrap Data and Beyond},
    author = {Tobias Kockmann and Christian Panse},
    journal = {Cold Spring Harbor Laboratory},
    year = {2020},
    doi = {10.1101/2020.10.30.362533},
  }

R> 
```

closed 2020-12-23 by cpanse