fgcz/rawrr #28: replace purrr::map_df  by base R function

cpanse, 2020-12-28:
https://github.com/fgcz/rawrr/blob/f9974c1ef4bb8b0f3ad063ab161a84a16e946f0e/R/accessorFunctions.R#L147

maybe one of the apply function fill do the job

cpanse, 2020-12-28:
[image]

tobiasko, 2020-12-28:
```
==> R CMD build rawrr

* checking for file ‘rawrr/DESCRIPTION’ ... OK
* preparing ‘rawrr’:
* checking DESCRIPTION meta-information ... OK
* installing the package to build vignettes
* creating vignettes ... OK
* checking for LF line-endings in source and make files and shell scripts
* checking for empty or unneeded directories
Removed empty directory ‘rawrr/ACS_article/ACS_article_files’
* building ‘rawrr_0.2.0.tar.gz’

==> R CMD check rawrr_0.2.0.tar.gz

* using log directory ‘/Users/tobiasko/Documents/RStudio/rawrr.Rcheck’
* using R version 4.0.2 (2020-06-22)
* using platform: x86_64-apple-darwin17.0 (64-bit)
* using session charset: UTF-8
* checking for file ‘rawrr/DESCRIPTION’ ... OK
* checking extension type ... Package
* this is package ‘rawrr’ version ‘0.2.0’
* package encoding: UTF-8
* checking package namespace information ... OK
* checking package dependencies ... OK
* checking if this is a source package ... NOTE
Found the following apparent object files/libraries:
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.BackgroundSubtraction.dll
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.Data.dll
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.MassPrecisionEstimator.dll
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.RawFileReader.dll
Object files/libraries should not be included in a source package.
* checking if there is a namespace ... OK
* checking for executable files ... WARNING
Found the following executable files:
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.BackgroundSubtraction.dll
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.Data.dll
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.MassPrecisionEstimator.dll
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.RawFileReader.dll
  inst/rawrrassembly/bin/rawrr.exe
Source packages should not contain undeclared executable files.
See section ‘Package structure’ in the ‘Writing R Extensions’ manual.
* checking for hidden files and directories ... OK
* checking for portable file names ... OK
* checking for sufficient/correct file permissions ... OK
* checking whether package ‘rawrr’ can be installed ... OK
* checking installed package size ... OK
* checking package directory ... OK
* checking ‘build’ directory ... OK
* checking DESCRIPTION meta-information ... OK
* checking top-level files ... OK
* checking for left-over files ... OK
* checking index information ... OK
* checking package subdirectories ... OK
* checking R files for non-ASCII characters ... OK
* checking R files for syntax errors ... OK
* checking whether the package can be loaded ... OK
* checking whether the package can be loaded with stated dependencies ... OK
* checking whether the package can be unloaded cleanly ... OK
* checking whether the namespace can be loaded with stated dependencies ... OK
* checking whether the namespace can be unloaded cleanly ... OK
* checking dependencies in R code ... OK
* checking S3 generic/method consistency ... OK
* checking replacement functions ... OK
* checking foreign function calls ... OK
* checking R code for possible problems ... OK
* checking Rd files ... OK
* checking Rd metadata ... OK
* checking Rd cross-references ... NOTE
Unknown package ‘rawDiag’ in Rd xrefs
* checking for missing documentation entries ... OK
* checking for code/documentation mismatches ... OK
* checking Rd \usage sections ... OK
* checking Rd contents ... OK
* checking for unstated dependencies in examples ... OK
* checking sizes of PDF files under ‘inst/doc’ ... OK
* checking installed files from ‘inst/doc’ ... OK
* checking files in ‘vignettes’ ... NOTE
Files named as vignettes but with no recognized vignette engine:
   ‘vignettes/JPR_supplement.Rmd’
(Is a VignetteBuilder field missing?)
* checking examples ... OK
* checking for unstated dependencies in ‘tests’ ... OK
* checking tests ...
  Running ‘testthat.R’ OK

* checking for unstated dependencies in vignettes ... OK
* checking package vignettes in ‘inst/doc’ ... OK
* checking running R code from vignettes ...
  ‘JPR_TechnicalNote.Rmd’ using ‘UTF-8’... OK
 NONE
* checking re-building of vignette outputs ... OK
* checking PDF version of manual ... OK
* DONE
Status: 1 WARNING, 3 NOTEs

See
  ‘/Users/tobiasko/Documents/RStudio/rawrr.Rcheck/00check.log’
for details.

R CMD check succeeded
```

tobiasko, 2020-12-28:
I tried using `vapply() but gave up, since I could not figure out how to assign the correct template for the return value (a data.frame of variable row counts!) `map_df` worked out of the box!

cpanse, 2020-12-28:
Just use `lapply` or `sapply` for the moment.  I fix the template later.

closed 2020-12-29 by tobiasko