fgcz/rawrr #29: Usage of `@importFrom`

tobiasko, 2020-12-28:
"If you are using just a few functions from another package, the recommended option is to note the package name in the Imports: field of the DESCRIPTION file and call the function(s) explicitly using ::, e.g., pkg::fun(). Alternatively, though no longer recommended due to its poorer readability, use @importFrom, e.g., @importFrom pgk fun, and call the function(s) without ::."

taken from https://roxygen2.r-lib.org/articles/namespace.html#imports

Example found in `rawrr.R`:

```
#' Plot \code{rawrrChromatogramSet} objects
#'
#' @param x A \code{rawrrChromatogramSet} object to be plotted.
#' @param ... Passes additional arguments.
#' @param diagnostic Show diagnostic legend?
#' @author Tobias Kockmann, 2020.
#' @export
#' @importFrom grDevices hcl.colors
#' @importFrom graphics lines text
```
and many many more!

cpanse, 2020-12-28:
This are all base R packages.

cpanse, 2020-12-28:
@tobiasko https://cran.r-project.org/doc/manuals/r-release/R-exts.html#Specifying-imports-and-exports

tobiasko, 2020-12-29:
In the past? The golden age of Splus? ;-)

bash-3.2$ R CMD check rawrr
rawrr/              rawrr_0.1.7.tar.gz  rawrr_0.2.0.tar.gz  
bash-3.2$ R CMD check rawrr_0.2.0.tar.gz 
* using log directory ‘/Users/tobiasko/Documents/RStudio/rawrr.Rcheck’
* using R version 4.0.2 (2020-06-22)
* using platform: x86_64-apple-darwin17.0 (64-bit)
* using session charset: UTF-8
* checking for file ‘rawrr/DESCRIPTION’ ... OK
* checking extension type ... Package
* this is package ‘rawrr’ version ‘0.2.0’
* package encoding: UTF-8
* checking package namespace information ... OK
* checking package dependencies ... OK
* checking if this is a source package ... NOTE
Found the following apparent object files/libraries:
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.BackgroundSubtraction.dll
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.Data.dll
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.MassPrecisionEstimator.dll
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.RawFileReader.dll
Object files/libraries should not be included in a source package.
* checking if there is a namespace ... OK
*** checking for executable files ... WARNING**
Found the following executable files:
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.BackgroundSubtraction.dll
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.Data.dll
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.MassPrecisionEstimator.dll
  inst/rawrrassembly/bin/ThermoFisher.CommonCore.RawFileReader.dll
  inst/rawrrassembly/bin/rawrr.exe
Source packages should not contain undeclared executable files.
See section ‘Package structure’ in the ‘Writing R Extensions’ manual.
* checking for hidden files and directories ... OK
* checking for portable file names ... OK
* checking for sufficient/correct file permissions ... OK
* checking whether package ‘rawrr’ can be installed ... OK
* checking installed package size ... OK
* checking package directory ... OK
* checking ‘build’ directory ... OK
* checking DESCRIPTION meta-information ... OK
* checking top-level files ... OK
* checking for left-over files ... OK
* checking index information ... OK
* checking package subdirectories ... OK
* checking R files for non-ASCII characters ... OK
* checking R files for syntax errors ... OK
* checking whether the package can be loaded ... OK
* checking whether the package can be loaded with stated dependencies ... OK
* checking whether the package can be unloaded cleanly ... OK
* checking whether the namespace can be loaded with stated dependencies ... OK
* checking whether the namespace can be unloaded cleanly ... OK
* checking dependencies in R code ... OK
* checking S3 generic/method consistency ... OK
* checking replacement functions ... OK
* checking foreign function calls ... OK
* checking R code for possible problems ... OK
* checking Rd files ... OK
* checking Rd metadata ... OK
*** checking Rd cross-references ... WARNING**
Unbekanntes Paket ‘rawDiag’ in Rd Kreuzreferenzen
* checking for missing documentation entries ... OK
* checking for code/documentation mismatches ... OK
* checking Rd \usage sections ... OK
* checking Rd contents ... OK
* checking for unstated dependencies in examples ... OK
* checking installed files from ‘inst/doc’ ... OK
* checking files in ‘vignettes’ ... NOTE
Files named as vignettes but with no recognized vignette engine:
   ‘vignettes/JPR_supplement.Rmd’
(Is a VignetteBuilder field missing?)
* checking examples ... OK
* checking for unstated dependencies in ‘tests’ ... OK
* checking tests ...
  Running ‘testthat.R’
 OK
* checking for unstated dependencies in vignettes ... OK
* checking package vignettes in ‘inst/doc’ ... OK
* checking running R code from vignettes ...
  ‘JPR_TechnicalNote.Rmd’ using ‘UTF-8’... OK
 NONE
* checking re-building of vignette outputs ... OK
* checking PDF version of manual ... OK
* DONE

Status: 2 WARNINGs, 2 NOTEs
See
  ‘/Users/tobiasko/Documents/RStudio/rawrr.Rcheck/00check.log’
for details.

No warnings due to `purrr::map_df(x, tabulateSpectrum, accNames)`. Maybe you didn't Import the package in the `DESCRIPTION` file?

tobiasko, 2020-12-29:
Hmmm...and reading this (from #Specifying-imports-and-export)

"If a package only needs a few objects from another package it can use a fully qualified variable reference in the code instead of a formal import. A fully qualified reference to the function f in package foo is of the form foo::f. This is slightly less efficient than a formal import and also loses the advantage of recording all dependencies in the NAMESPACE file (but they still need to be recorded in the DESCRIPTION file). Evaluating foo::f will cause package foo to be loaded, but not attached, if it was not loaded already—this can be an advantage in delaying the loading of a rarely used package."

Somehow reminds of what I copied above from the roxygen2 webside.

tobiasko, 2020-12-29:
Something related to roxygen tags and the `NAMESPACE`:

I would avoid the usage of `@exportClass`, since exporting classes is S4-specific (which makes total sense because there is not even a formal class definition in S3). As the roxygen2 docu says:

"S4 classes: If you want others to be able to extend your class, @export it. If you want others to create instances of your class, but not extend it, @export the constructor function, but not the class."

rawrr does not define any S4 classes, so there is nothing to export.

The tag `@exportS3Method` is not needed to get correct directives for S3 methods like `plot.rawrrSpectrum`. A simple `@export` tag in each method definition does the job. I removed all of these tags locally:

```
#' @export
## #' @export readChromatogram
## #' @exportClass rawrrChromatogram
## #' @exportClass rawrrChromatogramSet
## #' @exportS3Method plot rawrrChromatogram
## #' @exportS3Method plot rawrrChromatogramSet
#' @importFrom utils read.csv2
#' @examples
```
to arrive at a `NAMESPACE` like:

```
# Generated by roxygen2: do not edit by hand

S3method(plot,rawrrChromatogram)
S3method(plot,rawrrChromatogramSet)
S3method(plot,rawrrSpectrum)
S3method(print,rawrrSpectrum)
S3method(summary,rawrrChromatogram)
S3method(summary,rawrrSpectrum)
export(basePeak)
export(dependentScan)
export(faimsVoltageOn)
export(is.rawrrChromatogram)
export(is.rawrrSpectrum)
export(makeAccessor)
export(massRange)
export(masterScan)
export(rawrrSpectrum)
export(readChromatogram)
export(readFileHeader)
export(readIndex)
export(readSpectrum)
export(sampleFilePath)
export(scanNumber)
export(tic)
export(validate_rawrrIndex)
export(validate_rawrrSpectrum)
importFrom(grDevices,hcl.colors)
importFrom(graphics,legend)
importFrom(graphics,lines)
importFrom(graphics,text)
importFrom(stats,na.omit)
importFrom(utils,packageVersion)
importFrom(utils,read.csv2)
importFrom(utils,read.table)
```
All checks pass.

closed 2021-12-02 by cpanse