fgcz/rawrr #31: Titles in .plotChromatogramAndFit?

BioAlvaro, 2021-02-03:
Hi guys,

When plotting multiple raw files with iRT peptides, I'm using the function .plotChromatogramAndFit that you showed.
I want to add a title being the name of each of those raw files to the plot, though I'm not being successful. Any ideas?

`plot(x, main=???); legend("topright", legend=i, title='Instrument Model', bty = "n", cex=0.75)`

Thanks a lot for the great library :)

cpanse, 2021-02-03:
@BioAlvaro would that commit solve the issue?

 https://github.com/fgcz/rawrr/commit/583117002613718b08461795b3f81410ad7e8797

[image]

taken from

http://fgcz-ms.uzh.ch/~cpanse/rawrr/test/functional_test.html

if yes 
```
install.packages('http://fgcz-ms.uzh.ch/~cpanse/rawrr_0.2.2.tar.gz', repo=NULL)
```

and then just provide a value for the argument `main`, e.g., 
```
# http://fgcz-ms.uzh.ch/~cpanse/20181113_010_autoQC01.raw
rawfile <- file.path(Sys.getenv('HOME'), "Downloads","20181113_010_autoQC01.raw")  
XIC <- rawrr::readChromatogram(rawfile, mass=c(669.8381, 726.8357), tol=1000)
plot(XIC, main=paste0("filename: ", basename(rawfile)))
```

BioAlvaro, 2021-02-03:
Yes! It worked 👍🏻 

Thanks a lot for the fast answer :)

closed 2021-02-03 by BioAlvaro