fgcz/rawrr #32: Noise value for individual mass peaks

YasinEl, 2021-03-09:
Hello, and thank you for your package!

I was wondering if it is somehow possible to get the Noise value that is reported for every single mass peak in Thermo's .raw files.

Thanks!

tobiasko, 2021-03-10:
Dear @YasinEl,

sure that is possible! Following what we have published yesterday :-) https://pubs.acs.org/doi/10.1021/acs.jproteome.0c00866 you can even generate your own accessor function for peak noise values:

```
> rawfile <- file.path(Sys.getenv('HOME'), "Downloads", "20181113_010_autoQC01.raw")
> rawfile
[1] "/Users/tobiasko/Downloads/20181113_010_autoQC01.raw"
> i <- 9594
> S <- rawrr::readSpectrum(rawfile = rawfile, scan = i)
> S
[[1]]
Total Ion Current:	 62374688
Scan Low Mass:	 100
Scan High Mass:	 1015
Scan Start Time (Min):	 15.42
Scan Number:	 9594
Base Peak Intensity:	 12894520
Base Peak Mass:	 860.4223
Scan Mode:	 FTMS + c NSI Full ms2 487.2567@hcd27.00 [100.0000-1015.0000]
======= Instrument data =====   :	NULL
Multiple Injection:	h 
Multi Inject Info:	IT=2.8 
AGC:	On
Micro Scan Count:	1
Scan Segment:	1
Scan Event:	2
Master Index:	0
Charge State:	2
Monoisotopic M/Z:	0.0000
Ion Injection Time (ms):	2.811
Max. Ion Time (ms):	55.00
FT Resolution:	30000
MS2 Isolation Width:	1.40
MS2 Isolation Offset:	0.00
AGC Target:	100000
HCD Energy:	27.00
Analyzer Temperature:	30.29
=== Mass Calibration:	NULL
Conversion Parameter B:	211689007.9670
Conversion Parameter C:	-88620884.7343
Temperature Comp. (ppm):	-6.03
RF Comp. (ppm):	0.00
Space Charge Comp. (ppm):	-0.06
Resolution Comp. (ppm):	-0.04
Number of Lock Masses:	2
Lock Mass #1 (m/z):	371.1010
Lock Mass #2 (m/z):	445.1200
Lock Mass #3 (m/z):	0.0000
LM Search Window (ppm):	15.0
LM Search Window (mmu):	0.0
Number of LM Found:	0
Last Locking (sec):	0.0
LM m/z-Correction (ppm):	3.15
=== Ion Optics Settings:	NULL
S-Lens RF Level:	60.00
S-Lens Voltage (V):	21.00
Skimmer Voltage (V):	15.00
Inject Flatapole Offset (V):	5.00
Bent Flatapole DC (V):	2.00
MP2 and MP3 RF (V):	628.00
Gate Lens Voltage (V):	1.00
C-Trap RF (V):	1010.0
====  Diagnostic Data:	NULL
Dynamic RT Shift (min):	0.00
Intens Comp Factor:	0.6852
Res. Dep. Intens:	1.333
CTCD NumF:	0
CTCD Comp:	5.321
CTCD ScScr:	0.000
RawOvFtT:	225788.1
LC FWHM parameter:	15.0
Rod:	0
PS Inj. Time (ms):	13.750
AGC PS Mode:	-1
AGC PS Diag:	0
HCD Energy eV:	23.681
AGC Fill:	1.00
Injection t0:	-0.020
t0 FLP:	860.42
Access Id:	0
Analog Input 1 (V):	0.000
Analog Input 2 (V):	0.000

attr(,"class")
[1] "rawrrSpectrumSet"
> S[[1]]$noises
  [1] 24563.38 24559.59 25864.96 25866.78 25427.94 25113.02 24894.63 26068.89 26085.78 26253.31 26342.82 27035.06 27033.53
 [14] 27020.81 26991.55 26970.78 26607.42 26574.18 26572.64 26468.55 26400.05 25540.01 26206.67 26684.34 26786.95 26877.38
 [27] 26939.79 26968.94 26817.41 26723.80 26561.83 26464.12 26463.38 26447.87 26265.03 26261.98 26243.98 26241.00 26228.81
 [40] 26250.30 26257.45 26278.55 26287.98 26357.33 26371.79 26379.98 26363.67 26358.24 26336.62 26298.83 26282.51 26277.18
 [53] 26271.39 26238.96 26200.41 26197.32 26183.76 26183.35 26178.21 26106.25 26097.72 26089.20 26084.02 26083.65 26175.71
 [66] 26197.22 26556.51 27122.80 27162.43 27282.32 27324.60 27639.10 27645.09 27693.55 27705.69 27748.30 27814.58 27996.20
 [79] 28021.08 28033.02 28051.78 28057.82 28160.99 28136.65 28120.05 28082.64 27954.59 27945.63 27826.57 27809.94 27808.66
 [92] 27790.52 27772.53 27498.53 27480.24 27354.27 27316.66 26752.06 26566.07 26551.98 26509.03 26314.59 26317.16 26327.46
[105] 26342.01 26342.86 26357.41 26358.26 26365.94 26467.42 26616.55 26620.82 26690.80 26695.06 26699.44 26769.32 26777.12
[118] 26959.56 27029.97 27041.84 27235.08 27246.98 27259.08 27452.10 27463.33 27464.19 27465.21 27632.84 27716.98 27728.90
[131] 27922.12 27934.00 27946.07 28069.81 28060.78 28051.75 27979.89 27970.97 27887.41
> makeAccessor(key = "noises", returnType = "double")
function(x) {

    stopifnot(is.rawrrSpectrum(x))
    if (key %in% names(x)) {

      cl <- call(paste0("as.", returnType), x[[key]])
      eval(cl)

    } else {

      stop(paste0(key, " is not available!"))
    }
  }
<bytecode: 0x7fb257d45248>
<environment: 0x7fb25b5476a8>
> noises <- makeAccessor(key = "noises", returnType = "double")
> noises(S[[1]])
  [1] 24563.38 24559.59 25864.96 25866.78 25427.94 25113.02 24894.63 26068.89 26085.78 26253.31 26342.82 27035.06 27033.53
 [14] 27020.81 26991.55 26970.78 26607.42 26574.18 26572.64 26468.55 26400.05 25540.01 26206.67 26684.34 26786.95 26877.38
 [27] 26939.79 26968.94 26817.41 26723.80 26561.83 26464.12 26463.38 26447.87 26265.03 26261.98 26243.98 26241.00 26228.81
 [40] 26250.30 26257.45 26278.55 26287.98 26357.33 26371.79 26379.98 26363.67 26358.24 26336.62 26298.83 26282.51 26277.18
 [53] 26271.39 26238.96 26200.41 26197.32 26183.76 26183.35 26178.21 26106.25 26097.72 26089.20 26084.02 26083.65 26175.71
 [66] 26197.22 26556.51 27122.80 27162.43 27282.32 27324.60 27639.10 27645.09 27693.55 27705.69 27748.30 27814.58 27996.20
 [79] 28021.08 28033.02 28051.78 28057.82 28160.99 28136.65 28120.05 28082.64 27954.59 27945.63 27826.57 27809.94 27808.66
 [92] 27790.52 27772.53 27498.53 27480.24 27354.27 27316.66 26752.06 26566.07 26551.98 26509.03 26314.59 26317.16 26327.46
[105] 26342.01 26342.86 26357.41 26358.26 26365.94 26467.42 26616.55 26620.82 26690.80 26695.06 26699.44 26769.32 26777.12
[118] 26959.56 27029.97 27041.84 27235.08 27246.98 27259.08 27452.10 27463.33 27464.19 27465.21 27632.84 27716.98 27728.90
[131] 27922.12 27934.00 27946.07 28069.81 28060.78 28051.75 27979.89 27970.97 27887.41
> 
```

Hope this helps!

YasinEl, 2021-03-11:
Thank you, that's exactly what I was looking for :)

closed 2021-03-11 by YasinEl