fgcz/rawrr #34: ReadChromatogram intensity or area under the curve

Linda24bc, 2021-03-11:
Hi, I really appreciated your efforts in making this package.
I have a few questions about the readChromatogram function, I get the XIC for an analyte, such as the 836.07492  at tol:100.
Once I got the XIC, there are an equal number of retention time and intensity, when I look at the details of the RT and intensity, for example, at rt33.03 min, the output intensity from this function is 23020686, but for the raw data, the NL is 7.11E6. So I was wondering, how is the output 23020686 calculated?

Another question is that, is there any way to get the area under the curve of XIC? I want to do the quantitation analysis. 

Thank you.

tobiasko, 2021-03-11:
Hi @Linda24bc,

what do you mean by "for the raw data, the NL is 7.11E6". Where is this value coming from (Thermo QualBrowser, Freestyle) and normalised level (NL) of what?

Best,
Tobi

cpanse, 2021-03-11:
#  AUC 

for example to compute the AUC of the iRT peptide XICs in [Figure 4](https://pubs.acs.org/doi/10.1021/acs.jproteome.0c00866)

```
# Download via FTP
f <- "ftp://massive.ucsd.edu/MSV000086542/raw/20181113_010_autoQC01.raw"
rawfile <- file.path(Sys.getenv('HOME'), "Downloads", "20181113_010_autoQC01.raw")
if (!file.exists(rawfile)) { download.file(f, rawfile) }

iRTmz <- c(487.2571, 547.2984, 622.8539, 636.8695, 644.8230, 669.8384, 683.8282,
            683.8541, 699.3388, 726.8361, 776.9301)
names(iRTmz) <- c("LGGNEQVTR", "YILAGVENSK", "GTFIIDPGGVIR", "GTFIIDPAAVIR",
                 "GAGSSEPVTGLDAK", "TPVISGGPYEYR", "VEATFGVDESNAK",
                 "TPVITGAPYEYR", "DGLDAASYYAPVR", "ADVTPADFSEWSK",
                 "LFLQFGAQGSPFLK")
C <- rawrr::readChromatogram(rawfile, mass = iRTmz, tol = 10, type = "xic", filter = "ms")
plot(C)

auc.rawrrChromatogram <- function(x){
        times <- x$times; intensities <- x$intensities
        sum(diff(times) * (head(intensities, -1) + tail(intensities, -1))) / 2
}

(auc <- sapply(C, auc.rawrrChromatogram)  )
```

output:
```
 [1] 2836329.9 1826909.0  449959.6 1108052.4 3178564.6 1655764.7
 [7] 1624008.0 2186544.0 3002209.3 1717397.3  261542.6
```

tobiasko, 2021-03-15:
No reply...

Linda24bc, 2021-03-15:
> Hi @Linda24bc,
> 
> what do you mean by "for the raw data, the NL is 7.11E6". Where is this value coming from (Thermo QualBrowser, Freestyle) and normalised level (NL) of what?
> 
> Best,
> Tobi

Sorry for the delayed reply. I am using the Thermo Qualbrowser. The NL value I referred to comes from the picture here. 
[image]
Thank you.

tobiasko, 2021-03-15:
Ok. And what are we looking at? Your screenshot suggests it is a single mass spectrum (a specific scan). Is this true?

Linda24bc, 2021-03-15:
> Ok. And what are we looking at? Your screenshot suggests it is a single mass spectrum (a specific scan). Is this true?

This is LC-MS run. It detected the intact proteins, at charge state 19+, it shows the isotopologue, I specifically zoom into the 836.07492 range.

Linda24bc, 2021-03-15:
> Could you please share the complete window?

[image]
[image]

tobiasko, 2021-03-15:
ok. as suspected, these are mass spectra, not chromatograms (the x dim. shows m/z values!). The normalised level (NL) is the equal to what one calls a base peak intensity (the most intense signal in the spectrum), since the default view shows relative values.

Linda24bc, 2021-03-15:
> t chromatograms (the x dim. shows m/z values!). The normalised level (NL) is the equal to what one calls a base peak intensity (the most intense signal in the spectrum), since the default view shows relative values.

This is the spectra at RT 33.03 min.  My question is that when I do the XIC for  836.07492 at tol 100, with your function, the intesity is 23020686, while in the raw data, it shows 7.11E6 at RT 33.03min. So I was wondering, how is the output 23020686 calculated?

tobiasko, 2021-03-15:
Sorry, I don't understand your question. An XIC integrates all signals around a target mass with a given tolerance (here 100 ppm!) over all ms scans in your files. Why do you even expect to get similar values?

Linda24bc, 2021-03-15:
Sorry, I am making the question complicated. Let me start over. This time. the target m/z is still 836.07492, the tol is 100ppm, so the mm/z range is 835.990743-836.157957. From my spectrum, the peaks that fall within the range are 836.02218, 836.07484, 836.12693 at RT 33.03min. These three m/z peaks are from the same intact protein, all are at 19+ charge state. From the raw data, the NL for these three peaks are 3736123.5, 4202202.5, 3550020.3, the sum of these three values is 11488346.3. 
After I use your function to get the XIC of 836.07492 with tol 100ppm, the intensity at 33.03min is 8166443.520. I don't know how you calculated the 8166443.520 value. 
In my opinion, we use the R function to simplify the data analysis process, but we can still get the same value by manual validation.

tobiasko, 2021-03-16:
In general: You can not just sum the apex hight of m/z peaks acquired in profile mode (your scan type shows `p`). Your peaks are not discrete data items (centroids). If your apex is located at position 836.07492 what kind of signal do you expect at 836.07491? Yes, a non zero, not to say strongly positive value. Depending on the characteristics of your scan (like resolving power) the signal shows some kind of gaussian distribution around the apex. You need to integrate here. Regarding your assumption that we calculate something: No, we don't. We ask the RawFileReader API to do the calculation and return the result as-is.

Linda24bc, 2021-03-16:
> . We ask the RawFileReader API to do the calculation and return the result as-is.

Thank you. I got your point. I got another question about the AUC function, I use the same targeted m/z, and tol, the AUC output from your function and the automated area calculated from xcalibur is different, do you know the reason behind this?

cpanse, 2021-03-16:
@Linda24bc it is a question of the interval boundaries.

tobiasko, 2021-03-17:
Not sure which Xcalibur function you mean. Thermo has a couple of peak caller algorithms implemented. They all estimate peak areas a bit differently and have different underlying assumptions about theoretical peak properties. The difference could also come from a different peak filter setting.

closed 2021-03-24 by cpanse