fgcz/rawrr #41: Error if too many scans are selected

rokaempf, 2021-07-26:
Hello again :-)

Unfortunately I have a little problem, which I don't know how to solve...

readSpectrum() gives the following error message:
`Error in source(tfo, local = TRUE) : 
  negative length vectors are not allowed`

A little example how my code looks:

```
library(rawrr)
library(tidyverse)

#reading Index and selecting scans with ms_order = "Ms"
ms_order <- "Ms"
IDX <- as_tibble(readIndex(path))
scans <- IDX %>% filter(MSOrder == ms_order) %>% pull(scan)

SPC <- readSpectrum(path, scan = scans)
```

I have about 12000 scans in total and about 2500 MS1 scans in my raw file. It works fine as long as I only read about 2000 scans. After that I receive the error message. I don't know if it due to memory limits on my machine.

Thanks for your help in advance!
kaempfro

tobiasko, 2021-07-26:
Hi @rokaempf

could you please post the content of `path` and `scans` and your session info?

tobiasko, 2021-07-26:
hmmm. ok. Most likely it is connected to some sort of resource shortage. But just to make sure: Could you please move to the latest rawrr version? We have started using Bioconductor:

https://bioconductor.org/packages/release/bioc/html/rawrr.html

and pushed the version number to 1.0.X. We also recommend using the BiocManager for package installation:

```
if (!requireNamespace("BiocManager", quietly = TRUE))
    install.packages("BiocManager")

BiocManager::install("rawrr")
```

cpanse, 2021-07-26:
Hi @rokaempf 

if you apply the divide-and-conquer paradigm it should work.

```{r}
#R >= 4.1
(rawfile <- rawrr::sampleFilePath())
S <- (rawrr::readIndex(rawfile) |>
  subset (MSOrder == 'Ms'))$scan |>
  BiocParallel::bplapply(FUN=function(i){rawrr::readSpectrum(rawfile=rawfile,i)[[1]]})
```

C

rokaempf, 2021-07-26:
Hey @tobiasko  and @cpanse 
Perfect! Thanks for the update!
I'll try both and let you know!

Great support and thanks again for all you work!

rokaempf, 2021-07-26:
Maybe not the right place to report this issue, but the Bioconductor installation does not work for me:

```
> rawrr::installRawrrExe()
Error: .onLoad failed in loadNamespace() for 'rawrr', details:
  call: buildRawrrExe()
  error: Could not find 'msbuild' or 'xbuild' in the path. Therefore, it is not possible to build the 'rawrr.exe' assembly from source code.
Try to run rawrr::installRawrrExe().
```

cpanse, 2021-07-26:
Dear @rokaempf,

 thanks for reporting.

can you please terminate R and restart with the install procedure.
```
if (!requireNamespace("BiocManager", quietly = TRUE))
    install.packages("BiocManager")

BiocManager::install("rawrr")
rawrr::installRawFileReaderDLLs()
rawrr::installRawrrExe()
```

If it is not working. what OS are you running?

TNX
C

rokaempf, 2021-07-27:
Dear @cpanse 

I tried it, but still no success!
I'm running it on Windows 10 Education Version 2004 (see SessionInfo below)

```
> if (!requireNamespace("BiocManager", quietly = TRUE))
+     install.packages("BiocManager")
> 
> BiocManager::install("rawrr", force = TRUE)
'getOption("repos")' replaces Bioconductor standard repositories, see '?repositories' for
details

replacement repositories:
    CRAN: https://cran.rstudio.com/

Bioconductor version 3.13 (BiocManager 1.30.16), R 4.1.0 (2021-05-18)
Installing package(s) 'rawrr'
trying URL 'https://bioconductor.org/packages/3.13/bioc/bin/windows/contrib/4.1/rawrr_1.0.2.zip'
Content type 'application/zip' length 1398523 bytes (1.3 MB)
downloaded 1.3 MB

package ‘rawrr’ successfully unpacked and MD5 sums checked

The downloaded binary packages are in
	C:\...\Local\Temp\RtmpMlMb3z\downloaded_packages
> rawrr::installRawFileReaderDLLs()
Error: .onLoad failed in loadNamespace() for 'rawrr', details:
  call: buildRawrrExe()
  error: Could not find 'msbuild' or 'xbuild' in the path. Therefore, it is not possible to build the 'rawrr.exe' assembly from source code.
Try to run rawrr::installRawrrExe().
> rawrr::installRawrrExe()
Error: .onLoad failed in loadNamespace() for 'rawrr', details:
  call: buildRawrrExe()
  error: Could not find 'msbuild' or 'xbuild' in the path. Therefore, it is not possible to build the 'rawrr.exe' assembly from source code.
Try to run rawrr::installRawrrExe().
> sessionInfo()
R version 4.1.0 (2021-05-18)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 19041)
```

Thanks!

cpanse, 2021-07-27:
Good morning @rokaempf

you found a bug. Can you please try that version?
 
```{r}
#R 4.1
install.packages('http://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr_1.1.13.tar.gz', repo=NULL)
```
If it works (please let us know) I will push it to Bioconductor.

Thank you again for reporting,

Christian

rokaempf, 2021-07-27:
Thanks @cpanse and @tobiasko 

The new version works fine and divide-and-conquer also helped!
Thanks for all your work on this package!

closed 2021-07-27 by cpanse