fgcz/rawrr #46: How to access S/N values

papanlipotype, 2022-01-17:
Dear all,
is there a way of accessing the S/N values from a readSpectrum-Object? I cant seem to find the info in the list. 
Thanks

tobiasko, 2022-02-14:
Dear @papanlipotype,

the S/N values are never stored in the raw file. They need to be calculated. This is for instance done by the `plot` method for `rawrrSpectrum` objects in case `SN = TRUE`:

```
plot.rawrrSpectrum <- function(x, relative = TRUE, centroid = FALSE, SN = FALSE,
                              legend = TRUE, diagnostic = FALSE, ...){

    stopifnot(is.rawrrSpectrum(x))

    if (centroid) {

        stopifnot(x$centroidStream)

        if (SN) {

            plot(x = x$centroid.mZ, y = x$centroid.intensity/x$noise,
                 type = "h",
                 xlim = x$massRange,
                 xlab = "Centroid m/z",
                 ylab = "Centroid Signal/Noise",
                 frame.plot = FALSE, ...
            )
```

Currently the plot function only supports this for centroided data, although it has meanwhile become clear that the noise attribute is also available for (reduced) profile mode data.

But you could easily write a function that combines data retrieval and S/N calculation.

Hope this helps,
Tobi

papanlipotype, 2022-02-14:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Sorry, my bad- of course I meant the noise value, not S/N... However, we always acquire our data in profile mode, from what I know, the recuced profile mode was only available on the LTQ-Orbi, not the Q-Exactives. Is there any way to get the noise values for profile mode data?
Thanks,
Cyrus

[https://www.lipotype.com/cms/wp-content/uploads/2020/07/lipotype_logo_signatur-e1593700444849.png] <https://www.lipotype.com>

Lipotype GmbH
Tatzberg 47
01307 Dresden
Germany

Tel.:   +49 351 796 5345
Fax:    +49 351 796 5349
Email:  ***@***.******@***.***>
Web:    www.lipotype.com<https://www.lipotype.com/>

Follow us on [LinkedIn] <https://www.linkedin.com/company/lipotype/>  [Twitter] <https://twitter.com/lipotype_gmbh/>  [Instagram] <https://www.instagram.com/lipotype_gmbh/>  [Facebook] <https://www.facebook.com/lipotype/> , and subscribe to our newsletter<https://www.lipotype.com/mailing-list/> to receive the latest news about lipidomics research and Lipotype.

VAT ID: DE282974828
County Court Dresden HRB 31124
Managing Directors: Prof. Dr. Kai Simons (CEO), Dr. Oliver Uecke (COO) and Dr. Christian Klose (CTO)

This message is intended for the addressee only as it contains private and confidential information. The contents are not to be disclosed to anyone other than the addressee. Unauthorised recipients are requested to comply with the above and to inform the sender immediately of any errors in transmission.

tobiasko, 2022-02-14:
Apparently reduced profile mode is what the Q Exactive series uses when you select profile mode in the instrument method. see https://github.com/fgcz/rawrr/issues/37. I also thought that full profile mode would be triggered by selecting profile mode acquisition,  but that doesn't seem to be the case.

I had a look at recent data from our Q Exactive for a "FTMS + p NSI Full ms [350.0000-1500.0000]" scan. `rawrr` just gives access to the intensity and m/z values (no noise attributes). But I guess the noise was written to the raw file. Most likely to a different data stream that our managed code currently ignores (again, have a look at issue #37).

@cpanse We should really push the code changes to make this stream available!

papanlipotype, 2022-02-14:
Hi Tobias,
Yes, the noise value is definitely written to the raw files, as we read it out with out own raw-converter. We had just discovered the rawrr library (I guess it is still pretty new after all). It would be very useful if that could become available for profile mode in rawrr too.
Thanks for your help and replies,
Cyrus

[https://www.lipotype.com/cms/wp-content/uploads/2020/07/lipotype_logo_signatur-e1593700444849.png] <https://www.lipotype.com>

tobiasko, 2022-02-17:
ok @tobiasko . We will try to do our best to make this happen in the near future, since quite some people requested this already!

Linda24bc, 2024-09-06:
I was wondering if the noise value can be extract from the raw file now with readSpectrum? Thank you

cpanse, 2024-09-09:
> I was wondering if the noise value can be extract from the raw file now with readSpectrum? Thank you

sorry not yet