fgcz/rawrr #47: Issue with readChromatogram when type = "xic"

Arthfael, 2022-02-01:
I have been using this package for ~1 year now, specifically the readChromatogram function for extracting "tic" and "base peak". It works nicely and has been very useful. However, I have just for the first time tried to extract an "xic" for some masses of interest, and here is what I get:

```
XICs <- lapply(Raws, function(raw) { readChromatogram(raw, masses, tols) })
Error in .rawrrSystem2Source(rawfile, input = mass, rawrrArgs = sprintf("xic %f %s",  : 
  **Rcode file to parse does not exist. 'C:\Users\MyUserName\AppData\Local/R/cache/R/rawrr/rawrrassembly/rawrr.exe' failed for an unknown reason.
Please check the debug files:
	C:\Users\MyUserName\AppData\Local\Temp\2\RtmpeQSRv0\file62986016698d.stderr
	C:\Users\MyUserName\AppData\Local\Temp\2\RtmpeQSRv0\file629849cf6634.stdout
and the System Requirements
Called from: .rawrrSystem2Source(rawfile, input = mass, rawrrArgs = sprintf("xic %f %s", 
    tol, shQuote(filter)))**
```
This is on a Windows 2019 Server machine, using R version 4.1.0 (2021-05-18) in RStudio 1.4.1717.
File "C:/Users/MyUserName/AppData/Local/R/cache/R/rawrr/rawrrassembly/rawrr.exe" does exist, but maybe this is an issue with slashes in Windows, since the error uses inconsistently backwards (Windows) and forward (Linux) slashes? In which case, including normalizePath(..., winslash = "/") would probably be enough to fix it?

tobiasko, 2022-02-14:
Hi @Arthfael 

That is indeed a bit strange! Could you please upload your `sessionInfo`? Do you get the same error when reading from a single file (outside of the `lapply` function)? Are other reading functions of the package also affected (`readIndex`)?

Best,
Tobi

Arthfael, 2022-02-16:
Dear Tobi,

Yes, same error if I run on a single file, regardless of whether I use
forward or backwards slashes:
> Raws[1]
[1] "Q:\\MS\\Acquired data\\test_QC02_25ng_20210727161519.raw"
Both
> XICs <- readChromatogram(Raws[1], masses, tols)
and
> XICs <- readChromatogram(normalizePath(Raws[1], winslash = "/"), masses,
tols)
produce the error.

(*NB:*
I get the same result whether I install the package with >
devtools::install_github("cpanse/rawrr") or >
devtools::install_github("fgcz/rawrr")
However, if I do it with > install.packages('
http://fgcz-ms.uzh.ch/~cpanse/rawrr_0.2.1.tar.gz', repo = NULL) I instead
get this error:
> XICs <- readChromatogram(Raws[1], masses, tols)
Error in !silent : invalid argument type
In addition: Warning message:
In file(filename, "r", encoding = encoding) :
  cannot open file
'C:\Users\MyUser\AppData\Local\Temp\2\Rtmp29h2p2\file31187bb1737': No such
file or directory
But this is an older version I think?)

Here it my sessionInfo output:
> sessionInfo()
R version 4.1.0 (2021-05-18)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows Server x64 (build 17763)

Matrix products: default

locale:
[1] LC_COLLATE=English_United States.1252  LC_CTYPE=English_United
States.1252    LC_MONETARY=English_United States.1252
[4] LC_NUMERIC=C                           LC_TIME=English_United
States.1252

attached base packages:
 [1] stats4    grid      parallel  compiler  stats     graphics  grDevices
utils     datasets  methods   base

other attached packages:
 [1] rawrr_1.3.3          aRmel_2.0.1.15       DEP_1.14.0
MSnbase_2.18.0       ProtGenerics_1.24.0
 [6] mzR_2.26.1           Rcpp_1.0.7           qvalue_2.24.0
 sva_3.40.0           BiocParallel_1.26.1
[11] genefilter_1.74.0    mgcv_1.8-35          nlme_3.1-152
limma_3.48.1         UniProt.ws_2.32.0
[16] RCurl_1.98-1.3       RSQLite_2.2.7        GO.db_3.13.0
AnnotationDbi_1.54.1 IRanges_2.26.0
[21] S4Vectors_0.30.0     Biobase_2.52.0       BiocGenerics_0.38.0
 biomaRt_2.48.2       fs_1.5.0
[26] tibble_3.1.3         magrittr_2.0.1       htmlwidgets_1.5.3
 svDialogs_1.0.3      gridExtra_2.3
[31] VennDiagram_1.6.20   futile.logger_1.4.3  statmod_1.4.36
pdftools_3.0.1       xml2_1.3.2
[36] Peptides_2.4.4       plotly_4.9.4.1       openxlsx_4.2.4.9000
 coin_1.4-1           survival_3.2-11
[41] snow_0.4-3           minpack.lm_1.2-1     modeest_2.4.0
 rstudioapi_0.13      ggrepel_0.9.1
[46] rgl_0.107.10         reshape2_1.4.4       reshape_0.8.8
 gtools_3.9.2         ggpubr_0.4.0
[51] ggplot2_3.3.5        gplots_3.1.1         stringr_1.4.0

[... 58 lines cut ...]
foreign_0.8-81              statip_0.2.3
[117] farver_2.1.0                mzID_1.30.0                 stable_1.1.4
               digest_0.6.27
[121] BiocManager_1.30.16         shiny_1.6.0
GenomicRanges_1.44.0        car_3.0-11
[125] broom_0.7.8                 later_1.2.0                 ncdf4_1.17
               httr_1.4.2
[129] ComplexHeatmap_2.8.0        colorspace_2.0-2            XML_3.99-0.6
               splines_4.1.0
[133] rmutil_1.1.5                gmm_1.6-6
sessioninfo_1.1.1           xtable_1.8-4
[137] jsonlite_1.7.2              futile.options_1.0.1
 timeDate_3043.102           testthat_3.0.4
[141] modeltools_0.2-23           R6_2.5.0                    mime_0.11
              pillar_1.6.1
[145] htmltools_0.5.1.1           svGUI_1.0.1                 glue_1.4.2
               fastmap_1.1.0
[149] DT_0.18                     codetools_0.2-18
 pkgbuild_1.2.0              mvtnorm_1.1-2
[153] utf8_1.2.1                  lattice_0.20-44             curl_4.3.2
               zip_2.2.0
[157] desc_1.3.0                  munsell_0.5.0
GetoptLong_1.0.5            GenomeInfoDbData_1.2.6
[161] iterators_1.0.13            impute_1.66.0               haven_2.4.1
              gtable_0.3.0

Arthfael, 2022-02-16:
(I also tried the two branches, e.g. >
***@***.***_3_13", force = TRUE), but get
the same error as with the master.)

tobiasko, 2022-02-16:
Hi @Arthfael 

since release 3.13 rawrr is a Bioconductor package, see

https://bioconductor.org/packages/release/bioc/html/rawrr.html

I would recommend to use the latest BioC release 3.14 to install rawrr with the help of the BiocManager:

```
if (!require("BiocManager", quietly = TRUE))
    install.packages("BiocManager")

BiocManager::install("rawrr")
```

I am honestly not sure what you get when installing from GitHub directly using `devtools::`. Version 0.2.1 is stone age. Please don't use it.

Arthfael, 2022-02-16:
Dear Tobi,

Thank you, I now remember discovering and also trying the Bioconductor
version when I originally encountered the problem. Unfortunately I also get
the same error with it.
Kind regards,

Armel

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

tobiasko, 2022-02-16:
Did you follow this

```
https://bioconductor.org/packages/release/bioc/install/rawrr/INSTALL
``` 

or specifically did you install the RawFileReader DLLs and rawrr exe on your system:

```
> installRawFileReaderDLLs()
Do you accept the Thermo License agreement 'X:/Rlibs/rawrr/rawrrassembly/RawFileReaderLicense.txt'? [Y/n]: Y
trying URL 'https://github.com/thermofisherlsms/ThermoRawFileParser/raw/master/packages/ThermoFisher.CommonCore.RawFileReader.4.0.26/lib//ThermoFisher.CommonCore.Data.dll'
Content type 'application/octet-stream' length 360448 bytes (352 KB)
downloaded 352 KB

trying URL 'https://github.com/thermofisherlsms/ThermoRawFileParser/raw/master/packages/ThermoFisher.CommonCore.RawFileReader.4.0.26/lib//ThermoFisher.CommonCore.MassPrecisionEstimator.dll'
Content type 'application/octet-stream' length 11264 bytes (11 KB)
downloaded 11 KB

trying URL 'https://github.com/thermofisherlsms/ThermoRawFileParser/raw/master/packages/ThermoFisher.CommonCore.RawFileReader.4.0.26/lib//ThermoFisher.CommonCore.RawFileReader.dll'
Content type 'application/octet-stream' length 620544 bytes (606 KB)
downloaded 606 KB

                  ThermoFisher.CommonCore.Data.dll ThermoFisher.CommonCore.MassPrecisionEstimator.dll 
                                                 0                                                  0 
         ThermoFisher.CommonCore.RawFileReader.dll 
                                                 0 
> installRawrrExe()
trying URL 'http://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr.1.1.12.exe'
Content type 'application/x-msdos-program' length 26624 bytes (26 KB)
downloaded 26 KB

MD5 6d5f6d19512eaba73e92d2bfb474e6f2 C:\Users\tobiasko.FGCZ-NET\AppData\Local/R/cache/R/rawrr/rawrrassembly/rawrr.exe
[1] 0
> 
```

tobiasko, 2022-02-16:
The problem is:

Bioc was not willing to host the binary rawrr.x.exe as part of the package. So it is downloaded on demand after the package installation through `BiocManager`. Same-same for the RawFileReader dlls. They are fetched from a Thermo GitHub Repo. I know this is kind of annoying, but it was the only option.

Arthfael, 2022-02-16:
Dear Tobi,

Thanks, but unfortunately this still doesn't fix it: even after a fresh
install from Bioconductor, running both installRawrrExe() and
installRawFileReaderDLLs(), and restarting my R session, I still get the
same error.
NB: I had run both lines originally before reporting my issue here, and can
confirm that all 3 dlls and the rawrr.exe were already and are still
present at the location (hence why I assumed the issue was with the slash
orientation in the path causing it to somehow not be read properly.)
Best regards,

Armel

Arthfael, 2022-02-16:
If that helps:
The debug files referred by the error message do not exist. But at the
debug location, there is a txt file with a similar name (and an empty
rs-graphics-someweirdcode folder) containing what looks like two error
codes: 524 and 670. (I tried searching what they mean but not sure they are
the same respectively CloudFlare- and Silverlight-related codes I found
through google?)
Best,

Armel

tobiasko, 2022-02-17:
That is all pretty wired. I meanwhile installed rawrr 1.2.0 using `BiocManager` on a very similar system (Windows Server 2021):

```
> sessionInfo()
R version 4.1.0 (2021-05-18)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows Server 2012 R2 x64 (build 9600)

Matrix products: default

locale:
[1] LC_COLLATE=English_United States.1252  LC_CTYPE=English_United States.1252   
[3] LC_MONETARY=English_United States.1252 LC_NUMERIC=C                          
[5] LC_TIME=English_United States.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] rawrr_1.2.0         BiocManager_1.30.16

loaded via a namespace (and not attached):
 [1] magrittr_2.0.2    tidyselect_1.1.1  R6_2.5.1          rlang_1.0.1       fastmap_1.1.0     fansi_1.0.2      
 [7] dplyr_1.0.8       tools_4.1.0       DT_0.20           utf8_1.2.2        cli_3.2.0         DBI_1.1.2        
[13] htmltools_0.5.2   ellipsis_0.3.2    yaml_2.2.2        assertthat_0.2.1  digest_0.6.29     tibble_3.1.6     
[19] lifecycle_1.0.1   crayon_1.5.0      purrr_0.3.4       htmlwidgets_1.5.4 vctrs_0.3.8       glue_1.6.1       
[25] compiler_4.1.0    pillar_1.7.0      generics_0.1.2    pkgconfig_2.0.3  
```
What do you get when reading from the sample data that comes with the package?

```
> rawfile <- sampleFilePath()
> Idx <- readIndex(rawfile)
> str(Idx)
'data.frame':	573 obs. of  8 variables:
 $ scan          : int  1 2 3 4 5 6 7 8 9 10 ...
 $ scanType      : chr  "FTMS + c NSI Full ms [350.0000-1800.0000]" "FTMS + c NSI Full ms2 487.2567@hcd28.00 [140.0000-1015.0000]" "FTMS + c NSI Full ms2 644.8226@hcd28.00 [140.0000-1335.0000]" "FTMS + c NSI Full ms2 683.8279@hcd28.00 [140.0000-1415.0000]" ...
 $ rtinseconds   : num  0.097 0.35 0.419 0.489 0.558 0.627 0.696 0.766 0.835 0.904 ...
 $ precursorMass : num  1075 487 645 684 547 ...
 $ MSOrder       : chr  "Ms" "Ms2" "Ms2" "Ms2" ...
 $ charge        : int  0 2 2 2 2 2 2 2 2 2 ...
 $ masterScan    : int  NA NA NA NA NA NA NA NA NA NA ...
 $ dependencyType: logi  NA NA NA NA NA NA ...
```
My path to the sample data is:

```
> rawfile
[1] "X:/Rlibs/rawrr/extdata/sample.raw"
```

The rawrr assembly path (the path were the dll are placed) is

```
> rawrrAssemblyPath()
[1] "C:\\Users\\tobiasko.FGCZ-NET\\AppData\\Local/R/cache/R/rawrr/rawrrassembly"
```
as you can see, also here I am getting this strange mixture of `\\` and `/` in a single path. But that is obviously not a problem, since `readIndex` works.

The exe and the dlls should be at the assembly path location together with the EULA text:

```
> list.files(rawrrAssemblyPath())
[1] "eula.txt"                                           "rawrr.exe"                                         
[3] "ThermoFisher.CommonCore.Data.dll"                   "ThermoFisher.CommonCore.MassPrecisionEstimator.dll"
[5] "ThermoFisher.CommonCore.RawFileReader.dll"         
> 
```

Could you just execute the same commands and upload the returns?

Arthfael, 2022-02-17:
>  $ scanType      : chr  "FTMS + c NSI Full ms [350.0000-1800.0000]" "FTMS + c NSI Full ms2 ***@***.*** [140.0000-1015.0000]" "FTMS + c NSI Full ms2 ***@***.*** [140.0000-1335.0000]" "FTMS + c NSI Full ms2 ***@***.*** [140.0000-1415.0000]" ...

Dear Tobi,

I just figured it out. All of the paths and sessionInfo details matched as
closely as possible, so I decided to think again from the start. If
readChromatogram worked for me in TIC/Base Peak mode, but not XIC mode,
then maybe my usage was wrong for that mode. Indeed, it turns out I had
misread the help file, I thought one needs to provide one tolerance value
for each mass, not a single one for all:
masses <- c(524, 670)
tols <- rep(100, length(masses)) # in ppm
XICs <- readChromatogram(Raws[1], masses, tols) # this fails
XICs <- readChromatogram(Raws[1], masses, tols[1) # this works

My apologies for wasting your time. Maybe it would be a good idea to change
slightly the function so that if several tolerances are provided, it will
throw a more helpful error, as the error message which was printed now did
not actually relate to the cause of the error. Anyway, thank you for your
help!
All the best,

Armel

tobiasko, 2022-02-17:
Puh! Yip, we need to check more carefully if the parameters are valid (length = 1) and raise an error right away. That could be avoided very easily by a `stopifnot()` statement at the start of `readChromatogram`.

tobiasko, 2022-02-17:
@cpanse 

We need to check this here:

```
readChromatogram <- function(rawfile,
                             mass = NULL,
                             tol = 10,
                             filter = "ms",
                             type = 'xic'){
    
    .isAssemblyWorking()
    .checkRawFile(rawfile)
    
    stopifnot(type %in% c('xic', 'bpc', 'tic'))

```

closed 2022-02-17 by tobiasko