fgcz/rawrr #52: Peak charges for MS1 spectras

lucaraess, 2022-03-30:
Dear all,
thanks for this very useful package! 
Is there a way to extract the peak "charges" of MS1 spectras with rawrr::readSpectrum? It seems to only work for MS2 spectras at the moment.
Thanks a lot.

lucaraess, 2022-04-06:
Yes, this is Orbitrap data recorded in centroid mode. My code is as simple as that:
Idx_run1 <- readIndex("DIR")
rawrr::readSpectrum("DIR", scan = which(Idx_run1$MSOrder == "Ms"))

whereof I'm then trying to extract the charges, but as you pointed out is apparently not possible. Thanks for your help!

tobiasko, 2022-04-11:
ok. In this case let me end this discussion with a final comment: I have never seen centroided orbitrap data was missing charge state assignments in general. Sometimes the isotope patterns can not be fully resolved, so you will see no assignment for the low S/N patterns. Example code from the package:

```
> library(rawrr)
Package 'rawrr' version 1.2.0 using
RawFileReader reading tool. Copyright © 2016 by Thermo Fisher Scientific, Inc. All rights reserved.
> # Example 1
> (rawfile <- rawrr::sampleFilePath())
[1] "/Library/Frameworks/R.framework/Versions/4.1/Resources/library/rawrr/extdata/sample.raw"
> 
> S <- rawrr::readSpectrum(rawfile, scan = 1:9)
> 
> S[[1]]
Total Ion Current:	 78725448
Scan Low Mass:	 350
Scan High Mass:	 1800
Scan Start Time (Min):	 0
Scan Number:	 1
Base Peak Intensity:	 5979308
Base Peak Mass:	 445.1181
Scan Mode:	 FTMS + c NSI Full ms [350.0000-1800.0000]
======= Instrument data =====   :	NULL
Multiple Injection:	ii
Multi Inject Info:	IT=28;28 
AGC:	On
Micro Scan Count:	1
Scan Segment:	1
Scan Event:	1
Master Index:	0
Charge State:	0
Monoisotopic M/Z:	0.0000
Ion Injection Time (ms):	55.000
Max. Ion Time (ms):	55.00
FT Resolution:	120000
MS2 Isolation Width:	1450.00
MS2 Isolation Offset:	0.00
AGC Target:	3000000
HCD Energy:	
Analyzer Temperature:	27.06
=== Mass Calibration:	NULL
Conversion Parameter B:	211820589.2896
Conversion Parameter C:	-41163067.8349
Temperature Comp. (ppm):	-2.35
RF Comp. (ppm):	-0.09
Space Charge Comp. (ppm):	-2.30
Resolution Comp. (ppm):	0.57
Number of Lock Masses:	0
Lock Mass #1 (m/z):	0.0000
Lock Mass #2 (m/z):	0.0000
Lock Mass #3 (m/z):	0.0000
LM Search Window (ppm):	0.0
LM Search Window (mmu):	0.0
Number of LM Found:	0
Last Locking (sec):	0.0
LM m/z-Correction (ppm):	0.00
=== Ion Optics Settings:	NULL
S-Lens RF Level:	40.00
S-Lens Voltage (V):	21.00
Skimmer Voltage (V):	15.00
Inject Flatapole Offset (V):	5.00
Bent Flatapole DC (V):	2.00
MP2 and MP3 RF (V):	900.00
Gate Lens Voltage (V):	2.25
C-Trap RF (V):	2400.0
====  Diagnostic Data:	NULL
Dynamic RT Shift (min):	0.00
Intens Comp Factor:	1.4281
Res. Dep. Intens:	0.981
CTCD NumF:	0
CTCD Comp:	1.009
[... 45 lines cut ...]
  [1]  1  0  0  0  1  0  0  0  0  1  1  1  1  1  1 28  0 28  0  1 28 28  1 28  1 28  1  1  0  1  0  1
 [33]  1  1  1  1  0  1  0  1  1  0  1  1  1  1  0  1  1  0  0  1  1  1  1  1  0  0  1  1  1  0  0  1
 [65]  1  1  1  1  0  1  1  0  1  0  1  1  1  1  1  1  1  1  1  1  1  1  1  0  0  0  0  1  1  1  1  0
 [97]  1  1  0  1  1  0  1  1  1  1  0  0  1  0  0  1  0  0  0  1  0  1  0  0  0  1  1  0  1  1  0  0
[129]  1  1  1  1  1  1  1  1  0  0  1  1  1  1  1  1  0  0  0  1  1  1  0  0  0  1  0  1  1  1  1  1
[161]  1  1  1  1  1  1  0  1  1  1  0  1  1  1  1  0  0  1  1  1  1  1  1  0  0  1  0  1  1  1  0  1
[193]  1  0  0  0  1  1  0  0  0  0  0  0  0  0  0  0  0  1  1  1  1  1  1  1  0  0  1  1  1  1  1  1
[225]  1  1  1  0  0  0  1  1  1  1  0  0  1  1  0  1  1  1  0  0  1  1  1  0  1  1  0  1  1  1  0  1
[257]  1  1  0  0  0  1  0  1  0  0  0  1  1  0  0  0  0  0  0  1  0  1  0  0  0  1  1  0  0  1  1  0
[289]  0  1  0  1  0  1  1  0  1  1  0  0  1  1  1  0  0  0  0  0  0  1  1  1  0  1  0  1  1  0  1  0
[321]  1  0  1  0  1  1  1  1  1  0  0  0  0  1  1  1  1  1  1  0  1  1  1  0  1  0  0  1  0  0  1  1
[353]  1  1  1  1  1  0  1  1  1  1  1  0  1  1  1  1  1  1  1  0  0  1  1  1  2  2  0  0  1  1  1  1
[385]  0  1  1  0  0  0  1  1  0  0  1  1  1  0  1  1  1  0  1  1  1  1  1  1  1  1  0  0  1  1  1  1
[417]  0  0  1  1  0  1  1  1  1  1  1  1  1  1  1  1  1  1  1  0  1  1  1  1  1  0  0  0  0  0  0  0
[449]  1  0  1  0  0  1  3  0  0  1  0  1  0  1  0  1  1  1  0  0  0  1  1  1  1  0  1  1  0  0  1  1
[481]  1  0  1  1  1  1  1  0  1  0  1  1  1  0  1  1  0  0  1  1  0  1  1  1  1  1  0  1  1  1  1  1
[513]  0  1  1  1  0  1  1  1  1  1  1  1  0  1  1  1  0  0  1  1  0  0  0  1  0  1  0  1  0  1  0  1
[545]  1  1  1  1  1  1  1  1  1  1  1  0  1  1  0  1  1  1  1  1  1  1  1  1  1  0  1  1  1  1  1  1
[577]  1  1  1  1  1  1  1  1  1  1  1  0  1  0  1  1  1  0  0  1  1  0  1  1  1  1  1  1  1  1  1  1
[609]  1  1  1  1  1  1  1  1  1  1  1  0  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1
[641]  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  0  1  1  1  1  1  1  1  1  0
[673]  0  0  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1  1
[705]  1  1  1  1  1  1  1  1  1  1  1  1  0  0  0  0  0  0  1  1  1  1  1  1  1  0  0  0  0  0  0  0
[737]  0  0
```

closed 2022-04-11 by tobiasko