fgcz/rawrr #54: Error in Example: Length of "x" and "y" are not matching

DRiXD, 2022-07-21:
Hi everyone, 
to test this package I wanted to load the .raw file and follow the provided example code.
Somehow my R sends me an error message that the "x and "y" coordinates are not matching:
[image]

Now when I run I get:
```
> plot(S[[1]], centroid=TRUE)
Error in xy.coords(x, y, xlabel, ylabel, log) : 
  Length of 'x' and 'y' do not match
```

I have absolutly no Idea what I'm doing wrong and am super lost.
I would appreciate some help here!

Also I am new to R and working with bio-informatics data so if anyone could provide any help how to come up with the number in the scan vector (paper just mentions some database seach?) that would be awesome aswell.
Thanks in advance!

DRiXD, 2022-07-21:
Oh and if it's necessary for debug issues:
[image]
 And R-version 4.2.1

tobiasko, 2022-07-22:
That is most likely due to your OS settings regarding your decimal sep. character. The Thermo API expects `.`, not `,`. see for instance

https://answers.microsoft.com/en-us/msoffice/forum/all/changing-decimal-separator-from-comma-to-dot/953556fb-00d0-45ad-bd4f-6ca8afa5025a

for Windows.

tobiasko, 2023-07-19:
Did the above solve your problem @DRiXD?

closed 2023-08-29 by cpanse