fgcz/rawrr #59: Read_Spectrum - Sum Spectra

BerndMaier, 2023-03-14:
Hi there,
if i see correct, Read_Spectra returns a list of all the scans entered. 

Is there a way to get a Summed/Average_Spectra like Xcalibur or Freestyle do? 

When I merge the list from rawrr::readSpectrum i get some weird results as data is not XY even (e.g. 9/10 scans share the exact same m/z value und 1 is 0.01 Da off)

[image]
[image]

Not sure if Xcalibur / Freestyle just merges after reading the spectra or if this is already part of the ThermoRawFileReader.
But adding an "Average" method to readSpectrum would be nice and also would save space, as the list can get quite large when reading the scans for whole chromatographic peaks.

All the best, Bernd

BerndMaier, 2025-03-14:
Hi guys, just checking if you maybe got any idea around this.

All the best,
Bernd

cpanse, 2025-04-09:
If you can guide me to a C# code snippet; that would help.

tobiasko, 2025-04-11:
There is no example code in @jshofstahl RawFileReader .doc file, there is just a high level description of the respective functions.