fgcz/rawrr #60: unit should be minute

cpanse, 2023-03-15:
https://github.com/fgcz/rawrr/blob/b14db4f59b9ba2f641a1579ead489a0bc96c26c7/R/rawrr.R#L1286

the unit seconds is wrong. it should be minutes.
@tobiasko can you confirm?

C

cpanse, 2023-03-16:
@rolivella it should be fixed
https://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr_1.7.10.tar.gz

rolivella, 2023-03-16:
Thanks! I'll have a look at it ASAP and let you know.

cpanse, 2023-03-16:
@rolivella
```
#R
install.packages('https://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr_1.7.10.tar.gz', repos=NULL)
# install latest rawrr.exe assembly!
rawrr::installRawrrExe()
# sanity check
rawrr::sampleFilePath() |> rawrr::readIndex() |> head()
```

the availability on Bioconductor will take some days
http://bioconductor.org/packages/devel/bioc/html/rawrr.html

rolivella, 2023-03-17:
@cpanse I apparently get the same result `RT : 0.07637552 - 79.99182 s` (interval in seconds, when it should be minutes): 

```r
> summary(QC3[[1]])
XIC generated from 
190215_Q_QC03_01_04_25ng_6583a564-93dd-4500-a101-b2fe56496b25_QC03_2bf4293c4d1c8c891fab774cf973f7e9.raw 
consisting of 4570 data points.
   Filter : ms
   m/z : 566.83
   Tolerance : 10 ppm
   RT : 0.07637552 - 79.99182 s
> 
> 
> 
> sessionInfo()
R version 4.1.0 (2021-05-18)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 7 x64 (build 7601) Service Pack 1

Matrix products: default

locale:
[1] LC_COLLATE=Spanish_Spain.1252  LC_CTYPE=Spanish_Spain.1252    LC_MONETARY=Spanish_Spain.1252
[4] LC_NUMERIC=C                   LC_TIME=Spanish_Spain.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

loaded via a namespace (and not attached):
[1] rawrr_1.7.10   compiler_4.1.0 tools_4.1.0  
```

cpanse, 2023-03-17:
@rolivella have you really installed the latest rawrr package?

```
R> rawfile <- rawrr::sampleFilePath()
R> QC3 <- rawrr::readChromatogram(rawfile, mass = 566.8300, tol = 10, type = "xic", filter = "ms")
R> summary(QC3[[1]])
XIC generated from sample.raw consisting of 27 data points.
   Filter : ms
   m/z : 566.83
   Tolerance : 10 ppm
   RT : 0.001619751 - 0.7801059 min
R> 
```

rolivella, 2023-03-17:
@cpanse I followed your instructions and the `sessionInfo()` tells that: 

```
> sessionInfo()

loaded via a namespace (and not attached):
[1] rawrr_1.7.10   compiler_4.1.0 tools_4.1.0  
```

What else can I do to ensure that I'm using the right version?

rolivella, 2023-03-17:
When I run `> rawrr::installRawrrExe()` I got: 

```R
trying URL 'http://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr.1.1.12.exe'
Content type 'application/x-msdos-program' length 26624 bytes (26 KB)
downloaded 26 KB

MD5 6d5f6d19512eaba73e92d2bfb474e6f2 C:\Users\proteomics\AppData\Local/R/cache/R/rawrr/rawrrassembly/rawrr.exe
[1] 0
```

But it's the wrong version, right? I don't undrestand nothing...

rolivella, 2023-03-17:
So all the commands to update rawrr were: 

```R
> install.packages('https://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr_1.7.10.tar.gz', repos=NULL)
WARNING: Rtools is required to build R packages but is not currently installed. Please download and install the appropriate version of Rtools before proceeding:

https://cran.rstudio.com/bin/windows/Rtools/
Installing package into ‘C:/Users/proteomics/Documents/R/win-library/4.1’
(as ‘lib’ is unspecified)
trying URL 'https://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr_1.7.10.tar.gz'
Content type 'application/x-gzip' length 1672478 bytes (1.6 MB)
downloaded 1.6 MB

* installing *source* package 'rawrr' ...
** using staged installation
** R
** inst
** byte-compile and prepare package for lazy loading
** help
*** installing help indices
  converting help for package 'rawrr'
    finding HTML links ... done
    auc.rawrrChromatogram                   html  
    basePeak                                html  
    buildRawrrExe                           html  
    dependentScan                           html  
    dot-checkDllInMonoPath                  html  
    dot-thermofisherlsmsUrl                 html  
    faimsVoltageOn                          html  
    filter                                  html  
    installRawFileReaderDLLs                html  
    installRawrrExe                         html  
    is.rawrrChromatogram                    html  
    is.rawrrSpectrum                        html  
    is.rawrrSpectrumSet                     html  
    makeAccessor                            html  
    massRange                               html  
    masterScan                              html  
    new_rawrrSpectrum                       html  
    plot.rawrrChromatogram                  html  
    plot.rawrrChromatogramSet               html  
    plot.rawrrSpectrum                      html  
    print.rawrrSpectrum                     html  
    rawrrAssemblyPath                       html  
    rawrrSpectrum                           html  
    readChromatogram                        html  
    readFileHeader                          html  
    readIndex                               html  
    readSpectrum                            html  
    readTrailer                             html  
    sampleFilePath                          html  
    scanNumber                              html  
    summary.rawrrChromatogram               html  
    summary.rawrrSpectrum                   html  
    tic                                     html  
    validate_rawrrIndex                     html  
    validate_rawrrSpectrum                  html  
** building package indices
** installing vignettes
** testing if installed package can be loaded from temporary location
*** arch - i386
*** arch - x64
** testing if installed package can be loaded from final location
*** arch - i386
*** arch - x64
** testing if installed package keeps a record of temporary installation path
* DONE (rawrr)
```

```R
> rawrr::installRawrrExe()
trying URL 'http://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr.1.1.12.exe'
Content type 'application/x-msdos-program' length 26624 bytes (26 KB)
downloaded 26 KB

MD5 6d5f6d19512eaba73e92d2bfb474e6f2 C:\Users\proteomics\AppData\Local/R/cache/R/rawrr/rawrrassembly/rawrr.exe
[1] 0
```

```R
> rawrr::sampleFilePath() |> rawrr::readIndex() |> head()
  scan                                                     scanType rtinseconds precursorMass MSOrder charge
1    1                    FTMS + c NSI Full ms [350.0000-1800.0000]       0.097     1075.0000      Ms      0
2    2 FTMS + c NSI Full ms2 487.2567@hcd28.00 [140.0000-1015.0000]       0.350      487.2567     Ms2      2
3    3 FTMS + c NSI Full ms2 644.8226@hcd28.00 [140.0000-1335.0000]       0.419      644.8226     Ms2      2
4    4 FTMS + c NSI Full ms2 683.8279@hcd28.00 [140.0000-1415.0000]       0.489      683.8279     Ms2      2
5    5 FTMS + c NSI Full ms2 547.2980@hcd28.00 [140.0000-1135.0000]       0.558      547.2980     Ms2      2
6    6 FTMS + c NSI Full ms2 669.8381@hcd28.00 [140.0000-1385.0000]       0.627      669.8381     Ms2      2
  masterScan dependencyType
1         NA             NA
2         NA             NA
3         NA             NA
4         NA             NA
5         NA             NA
6         NA             NA
```

cpanse, 2023-03-17:
something is wrong. restart R. does that help?

cpanse, 2023-03-17:
that a minutes.

rolivella, 2023-03-17:
yes...and your output if you do: ` rawrr:::auc.rawrrChromatogram(QC3[[1]])`??

cpanse, 2023-03-17:
I hope 0.0 because all intensities are 0.0

```
R> QC3 <- rawrr::sampleFilePath() |> rawrr::readChromatogram(mass = 566.8300, tol = 10, type = "xic", filter = "ms")
R> rawrr:::auc.rawrrChromatogram(QC3[[1]])
[1] 0
R> QC3[[1]]$intensities
 [1] 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
R> 

```

cpanse, 2023-03-24:
@rolivella 
https://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr_1.7.15.tar.gz

rolivella, 2023-03-24:
Still wrong! :(

What I do: 

Check that I have the right rawrr version: 
```R
> sessionInfo() 
R version 4.1.0 (2021-05-18)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 7 x64 (build 7601) Service Pack 1

Matrix products: default

locale:
[1] LC_COLLATE=Spanish_Spain.1252  LC_CTYPE=Spanish_Spain.1252    LC_MONETARY=Spanish_Spain.1252
[4] LC_NUMERIC=C                   LC_TIME=Spanish_Spain.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

loaded via a namespace (and not attached):
[1] rawrr_1.7.15   compiler_4.1.0 tools_4.1.0  
```

And do the calculations: 

```R 
> QC3 <- rawrr::readChromatogram(rawfile, mass = 566.8300, tol = 10, type = "xic", filter = "ms")
> summary(QC3[[1]])
XIC generated from 
190215_Q_QC03_01_04_25ng_6583a564-93dd-4500-a101-b2fe56496b25_QC03_2bf4293c4d1c8c891fab774cf973f7e9.raw 
consisting of 4570 data points.
   Filter : ms
   m/z : 566.83
   Tolerance : 10 ppm
   RT : 0.07637552 - 79.99182 min
> 
> 
> 
> 
> rawrr:::auc.rawrrChromatogram(QC3[[1]])
[1] 42614408
```

cpanse, 2023-03-24:
@rolivella can you send me the output of `rawrr:::auc.rawrrChromatogram`

rolivella, 2023-03-24:
```R
> rawrr:::auc.rawrrChromatogram
function (x) 
{
    times <- x$times
    intensities <- x$intensities
    sum(diff(times) * (head(intensities, -1) + tail(intensities, 
        -1)))/2
}
<bytecode: 0x0000000011e00a98>
<environment: namespace:rawrr>

```

cpanse, 2023-03-24:
@rolivella 
R> install.packages('https://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr_1.7.15.tar.gz')
leite 'repos = NULL' aus 'pkgs' ab
versuche URL 'https://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr_1.7.15.tar.gz'
Content type 'application/x-gzip' length 2275310 bytes (2.2 MB)
==================================================
downloaded 2.2 MB

* installing *source* package 'rawrr' ...
** using staged installation
** R
** inst
** byte-compile and prepare package for lazy loading
** help
*** installing help indices
** building package indices
** installing vignettes
** testing if installed package can be loaded from temporary location
** testing if installed package can be loaded from final location
** testing if installed package keeps a record of temporary installation path
* DONE (rawrr)
R> rawrr:::auc.rawrrChromatogram
function (x) 
{
    times <- 60 * x$times
    intensities <- x$intensities
    sum(diff(times) * (head(intensities, -1) + tail(intensities, 
        -1)))/2
}
<bytecode: 0x10f03c6e8>
<environment: namespace:rawrr>
R>

rolivella, 2023-03-24:
This is very strange because if I check the rawrr version I'm using it says 

```R
loaded via a namespace (and not attached):
[1] **rawrr_1.7.15**   compiler_4.1.0 tools_4.1.0  
```

But the `rawrr:::auc.rawrrChromatogram` is still on the old version without `times <- 60 * x$times`

I don't know what to do...maybe I'll wait until it's being published to Bioconductor.

cpanse, 2023-03-24:
@rolivella  that version contains just a version inc to 1.7.16

```
install.packages('https://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr_1.7.16.tar.gz', repos=NULL)
rawrr:::auc.rawrrChromatogram
```

cpanse, 2023-03-24:
C