fgcz/rawrr #61: validate_rawrrSpectrum 'StartTime'

mblanzillotti, 2023-05-15:
Hi! Since updating to version 1.8.0 (from 1.4.0), I am no longer able to read my own Thermo raw files (recorded by direct infusion on a QE-HFX platform) and cannot run the example spectrum read-ins as part of the 'readSpectrum' documentation.

The error:

"Error in if (values$StartTime < 0) {: argument is of length zero" 

appears when trying to run 'readSpectrum,' and commenting out:

"Error in if (values$StartTime < 0) {stop(...)}"

in 'validate_rawrrSpectrum' appears to fix this issue.

cpanse, 2023-07-19:
@mblanzillotti thanks for the issue. Can you please provide me access to the raw file?

mblanzillotti, 2023-07-25:
Hi! attached is a rawfile from a published data set that I had previously
used rawrr to manipulate in R, it is a single, averaged scan of tandem MS
of carbonic anhydrase acquired on a Lumos Orbitrap platform.

Also, I am currently running R 4.3.0 and rawrr version 1.8.0. When trying
to read this file into my environment, I get the following error:

"Error in if (values$StartTime < 0) { : argument is of length zero"

However if I manipulate the validate_rawrrSpectrum function to either
ignore the "StartTime" parameter or replace it with the "rtinseconds"
parameter used in older versions of rawrr, the spectrum will read in
appropriately.

Michael

cpanse, 2023-08-28:
@mblanzillotti  can you run  `rawrr::installRawrrExe()` to also update the mono assembly!? I think that will solve the issue.

closed 2023-10-25 by mblanzillotti