fgcz/rawrr #62: "Error: line 1 did not have 9 elements" for readIndex() and readChromatogram() + "Error : No scan vector is provided"for readSpectrum

candidaorelmex, 2023-08-25:
Hi,

I tried to use your package for the first time and ran into some error. Here the code with the errors (placed in quotes below the code that I ran:

`rawf = file.path(path.package(package = 'rawrr'), 'extdata', 'sample.raw')

readIndex(rawf)
"Error in scan(file = file, what = what, sep = sep, quote = quote, dec = dec,  : 
  line 1 did not have 9 elements"

readSpectrum(rawf)
"Error in readSpectrum(rawf) : No scan vector is provided."

readChromatogram(rawf)
"Error in readChromatogram(rawf) : No mass vector is provided."`

Thanks a lot for your help, I cant wait to inspect my raw files with a rawrr!

cpanse, 2023-08-26:
If you use Windows, please check #54 and https://github.com/fgcz/rawDiag/issues/33
Otherwise, please post the output of `sessionInfo()`.
C

cpanse, 2023-08-26:
`rawrr::readChromatogram` requires a raw file and a vector of masses.

candidaorelmex, 2023-08-28:
the windows decimal operator wasnt the issue, I have it set as described in #33/#54.

my `sessionInfo()` output is:

loaded via a namespace (and not attached):
  [1] rgl_1.2.1              ps_1.7.5               class_7.3-20           crayon_1.5.2           nlme_3.1-163           backports_1.4.1        ellipse_0.5.0         
  [8] GOSemSim_2.22.0        rlang_1.1.1            SparseM_1.81           filelock_1.0.2         rjson_0.2.21           bit64_4.0.5            glue_1.6.2            
 [15] processx_3.8.2         parallel_4.2.2         DOSE_3.22.1            tidyselect_1.2.0       XML_3.99-0.14          xtable_1.8-4           MatrixModels_0.5-2    
 [22] magrittr_2.0.3         evaluate_0.21          cli_3.6.1              zlibbioc_1.44.0        rstudioapi_0.15.0      miniUI_0.1.1.1         rpart_4.1.19          
 [29] fastmatch_1.1-4        polynom_1.4-1          treeio_1.20.2          xfun_0.40              askpass_1.1            parameters_0.21.1      cluster_2.1.4         
 [36] caTools_1.18.2         tidygraph_1.2.3        KEGGREST_1.38.0        quantreg_5.97          ape_5.7-1              listenv_0.9.0          reshape_0.8.9         
 [43] png_0.1-8              future_1.33.0          ipred_0.9-14           zeallot_0.1.0          withr_2.5.0            bitops_1.0-7           cellranger_1.1.0      
 [50] e1071_1.7-13           coda_0.19-4            pillar_1.9.0           GlobalOptions_0.1.2    cachem_1.0.8           multcomp_1.4-25        GetoptLong_1.0.5      
 [57] paletteer_1.5.0        vctrs_0.6.3            ellipsis_0.3.2         generics_0.1.3         lava_1.7.2.1           tools_4.2.2            foreign_0.8-83        
 [64] munsell_0.5.0          tweenr_2.0.2           fgsea_1.22.0           emmeans_1.8.8          proxy_0.4-27           fastmap_1.1.1          compiler_4.2.2        
 [71] abind_1.4-5            httpuv_1.6.11          GenomeInfoDbData_1.2.9 prodlim_2023.03.31     lattice_0.20-45        visNetwork_2.1.2       utf8_1.2.3            
 [78] later_1.3.1            BiocFileCache_2.6.1    Quandl_2.11.0          jsonlite_1.8.7         ada_2.0-5              tidytree_0.4.5         carData_3.0-5         
 [85] estimability_1.4.1     lazyeval_0.2.2         promises_1.2.1         car_3.1-2              R.utils_2.12.2         checkmate_2.2.0        rmarkdown_2.24        
 [92] sandwich_3.0-2         textshaping_0.3.6      downloader_0.4         igraph_1.5.1           survival_3.4-0         systemfonts_1.0.4      htmltools_0.5.6       
 [99] memoise_2.0.1          graphlayouts_1.0.0     quadprog_1.5-8         viridisLite_0.4.2      assertthat_0.2.1       digest_0.6.33          mime_0.12             
[106] rappdirs_0.3.3         bayestestR_0.13.1      yulab.utils_0.0.8      future.apply_1.11.0    blob_1.2.4             R.oo_1.25.0            ragg_1.2.5            
[113] Formula_1.2-5          splines_4.2.2          rematch2_2.1.2         RCurl_1.98-1.12        broom_1.0.5            hms_1.1.3              colorspace_2.1-0      
[120] base64enc_0.1-3        shape_1.4.6            aplot_0.2.0            websocket_1.4.1        nnet_7.3-18            Rcpp_1.0.11            mvtnorm_1.2-2         
[127] enrichplot_1.16.2      fansi_1.0.4            tzdb_0.4.0             parallelly_1.36.0      svGUI_1.0.1            R6_2.5.1               lifecycle_1.0.3       
[134] statsExpressions_1.5.1 zip_2.3.0              datawizard_0.8.0       curl_5.0.2             ggsignif_0.6.4         snakecase_0.11.0       DO.db_2.9             
[141] Matrix_1.6-1           qvalue_2.28.0          TH.data_1.1-2          RColorBrewer_1.1-3     htmlwidgets_1.6.2      polyclip_1.10-4        shadowtext_0.1.2      
[148] chromote_0.1.2         timechange_0.2.0       gridGraphics_0.5-1     globals_0.16.2         openssl_2.1.0          insight_0.19.3         htmlTable_2.4.1       
[155] patchwork_1.1.3        codetools_0.2-18       GO.db_3.15.0           gtools_3.9.4           prettyunits_1.1.1      dbplyr_2.3.3           gridBase_0.4-7        
[162] R.methodsS3_1.8.2      correlation_0.8.4      gtable_0.3.4           ggfun_0.1.2            httr_1.4.7             KernSmooth_2.23-20     stringi_1.7.12        
[169] progress_1.2.2         reshape2_1.4.4         farver_2.1.1           uuid_1.1-1             viridis_0.6.4          ggtree_3.4.4           xml2_1.3.5            
[176] ade4_1.7-22            ggplotify_0.1.2        scatterpie_0.2.1       ggraph_2.1.0           janitor_2.2.0          pkgconfig_2.0.3        rstatix_0.7.2         
[183] evd_2.3-6.1            knitr_1.43

cpanse, 2023-08-28:
Thanks; interesting; I need the R, rawrr, and Windows OS versions.
Also, have you tried the `rawrr::readChromatogram` command with the correct arguments?
Also, when you changed the sep. setting, have you rebooted the OS  and restarted R?
C

candidaorelmex, 2023-08-28:
Sure:
R -> 4.2.2
rawrr -> 1.4.0
WindowsOS -> Windows 10, Version 22H2 (OS build 19045.3086)

readChromatogram works!

regarding the sep. settings: yes I have, it didn't help.

Thank you for your assistance!

cpanse, 2023-08-28:
OK; I see the problem. your rawrr version is out of date. current version is rawrr' version 1.8.1.

please update.

https://bioconductor.org/packages/rawrr/

```
if (!require("BiocManager", quietly = TRUE))
    install.packages("BiocManager")

BiocManager::install("rawrr")
```

the mono assembly is mismatching with the Rcode.

candidaorelmex, 2023-08-29:
I updated rawrr to 1.8.1, yet the same error messages persist.

cpanse, 2023-08-29:
@candidaorelmex and run `rawrr::installRawrrExe()` .
C

candidaorelmex, 2023-08-29:
It works! thanks a lot for your help:)

closed 2023-08-29 by cpanse