fgcz/rawrr #63: Problem executing readChromatogram inside Singularity container

toniher, 2023-08-29:
Hello,

I am trying to execute ``rawrr::readChromatogram`` inside a Singularity container (r-ver:4.1), concretely this command:
```
QCfile <- rawrr::readChromatogram( rawfile = args[1], mass = 566.83, tol = 10, type = "xic", filter = "ms", tmpdir="./tmp")
```
and I am getting this error, and after debugging a bit:

```
Error in .rawrrSystem2Source(rawfile, input = mass, rawrrArgs = procArgs,  :
  Rcode file to parse does not exist. '/home/toniher/.cache/R/rawrr/rawrrassembly/rawrr.exe' failed for an unknown reason.
Please check the debug files:
        ./tmp/file231265115b740c.stderr
        ./tmp/file231265636edd0f.stdout
and the System Requirements
Calls: <Anonymous> -> .rawrrSystem2Source
Execution halted
```
checking stderr I see the following error:

``
failed to catch configfile and itol
``

I can confirm trying to run straight the mono executable:
```
/usr/bin/mono '/home/toniher/.cache/R/rawrr/rawrrassembly/rawrr.exe' '2022ZZ999_bovine.raw' xic 10.000000 'ms' './tmp/file2312652e2acf2f.txt' './tmp/file2312651aaea3c0.R'
```

Do you have any idea what might be failing?

cpanse, 2023-08-30:
Good morning; 

two things:
1. if possible use the latest R4.3.x and rawrr 1.8.1 https://github.com/fgcz/rawrr/ and followed by `rawrr::installRawrrExe()` 
2. it could be that the *libraries have an issue with relative paths. can you please check if you set the absolute path for temp? what is the output of file231265636edd0f.stdout?
3. lets have a debug zoom session , e.g., in the afternoon?!

thanks,
C

toniher, 2023-08-30:
Hi @cpanse I tried with your last package and R 4.3 version and I don't see more error messages.
Exe used is: https://github.com/fgcz/rawrr/releases/download/1.7.10/rawrr.1.7.10.exe

In tmpdir I generate these three files:
 file24146e21249eda.stderr   file24146e5f85e384.stdout   file24146e6d12a90f.txt

In txt file ```566.83``` is printed.

I cannot share screen in Zoom today, but I can arrange for tomorrow afternoon if fine with you, let's say, around 16, if fine with you.

cpanse, 2023-08-30:
> Hi @cpanse I tried with your last package and R 4.3 version and I don't see more error messages. Exe used is: https://github.com/fgcz/rawrr/releases/download/1.7.10/rawrr.1.7.10.exe
>
> In tmpdir I [...]

1600 (Berlin time zone) is fine.
C

toniher, 2023-08-31:
We were missing in our setup (Ubuntu 22.04) ``libmono-system-data4.0-cil`` as pointed in in https://github.com/fgcz/rawrr/blob/devel/INSTALL

closed 2023-08-31 by toniher