fgcz/rawrr #65: Error in if (rvs != "No RAW file specified!") { :    the condition has length > 1

xianxiongma, 2023-09-03:
S <- rawrr::sampleFilePath() |> rawrr::readSpectrum(scan = 1:9)
Error in if (rvs != "No RAW file specified!") { :    the condition has length > 1
Can you tell me how to deal with it

cpanse, 2023-09-04:
what is the output of `rawrr::sampleFilePath() ` ?

xianxiongma, 2023-09-04:
> what is the output of `rawrr::sampleFilePath() ` ?

> S <- rawrr::sampleFilePath()
> S
[1] "D:/ProgramFiles/R/R-4.2.0/library/rawrr/extdata/sample.raw"

xianxiongma, 2023-09-04:
> can you please provide 1. OS 2. rawrr version and 3. the output of `sessionInfo()` C

> sessionInfo()
R version 4.2.0 (2022-04-22 ucrt)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 18363)

Matrix products: default

locale:
[1] LC_COLLATE=Chinese (Simplified)_China.utf8  LC_CTYPE=Chinese (Simplified)_China.utf8   
[3] LC_MONETARY=Chinese (Simplified)_China.utf8 LC_NUMERIC=C                               
[5] LC_TIME=Chinese (Simplified)_China.utf8    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] rawrr_1.8.1

loaded via a namespace (and not attached):
[1] compiler_4.2.0 tools_4.2.0

cpanse, 2023-09-05:
can you please rerun ` rawrr::installRawrrExe()` and send me the output.

xianxiongma, 2023-09-07:
> can you please rerun ` rawrr::installRawrrExe()` and send me the output.

> rawrr::installRawrrExe()
trying URL 'https://github.com/fgcz/rawrr/releases/download/1.7.10/rawrr.1.7.10.exe'
Content type 'application/octet-stream' length 28672 bytes (28 KB)
downloaded 28 KB

MD5 5c3b6d0135ffe550bc04615cfc77aaba C:\Users\Administrator\AppData\Local/R/cache/R/rawrr/rawrrassembly/rawrr.exe
[1] 0

cpanse, 2023-09-07:
so far this looks good.  can you please try some other functions?

```{r}
f <- rawrr::sampleFilePath()
rawrr::readFileHeader(f)
rawrr::readIndex(f) |> head()
rawrr::readSpectrum(f, scan = 1:9)
```

if it still does not work, I would propose a quick zoom. 

C

xianxiongma, 2023-09-07:
Still reporting the same error, I still can't run the example

> rawrr::readFileHeader(f)
Error in if (rvs != "No RAW file specified!") { : 
  the condition has length > 1

> rawrr::readIndex(f) |> head()
Error in if (rvs != "No RAW file specified!") { : 
  the condition has length > 1

xianxiongma, 2023-09-07:
> so far this looks good. can you please try some other functions?
> 
> ```
> f <- rawrr::sampleFilePath()
> rawrr::readFileHeader(f)
> rawrr::readIndex(f) |> head()
> rawrr::readSpectrum(f, scan = 1:9)
> ```
> 
> if it still does not work, I would propose a quick zoom.
> 
> C

I ran in both computers, reported the same error, can you help me solve it?

cpanse, 2023-09-07:
@xianxiongma quick zoom  in 15minutes? 0710 UTC
https://ethz.zoom.us/j/63334146797

xianxiongma, 2023-09-07:
> @xianxiongma quick zoom in 15minutes? 0710 UTC https://ethz.zoom.us/j/63334146797

Quick zoom? I don't understand, what should I do? I can't open the above URL

xianxiongma, 2023-09-07:
> > @xianxiongma quick zoom in 15minutes? 0710 UTC https://ethz.zoom.us/j/63334146797
> 
> Quick zoom? I don't understand, what should I do? I can't open the above URL

I just want to how to edit obitrap raw file

cpanse, 2023-09-07:
> > @xianxiongma quick zoom in 15minutes? 0710 UTC https://ethz.zoom.us/j/63334146797
> 
> Quick zoom? I don't understand, what should I do? I can't open the above URL
I wonder if we can have a video session (using zoom) to fix the issue.  would that work for you?

cpanse, 2023-09-07:
> > > @xianxiongma quick zoom in 15minutes? 0710 UTC https://ethz.zoom.us/j/63334146797
> > 
> > 
> > Quick zoom? I don't understand, what should I do? I can't open the above URL
> 
> I just want to how to edit obitrap raw file

of note, rawrr can only read Orbitrap data.

xianxiongma, 2023-09-07:
> > > @xianxiongma quick zoom in 15minutes? 0710 UTC https://ethz.zoom.us/j/63334146797
> > 
> > 
> > Quick zoom? I don't understand, what should I do? I can't open the above URL
> > I wonder if we can have a video session (using zoom) to fix the issue.  would that work for you?

No problem!