fgcz/rawrr #67: different total number of Spectra in msconvert, compomics/ThermoRawFileParser and thermofisherlsms/RawFileReader

asepsiswu, 2023-10-31:
MsBackendRawFileReader generated spectra lose the end spectrum(scanindex=7244) in msconvert and ThermoRawFileParser version.
I tried two files and got the same results MsBackendRawFileReader lost the final  scanindex

``` 
#  docker run   -it --rm  -v $PWD:/data  chambm/pwiz-skyline-i-agree-to-the-vendor-licenses:3.0.22335-b595b19  wine msconvert  --filter "peakPicking true 1-" negms2-1.raw
thermomzml <- '/mnt/ms/msftp/test/negms2-1.mzML'
#  mono ThermoRawFileParser1.4.2/ThermoRawFileParser.exe -i negms2-1.raw -b negms2-1.therorawfile.mzML
thermomzml2 <- '/mnt/ms/msftp/test/negms2-1.therorawfile.mzML'
thermoraw <- '/mnt/ms/msftp/test/negms2-1.raw'
xe <-MsExperiment::readMsExperiment(thermomzml)
xe2 <-MsExperiment::readMsExperiment(thermomzml2)
be <- backendInitialize(MsBackendRawFileReader(), files = thermoraw)
xe
xe2
Spectra(be)
``` 

``` 
> xe
Object of class MsExperiment 
 Spectra: MS1 (1261) MS2 (5983) 
 Experiment data: 1 sample(s)
 Sample data links:
  - spectra: 1 sample(s) to 7244 element(s).
> xe2
Object of class MsExperiment 
 Spectra: MS1 (1261) MS2 (5983) 
 Experiment data: 1 sample(s)
 Sample data links:
  - spectra: 1 sample(s) to 7244 element(s).
> Spectra(be)
MSn data (Spectra) with 7243 spectra in a MsBackendRawFileReader backend:
       msLevel     rtime scanIndex
     <integer> <numeric> <integer>
1            1        NA         1
2            2        NA         2
3            2        NA         3
4            2        NA         4
5            2        NA         5
...        ...       ...       ...
7239         2        NA      7239
7240         2        NA      7240
7241         1        NA      7241
7242         2        NA      7242
7243         2        NA      7243
 ... 31 more variables/columns.

file(s):
negms2-1.raw
```

asepsiswu, 2023-10-31:
unequal spectra
```
> Idx <- rawrr::sampleFilePath() |> rawrr::readIndex()
> dim(Idx)
[1] 573   9
> spc <- rawrr::sampleFilePath() |> rawrr::readSpectrum(scan = 1:600)
> length(spc)
[1] 574
```

cpanse, 2023-11-01:
@asepsiswu thanks!!!

using https://bioconductor.org/packages/rawrr/
version >1.10

I fixed the assembly. executing 
```
R> rawrr::installRawrrExe("https://github.com/fgcz/rawrr/releases/download/1.9.2/rawrr.1.9.2.exe")
versuche URL 'https://github.com/fgcz/rawrr/releases/download/1.9.2/rawrr.1.9.2.exe'
Content type 'application/octet-stream' length 28672 bytes (28 KB)
==================================================
downloaded 28 KB

MD5 5757c4f83c3bb18d143dc29421cf3c70 /Users/cp/Library/Caches/org.R-project.R/R/rawrr/rawrrassembly/rawrr.exe
[1] 0
```
should solve the issue.
```
R> rawrr::sampleFilePath() |> rawrr::readIndex() |> nrow()
[1] 574
R> rawrr::sampleFilePath() |> rawrr::readSpectrum(scan = 1:600) |> length()
[1] 574
R> 
```

As a token of appreciation, I would like to send you a rawrr sticker. If you're interested, please provide me with your mailing address (via email).

C

commit a67bd3fb (2023-11-08): fix #67 and bump to 1.10.1
closed 2023-11-01 by cpanse