fgcz/rawrr #68: Request for auc.rawrrChromatogram

rolivella, 2023-11-02:
Hello! 

I would like to suggest a little bit more informative output for `rawrr:::auc.rawrrChromatogram` fucntion. For instance with observed RT and mz, something like: 

```
XIC generated from 20230525_QCN1_001_03.raw consisting of 1454 data points.
   Filter : ms
   m/z : 259.0925
   m/z obs: 259.0889
   RT obs: 12.4 min
   Tolerance : 10 ppm
   RT : 0.001045075 - 41.0016 min
[1] 159020992
```

By "obs" I mean the RT and mz where the `rawrr:::auc.rawrrChromatogram` fucntion found the peak for integrating the area. 

This will be very useful for us and in general for any rawrr user. 

Thanks!

tobiasko, 2023-11-06:
To my knowledge `rawrr:::auc.rawrrChromatogram` returns the [integral](https://en.wikipedia.org/wiki/Integral) with f(x) being the chromatogram. It is not a peak picker. Accordingly, the return value of the function is a real value.

rolivella, 2023-11-06:
Maybe you can compute the local maximum as a measure of the observed RT and mz