fgcz/rawrr #69: profile mode in readSpectrum

asepsiswu, 2023-11-06:
I compared three tools: **msconvert**, **ThermoRawFileParser**, and **rawrr**, with the goal of skip the RAW to mzML conversion step. During this comparison, I noticed discrepancies in the number of peaks in the same MS1 scan and **msconvert** extracts much more peaks, primarily coused by "profile mode." 

Both ThermoRawFileParser and rawrr employ the same .NET (C#) ThermoFisher ThermoRawFileReader library. However, ThermoRawFileParser offers a '--noPeakPicking' parameter, which, when utilized, results in the same outcomes as msconvert. Reference link: https://github.com/compomics/ThermoRawFileParser/issues/34#issuecomment-493472923

I'm curious if rawrr supports the option for profile mode, similar to what ThermoRawFileParser provides.

tobiasko, 2023-11-06:
Not sure if I get your question, but `rawrr` does not provide access to Thermo data reduction routines like centroiding of profile data. It just provides access to the data as it was written to disk by the instrument control software. Meaning: If a scan was written in (reduced) profile mode, you can get profile data (a quasi continuous representation of the spectrum). If a scan was recorded in centroided mode, we can only provide access to centroided representation  (discrete peaks).

asepsiswu, 2023-11-06:
I apologize for not including  code and demo data.
please download the test raw file https://drive.google.com/open?id=1AWSEQV7TtnFtPHwzld5U1KWtz1aM6kUm from https://github.com/compomics/ThermoRawFileParser/issues/34#issuecomment-493472923.

the code below show readSpectrum has  different peak count  in scan 1. 
But, interestingly, Xcalibur show the peak count of 2842. But msconvert and ThermoRawFileParser has 2878.

Perhaps it is not caused by the "profile mode."

``` bash
ls
rawrrassembly  test_thermorawfileconverter.raw  ThermoRawFileParser

```

In rawrr: 2842
```
R -e ' "test_thermorawfileconverter.raw" |> rawrr::readSpectrum(scan = 1) -> spc; spc[[1]]$HasCentroidStream '

>  "test_thermorawfileconverter.raw" |> rawrr::readSpectrum(scan = 1) -> spc; spc[[1]]$HasCentroidStream
[1] "True, Length=2842"

```
msconvert: 2878
```
docker run -it --rm  -v `pwd`:/data chambm/pwiz-skyline-i-agree-to-the-vendor-licenses:3.0.22335-b595b19  wine msconvert  /data/test_thermorawfileconverter.raw --filter "peakPicking true 1-"

grep 'index="0"' test_thermorawfileconverter.mzML

        <spectrum index="0" id="controllerType=0 controllerNumber=1 scan=1" defaultArrayLength="2878">
        <chromatogram index="0" id="TIC" defaultArrayLength="58">
```
In ThermoRawFileParser :https://github.com/compomics/ThermoRawFileParser/releases/tag/v1.4.3

default: 2878
```
mono ThermoRawFileParser/ThermoRawFileParser.exe -i test_thermorawfileconverter.raw -b testdefault.mzML ; grep 'index="0"' testdefault.mzML

        <spectrum id="controllerType=0 controllerNumber=1 scan=1" index="0" defaultArrayLength="2878">
        <chromatogram id="BasePeak_0" index="0" defaultArrayLength="58">
```
with `--noPeakPicking`: 39032
```
 mono ThermoRawFileParser/ThermoRawFileParser.exe -i test_thermorawfileconverter.raw --noPeakPicking  -b testnopp.mzML ; grep 'index="0"' testnopp.mzML

 <spectrum id="controllerType=0 controllerNumber=1 scan=1" index="0" defaultArrayLength="39032">
  <chromatogram id="BasePeak_0" index="0" defaultArrayLength="58">

```

tobiasko, 2023-11-06:
Scan 1 is a profile mode scan:

```
> rawrr::readSpectrum(rawfile = "/Users/tobiasko//Downloads/test_thermorawfileconverter.raw", scan = 1)
[[1]]
Total Ion Current:	 974843968
Scan Low Mass:	 150
Scan High Mass:	 1600
Scan Start Time (Min):	 0.01
Scan Number:	 1
Base Peak Intensity:	 20530606
Base Peak Mass:	 874.7857
Scan Mode:	 FTMS + p NSI Full ms [150.0000-1600.0000]
======= Instrument data =====   :	NULL
AGC:	On
Micro Scan Count:	10
Charge State:	0
Monoisotopic M/Z:	0.0000
Ion Injection Time (ms):	0.292
MS2 Isolation Width:	1450.000
HCD Energy:	N/A
=== Mass Calibration:	NULL
Conversion Parameter B:	211778191.681400
Conversion Parameter C:	-30605686.250352
Temperature Comp. (ppm):	-4.85
RF Comp. (ppm):	0.00
Space Charge Comp. (ppm):	-0.15
Resolution Comp. (ppm):	-0.77
Number of LM Found:	1
=== Ion Optics Settings:	NULL
====  Diagnostic Data:	NULL
RawOvFtT:	468449.5
Injection t0:	-0.043

attr(,"class")
[1] "rawrrSpectrumSet"
> 
```
represented by 39032 intensity values and their positions in m/z:

```
> s <- rawrr::readSpectrum(rawfile = "/Users/tobiasko//Downloads/test_thermorawfileconverter.raw", scan = 1)
> length(s[[1]]$mZ)
[1] 39032
> length(s[[1]]$intensity)
[1] 39032
```

tobiasko, 2023-11-06:
What exactly is stored in the centroided stream in these cases is something I never quite understood:

```
> summary(s[[1]]$centroid.mZ)
   Min. 1st Qu.  Median    Mean 3rd Qu.    Max. 
  150.1   610.7   785.6   777.3   892.6  1599.3 
> summary(s[[1]]$centroid.intensity)
    Min.  1st Qu.   Median     Mean  3rd Qu.     Max. 
   19428    45542    66473   340286   178217 20530606 
```

but I remember that @jshofstahl (Thermo) once tried to explain us how these values a derived and in which cases. Maybe @cpanse remembers more details?

asepsiswu, 2023-11-07:
I'm concerned about the inconsistency in centroided peak count when using different software tools such as msconvert, ThermoRawFileParser, and rawrr.

rawrr can offer additional information for each centroided peak, including details like noise, resolutions, charges, and baselines.

> What exactly is stored in the centroided stream in these cases is something I never quite understood:
> 
> ```
> > summary(s[[1]]$centroid.mZ)
>    Min. 1st Qu.  Median    Mean 3rd Qu.    Max. 
>   150.1   610.7   785.6   777.3   892.6  1599.3 
> > summary(s[[1]]$centroid.intensity)
>     Min.  1st Qu.   Median     Mean  3rd Qu.     Max. 
>    19428    45542    66473   340286   178217 20530606 
> ```
> 
> but I remember that @jshofstahl (Thermo) once tried to explain us how these values are derived and in which cases. Maybe @cpanse remembers more details?

cpanse, 2023-11-07:
rawrr::readSpectrum is based on [L405](https://github.com/fgcz/rawrr/blob/926dbbcbb5863017a2db5c60f3a600e68fe32e93/inst/rawrrassembly/rawrr.cs#L405) and there are two methods:

```
var centroidStream = rawFile.GetCentroidStream(scanNumber, false);
centroidStream.Masses
```

and

```
var scan = Scan.FromFile(rawFile, scanNumber);
scan.CentroidScan.Masses.ToArray() 
```

I also have no idea why there is a difference in the number of peaks.

tobiasko, 2023-11-07:
I would not call this an "inconsistency". There is different ways of centroiding profile data, I guess they all have their pros and cons, theoretical foundations. My advice is: Use the centroiding recommended by the people that know the mass analyser best - the instrument vendor. This is why MSconvert recommends using the vendor dlls for peak picking:

```
peakPicking [<PickerType> [snr=<minimum signal-to-noise ratio>] [peakSpace=<minimum peak spacing>] [msLevel=<ms_levels>]]
This filter performs centroiding on spectra with the selected <ms_levels>, expressed as an int_set. The value for <PickerType> must be "cwt" or "vendor": when <PickerType> = "vendor", vendor (Windows DLL) code is used if available.
IMPORTANT NOTE: since this filter operates on the raw data through the vendor DLLs, IT MUST BE THE FIRST FILTER IN ANY LIST OF FILTERS when "vendor" is used.
The other option for PickerType is "cwt", which uses ProteoWizard's wavelet-based algorithm for performing peak-picking with a wavelet-space signal-to-noise ratio of <signal-to-noise ratio>.
Defaults:
<PickerType> is a low-quality (non-vendor) local maxima algorithm
<signal-to-noise ratio> = 1.0
<minimum peak spacing> = 0.1
<ms_levels> = 1-

```

https://proteowizard.sourceforge.io/tools/msconvert.html

To make this abs. clear: `rawrr` does **NOT** perform any centroiding/peak picking and has no access to vendor libs used for this. Whatever is stored in the files can usually be retrieved, but it needs to be stored there already.

asepsiswu, 2023-11-08:
The difference between ThermoRawFileParser and rawrr (2878 vs 2842) is confusing. ThermoRawFileParser utilizes the same GetCentroidStream function. https://github.com/compomics/ThermoRawFileParser/blob/42131e5f05e1e3684173e5b9a6dccc306fc5c353/Writer/ParquetSpectrumWriter.cs#L79

> rawrr::readSpectrum is based on [L405](https://github.com/fgcz/rawrr/blob/926dbbcbb5863017a2db5c60f3a600e68fe32e93/inst/rawrrassembly/rawrr.cs#L405) and there are two methods:
>
> ```
> var [...]

asepsiswu, 2023-11-08:
I used the Python version of **thermofisherlsms/RawFileReader** and the number of peaks in the first scan was 2842. However, since I'm not familiar with C#, I couldn't discern how **compomics/ThermoRawFileParser** gets different result. 

In the end, I decided to close the issue.

Regarding the issue title, it may be necessary to change it, as it is not related to profile mode.

> I would not call this an "inconsistency". There is different ways of centroiding profile data, I guess they all have their pros and cons, theoretical foundations. My advice is: Use the centroiding [...]

closed 2023-11-08 by asepsiswu