fgcz/rawrr #71: rawrr::buildRawrrExe() fails

SamThilmany, 2024-02-16:
I have created a Docker container based on [`rocker/r2u`](https://eddelbuettel.github.io/r2u/) for my statistical analyses. Unfortunately, `rawrr` is not included in the [pool of `r2u` packages](https://r2u.stat.illinois.edu/ubuntu/pool/dists/jammy/main/) and therefore has to be installed manually.
I have attempted to do so both in the Dockerfile and in the terminal of the finished container and failed in both cases. The `rawrr` package can be installed, but the `rawrr::buildRawrrExe()` process fails. According to the error message, a log file was stored under `/root/.cache/R/rawrr/rawrrassembly/rawrr_build.log.*`, but this file does not exist.

Here is the full error message:

```
Attempting to build 'rawrr.exe', one time setup ...
Error in rawrr::buildRawrrExe() : 
  Building 'rawrr.exe' failed. For details see the build report, supposed to be saved in:
'/root/.cache/R/rawrr/rawrrassembly/rawrr_build.log.9fa3cd0ef28'
Call 'rawrr::installRawrrExe()' to download and install a precompiled version
from a remote location. Note this requires internet connection.
In addition: Warning message:
In system2(cmd, cmdArgs, wait = TRUE, stderr = TRUE, stdout = TRUE) :
  running command ''xbuild' /p:OutputPath='/root/.cache/R/rawrr/rawrrassembly'/ /p:AdditionalLibPaths='/root/.cache/R/rawrr/rawrrassembly' /v:diagnostic /flp:LogFile='/root/.cache/R/rawrr/rawrrassembly/rawrr_build.log.9fa3cd0ef28' rawrr.csproj 2>&1' had status 134
Execution halted
```

Do you know what the problem might be?

<hr>

### Dockerfile

```dockerfile
# Use the r2u image as the base
FROM rocker/r2u

# Switch to the root user
USER root

# Set the working directory to /app
WORKDIR /app

# Install Mono
RUN apt-get update && \
    apt-get install -y --no-install-recommends \
    mono-mcs \
    mono-xbuild

# Install Jupyter Notebook and IRkernel
RUN apt-get update && apt-get install -y --no-install-recommends \
    python3 \
    python3-pip

RUN pip3 install --upgrade pip && \
    pip3 install jupyter

RUN apt-get install -y --no-install-recommends \
    r-cran-irkernel

RUN Rscript -e "IRkernel::installspec(user = FALSE)"

# Install R Packages
RUN apt-get install -y --no-install-recommends \
    r-cran-tidyverse \
    r-cran-showtext \
    r-cran-ggally \
    r-cran-cowplot \
    r-cran-ggrepel \
    r-cran-doparallel \
    r-cran-combinat \
    r-cran-biocmanager \
    r-bioc-limma \
    r-bioc-clusterprofiler \
    r-bioc-annotationdbi \
    r-bioc-org.hs.eg.db

# Install rawrr
RUN R -e "BiocManager::install('rawrr', update = TRUE, ask = FALSE)" \
    R -e "tools::R_user_dir('rawrr', which = 'cache')" \
    R -e "rawrr::installRawFileReaderDLLs()" \
    R -e "rawrr::buildRawrrExe()"

# Expose the Jupyter Notebook port
EXPOSE 8888

# Create JupyterLab configuration directory
RUN mkdir -p /etc/jupyter

# Add JupyterLab configuration to save notebooks in /app/notebooks
RUN echo "c.NotebookApp.notebook_dir = '/app/notebooks'" >> /etc/jupyter/jupyter_lab_config.py
RUN echo "c.MappingKernelManager.kernel_cmd_timeout = 3600" >> /etc/jupyter/jupyter_notebook_config.py

# # Command to run Jupyter Notebook
CMD ["jupyter", "lab", "--ip=0.0.0.0", "--port=8888", "--no-browser", "--allow-root"]
```

### docker-compose.yml

```yml
version: '3'

services:
  jupyter:
    build:
      context: ..
      dockerfile: ./jupyter-r/Dockerfile
    ports:
      - "8888:8888"
    volumes:
      - ../notebooks:/app/notebooks  # Mount a local folder to store notebooks
    environment:
      - TZ=UTC+1  # Set the container timezone
    image: jupyter_r
    container_name: jupyter_r
```

tobiasko, 2024-02-19:
Hi @SamThilmany,

I am unfortunately not to familiar with building docker containers. Does the location `/root/.cache/R/rawrr/rawrrassembly/` exist for the file system of the docker container? Can you place a file there using the command line of the container (is it writeable)? Have you tested the basic functionality of the mono compiler? Was is correctly build/installed in the container? On which platform do you run the docker container (OS)?

Have you ever tried to install rawrr in a bioc docker container?

https://www.bioconductor.org/help/docker/

Does it show the same behaviour?

Sorry for the many question.

Best,
Tobi

tobiasko, 2024-02-19:
The mono project homepage shows some basic tests for the mono compiler

https://www.mono-project.com/docs/getting-started/mono-basics/

tobiasko, 2024-02-19:
I think it would be nice to develop a recipe for `rawrr` application within containers.

tobiasko, 2024-02-21:
Maybe it would even make sense to publish your docker image on docker hub? A container image based on Debian Bookworm might be interesting for people that don't like the bioconductor containers for some reason.

tobiasko, 2024-02-21:
The bioc containers incl. stuff like RStudio server, which is of course a cool thing for people that depend on an IDE, but there is for sure also people how prefer a stripped down minimal image.

https://github.com/Bioconductor/bioconductor_docker