fgcz/rawrr #72: auc.rawrrChromatogram question

flolai, 2024-04-04:
Hello There,

I want to use auc.rawrrChromatogram to calculate area under the curve. However, not all of my peaks are 'clean' and some may have coelution issues. Any suggestions on how to resolve this? I suppose if I am not too worry about coelution then this function will calculate all the areas (all compounds) with the same m/z?

Thanks very much in advance

Florence

tobiasko, 2024-10-18:
Hi @flolai,

`auc.rawrrChromatogram()` just iterates over a `rawrrChromatogram` object.  It has no clue about compounds. Whatever you decide to put into the object is integrated. Example XIC for 487.25 m/z with 10 ppm tolerance.

```
> x <- readChromatogram(rawfile = sampleFilePath(), mass = 487.25)
> x
[[1]]
$filter
[1] "ms"

$ppm
[1] 10

$mass
[1] 487.25

$times
 [1] 0.001619751 0.031642766 0.061663615 0.091651065 0.121640750 0.151644970 0.181667770
 [8] 0.211526280 0.241284530 0.271307600 0.301222200 0.331145000 0.361147270 0.391168050
[15] 0.421057700 0.450970250 0.481045180 0.510989030 0.540955750 0.570893130 0.600724580
[22] 0.630620300 0.660428770 0.690318400 0.720320350 0.750197620 0.780105920

$intensities
 [1] 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

attr(,"filename")
[1] "/Library/Frameworks/R.framework/Versions/4.4-arm64/Resources/library/rawrr/extdata/sample.raw"
attr(,"type")
[1] "xic"
attr(,"class")
[1] "rawrrChromatogram"

attr(,"filter")
[1] "ms"
attr(,"filename")
[1] "/Library/Frameworks/R.framework/Versions/4.4-arm64/Resources/library/rawrr/extdata/sample.raw"
attr(,"type")
[1] "xic"
attr(,"tol")
[1] 10
attr(,"class")
[1] "rawrrChromatogramSet"
> rawrr:::auc.rawrrChromatogram(x)
[1] 0

```

tobiasko, 2024-10-18:
If you can't separate your compounds in m/z and RT rawrr can't do any magic!

closed 2024-10-18 by tobiasko