fgcz/rawrr #75: end of mono

tobiasko, 2024-10-03:
It looks like the mono project is coming to its end and will be replaced by .NET core

```
The [Mono Project (mono/mono)](https://github.com/mono/mono) (‘original mono’) has been an important part of the .NET ecosystem since it was launched in 2001. Microsoft became the steward of the Mono Project when it acquired Xamarin in 2016.

The last major release of the Mono Project was in July 2019, with minor patch releases since that time. The last patch release was February 2024.

We are happy to announce that the WineHQ organization will be taking over as the stewards of the Mono Project upstream at [wine-mono / Mono · GitLab (winehq.org)](https://gitlab.winehq.org/wine-mono/mono). Source code in existing [mono/mono](https://github.com/mono/mono) and other repos will remain available, although repos may be archived. Binaries will remain available for up to four years.

Microsoft maintains a modern fork of [Mono runtime in the dotnet/runtime repo](https://github.com/dotnet/runtime/tree/main/src/mono) and has been progressively moving workloads to that fork. That work is now complete, and we recommend that active Mono users and maintainers of Mono-based app frameworks migrate to [.NET](https://github.com/dotnet/core) which includes work from this fork.

We want to recognize that the Mono Project was the first .NET implementation on Android, iOS, Linux, and other operating systems. The Mono Project was a trailblazer for the .NET platform across many operating systems. It helped make cross-platform .NET a reality and enabled .NET in many new places and we appreciate the work of those who came before us.

Thank you to all the Mono developers!
```

Looking at https://github.com/thermofisherlsms/RawFileReader/tree/main/Libs/NetCore/Net8

makes me think that the CommonCore have meanwhile been tested for .NET 8.

https://dotnet.microsoft.com/en-us/download/dotnet/8.0

Maybe one should plan migrating  the C# code part of rawrr to .NET 8 before it's time to say: Mono RIP.

cpanse, 2024-10-22:
there seems to be an issue with arm64
https://github.com/thermofisherlsms/RawFileReader/issues/3

cpanse, 2024-10-28:
adding the source
```
cp@xerxes:~/src/github/thermofisherlsms/RawFileReader/ExampleProgram/NetCore/ > dotnet nuget add source /Users/cp/src/github/thermofisherlsms/RawFileReader/Libs/NetCore/Net8/
Package source with Name: Package source 1 added successfully.
```

publish the example code
```
cp@xerxes:~/src/github/thermofisherlsms/RawFileReader/ExampleProgram/NetCore/ > dotnet publish -a arm64
  Determining projects to restore...
  Restored /Users/cp/src/github/thermofisherlsms/RawFileReader/ExampleProgram/NetCore/RawFileReaderExample.csproj (in 148 ms).
CSC : warning CS8012: Referenced assembly 'ThermoFisher.CommonCore.MassPrecisionEstimator, Version=8.0.6.0, Culture=neutral, PublicKeyToken=1aef06afb5abd953' targets a different processor. [
/Users/cp/src/github/thermofisherlsms/RawFileReader/ExampleProgram/NetCore/RawFileReaderExample.csproj]
CSC : warning CS8012: Referenced assembly 'ThermoFisher.CommonCore.RawFileReader, Version=8.0.6.0, Culture=neutral, PublicKeyToken=1aef06afb5abd953' targets a different processor. [/Users/cp
/src/github/thermofisherlsms/RawFileReader/ExampleProgram/NetCore/RawFileReaderExample.csproj]
```

leoschwarz, 2024-10-28:
To reproduce this in the .NET Docker SDK containers:

Run with either:

- `docker run --platform linux/amd64 --rm -it mcr.microsoft.com/dotnet/sdk:8.0`
- `docker run --platform linux/arm64 --rm -it mcr.microsoft.com/dotnet/sdk:8.0`

The commands:

```bash
mkdir -p /work && cd /work
git clone --depth=1 https://github.com/thermofisherlsms/RawFileReader.git
cd RawFileReader/ExampleProgram/NetCore 
dotnet nuget add source /work/RawFileReader/Libs/NetCore/Net8/
wget https://github.com/fgcz/rawrr/blob/devel/inst/extdata/sample.raw?raw=true -O sample.raw
dotnet run sample.raw
```

It passes with the amd64 contanier, but fails with arm64.

tobiasko, 2024-10-28:
For my understanding: You can't even compile Jims example code on apple silicon? On all other platforms it works? Even on macOS with an intel processor?

cpanse, 2024-10-28:
> For my understanding: You can't even compile Jims example code on apple silicon? On all other platforms it works? Even on macOS with an intel processor?

correct

leoschwarz, 2024-10-29:
The reason it worked with mono but not dotnet core is that [mono does not support macOS ARM](https://github.com/mono/mono/issues/21092) and runs in x86_64 emulation mode always. For dotnet core we tried to build native binaries which does not work as we saw, as currently the libraries are only available for x86_64.

cpanse, 2024-10-29:
```
install.packages("https://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr_1.15.1.tar.gz")
library(rawrr)
RAWRRDOTNET <<- "Have Fun"

rawrr::installRawrrExe()

rawrr::sampleFilePath() |> readFileHeader()
rawrr::sampleFilePath() |> readIndex() |> tail()
rawrr::sampleFilePath() |> readSpectrum(1) |> tail()
```

cpanse, 2024-10-30:
# rbenchmark

Apple M3 Max

```
                   test replications elapsed relative user.self sys.self
6 dotnet-readFileHeader          100 387.753    2.728     0.125    0.128
4      dotnet-readIndex          100 735.654    5.175     5.854    8.358
5   dotnet-readSpectrum          100 233.909    1.645    46.554    1.149
3   mono-readFileHeader          100 142.157    1.000     0.202    0.280
1        mono-readIndex          100 324.513    2.283     5.983    7.276
2     mono-readSpectrum          100 222.157    1.563    45.245    1.229
```

```
#R

f <- 'Library/Caches/org.R-project.R/R/ExperimentHub/46314c3933e2_4590.raw'

library(rbenchmark)

benchmark("mono-readIndex" = {
            if (exists('RAWRRDOTNET')) { rm (RAWRRDOTNET) }
            rawrr::readIndex(f) -> idx.mono
          },
          "mono-readSpectrum" = {
            if (exists('RAWRRDOTNET')) { rm (RAWRRDOTNET) }
            rawrr::readSpectrum(f, 1:1000) -> spectra.mono
          },
          "mono-readFileHeader" = {
            if (exists('RAWRRDOTNET')) { rm (RAWRRDOTNET) }
            rawrr::readFileHeader(f) -> header.mono
          },
          "dotnet-readIndex" = {
            RAWRRDOTNET <- "FUN"
            rawrr::readIndex(f) -> idx.dotnet
          },
          "dotnet-readSpectrum" = {
            RAWRRDOTNET <- "FUN"
            rawrr::readSpectrum(f, 1:1000) -> spectra.dotnet
          },
          "dotnet-readFileHeader" = {
            RAWRRDOTNET <- "FUN"
            rawrr::readFileHeader(f) -> header.dotnet
          },
          replications = 100,
          columns = c("test", "replications", "elapsed",
                      "relative", "user.self", "sys.self"))

```

leoschwarz, 2024-11-05:
To make the mac executable `rawrr` executable run without signature (we could also look into buying these in the future) we should run
`xattr -d com.apple.quarantine /path/to/binary`
before trying the executable to remove the quarantine attribute and make it runnable.