fgcz/rawrr #76: Newbie: can't even get the first command to run properly

buzuddha, 2024-10-17:
Sorry to be completely inept with this...

I've installed Rawrr according to the instructions:

```
> BiocManager::install("rawrr")
$ sudo apt-get install mono-runtime libmono-system-data4.0-cil -y
> rawrr::installRawFileReaderDLLs()
> rawrr::installRawrrExe()
```
When I try to run the first command in the docs on a `.raw` file I run into problems. I don't really understand the `Rcode file to parse does not exist` message, but I'm new to rawrr.

```
> library(rawrr)
Package 'rawrr' version 1.10.2 using
RawFileReader reading tool. Copyright © 2016 by Thermo Fisher Scientific, Inc. All rights reserved.
> rawfile <- "~/Downloads/ftp_local/1284_01.raw"
> H <- rawrr::readFileHeader(rawfile = rawfile)
Error in .rawrrSystem2Source(rawfile, input = NULL, rawrrArgs = "headerR") :
  Rcode file to parse does not exist. '/home/user/.cache/R/rawrr/rawrrassembly/rawrr.exe' failed for an unknown reason.
Please check the debug files:
        /tmp/RtmpI8hKLq/file772e2ea5b01d.stderr
        /tmp/RtmpI8hKLq/file772e4799b292.stdout
and the System Requirements
```
 The `.dll` files appear to be in place and so is the `rawrr.exe`. 
```
> list.files("/home/user/.cache/R/rawrr/rawrrassembly/")
[1] "eula.txt"
[2] "rawrr.exe"
[3] "ThermoFisher.CommonCore.BackgroundSubtraction.dll"
[4] "ThermoFisher.CommonCore.Data.dll"
[5] "ThermoFisher.CommonCore.MassPrecisionEstimator.dll"
[6] "ThermoFisher.CommonCore.RawFileReader.dll"
```
The logs don't seem to tell me much:
```
> system("cat /tmp/RtmpI8hKLq/file772e2ea5b01d.stderr")
> system("cat /tmp/RtmpI8hKLq/file772e4799b292.stdout")
Unable to access the RAW file using the RawFileReader class!
```

Any help here would be greatly appreciated.

```
$ uname -r
5.15.153.1-microsoft-standard-WSL2
$ R --version
R version 4.3.3 (2024-02-29) -- "Angel Food Cake"
```
Similar issues trying this on macOS. The `.raw` file is from [PXD028532](https://proteomecentral.proteomexchange.org/cgi/GetDataset?ID=PXD028532) which appears to have been acquired on the correct Orbitrap Fusion instrument.

tobiasko, 2024-10-18:
My feeling is that your `rawrr.exe` is not functional. Based on your R commands and the content of `.../rawrrassembly/` you downloaded `rawer.exe`, see `?installRawrrExe`. Maybe try to delete the downloaded `rawer.exe` file and compile from the included C# code in the rawrr package. On my Mac this happens auto. on package startup, in case `rawrr.exe` is not found.

```
> BiocManager::install("rawrr")
'getOption("repos")' replaces Bioconductor standard repositories, see
'help("repositories", package = "BiocManager")' for details.
Replacement repositories:
    CRAN: https://cran.rstudio.com/
Bioconductor version 3.19 (BiocManager 1.30.25), R 4.4.1 (2024-06-14)
Installing package(s) 'rawrr'
trying URL 'https://bioconductor.org/packages/3.19/bioc/bin/macosx/big-sur-arm64/contrib/4.4/rawrr_1.12.0.tgz'
Content type 'application/x-gzip' length 2085113 bytes (2.0 MB)
==================================================
downloaded 2.0 MB

The downloaded binary packages are in
	/var/folders/_l/psxm9vkj6m11z1vws9plwj6h0000gn/T//RtmptdCbm8/downloaded_packages
> library("rawrr")
'ThermoFisher.CommonCore.BackgroundSubtraction.dll' is missing.
'ThermoFisher.CommonCore.Data.dll' is missing.
'ThermoFisher.CommonCore.MassPrecisionEstimator.dll' is missing.
'ThermoFisher.CommonCore.RawFileReader.dll' is missing.
'ThermoFisher.CommonCore.*.dll' files are not available on the system.
Run 'rawrr::installRawFileReaderDLLs()' or setenv MONO_PATH to the location where the assemblies are located.
For more information, type '?ThermoFisher'.
Package 'rawrr' version 1.12.0 using
RawFileReader reading tool. Copyright © 2016 by Thermo Fisher Scientific, Inc. All rights reserved.
> rawrr::installRawFileReaderDLLs()
Do you accept the Thermo License agreement '/Library/Frameworks/R.framework/Versions/4.4-arm64/Resources/library/rawrr/rawrrassembly/RawFileReaderLicense.txt'? [Y/n]: Y
trying URL 'https://github.com/thermofisherlsms/RawFileReader/raw/main/Libs/Net471//ThermoFisher.CommonCore.BackgroundSubtraction.dll'
Content type 'application/octet-stream' length 44544 bytes (43 KB)
==================================================
downloaded 43 KB

trying URL 'https://github.com/thermofisherlsms/RawFileReader/raw/main/Libs/Net471//ThermoFisher.CommonCore.Data.dll'
Content type 'application/octet-stream' length 406016 bytes (396 KB)
==================================================
downloaded 396 KB

trying URL 'https://github.com/thermofisherlsms/RawFileReader/raw/main/Libs/Net471//ThermoFisher.CommonCore.MassPrecisionEstimator.dll'
Content type 'application/octet-stream' length 11264 bytes (11 KB)
==================================================
downloaded 11 KB

trying URL 'https://github.com/thermofisherlsms/RawFileReader/raw/main/Libs/Net471//ThermoFisher.CommonCore.RawFileReader.dll'
Content type 'application/octet-stream' length 654336 bytes (639 KB)
==================================================
downloaded 639 KB

 ThermoFisher.CommonCore.BackgroundSubtraction.dll                   ThermoFisher.CommonCore.Data.dll ThermoFisher.CommonCore.MassPrecisionEstimator.dll 
                                                 0                                                  0                                                  0 
         ThermoFisher.CommonCore.RawFileReader.dll 
                                                 0 

Restarting R session...

> library("rawrr")
Attempting to build 'rawrr.exe', one time setup ...
'rawrr.exe' successfully built in 
'/Users/tobiaskockmann/Library/Caches/org.R-project.R/R/rawrr/rawrrassembly/rawrr.exe'.
The build report should have been saved in
'/Users/tobiaskockmann/Library/Caches/org.R-project.R/R/rawrr/rawrrassembly/rawrr_build.log.28cb22339cb1'.
Package 'rawrr' version 1.12.0 using
RawFileReader reading tool. Copyright © 2016 by Thermo Fisher Scientific, Inc. All rights reserved.
```

You can also manually trigger the build using `buildRawrrExe()`. In each case you should see something like:

```
> list.files(rawrrAssemblyPath())
[1] "eula.txt"                                          
[2] "rawrr_build.log.28cb22339cb1"                      
[3] "rawrr.exe"                                         
[4] "ThermoFisher.CommonCore.BackgroundSubtraction.dll" 
[5] "ThermoFisher.CommonCore.Data.dll"                  
[6] "ThermoFisher.CommonCore.MassPrecisionEstimator.dll"
[7] "ThermoFisher.CommonCore.RawFileReader.dll"  
```

tobiasko, 2024-10-18:
... and better use

```
> sessionInfo()
R version 4.4.1 (2024-06-14)
Platform: aarch64-apple-darwin20
Running under: macOS Sonoma 14.6

Matrix products: default
BLAS:   /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib 
LAPACK: /Library/Frameworks/R.framework/Versions/4.4-arm64/Resources/lib/libRlapack.dylib;  LAPACK version 3.12.0

locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8

time zone: Europe/Zurich
tzcode source: internal

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] rawrr_1.12.0

loaded via a namespace (and not attached):
[1] compiler_4.4.1 tools_4.4.1   
```

for environment infos.

tobiasko, 2024-10-18:
You could also check the integrity/which `rawrr.exe` you currently have by comparing the md5 sum to

https://github.com/fgcz/rawrr/releases

buzuddha, 2024-10-18:
Hi @tobiasko,

Many thanks for the detail response! In fact I am able to read the sample file headers using the test `sample.raw`.

```
> library(rawrr)
Package 'rawrr' version 1.10.2 using
RawFileReader reading tool. Copyright © 2016 by Thermo Fisher Scientific, Inc. All rights reserved.
> sampleFilePath()
[1] "/home/user/R/x86_64-pc-linux-gnu-library/4.3/rawrr/extdata/sample.raw"
> readFileHeader(rawfile = sampleFilePath())
$`RAW file`
[1] "sample.raw"

$`RAW file version`
[1] "66"

$`Creation date`
[1] "02/13/2018 11:09:03"

$Operator
[1] "ExactiveUser"

$`Number of instruments`
[1] 1

...

$`User text 4`
[1] ""
```

I don't believe it's an issue with the specific files I've attempted to load, as I've pulled a few different ones from the proteomExchange and they all have the same error.

[PXD042189](https://www.ebi.ac.uk/pride/archive/projects/PXD042189)
[PXD042189](https://www.ebi.ac.uk/pride/archive/projects/PXD042189)
[PXD045263](https://www.ebi.ac.uk/pride/archive/projects/PXD045263)

```
> readFileHeader(rawfile = "~/Downloads/ftp_local/1284_01.raw")
Error in .rawrrSystem2Source(rawfile, input = NULL, rawrrArgs = "headerR") :
  Rcode file to parse does not exist. '/home/user/.cache/R/rawrr/rawrrassembly/rawrr.exe' failed for an unknown reason.
Please check the debug files:
        /tmp/RtmpBUXYFG/file79bd1ab850d3.stderr
        /tmp/RtmpBUXYFG/file79bd690f7e50.stdout
and the System Requirements
> readFileHeader(rawfile = "~/Downloads/ftp_local/2192_Hg_ARP_10_.raw")
Error in .rawrrSystem2Source(rawfile, input = NULL, rawrrArgs = "headerR") :
  Rcode file to parse does not exist. '/home/user/.cache/R/rawrr/rawrrassembly/rawrr.exe' failed for an unknown reason.
Please check the debug files:
        /tmp/RtmpBUXYFG/file79bd2d9b0196.stderr
        /tmp/RtmpBUXYFG/file79bd6b74cdb9.stdout
and the System Requirements
> readFileHeader(rawfile = "~/Downloads/ftp_local/50kDato75kDa.raw")
Error in .rawrrSystem2Source(rawfile, input = NULL, rawrrArgs = "headerR") :
  Rcode file to parse does not exist. '/home/user/.cache/R/rawrr/rawrrassembly/rawrr.exe' failed for an unknown reason.
Please check the debug files:
        /tmp/RtmpBUXYFG/file79bdeccccc4.stderr
        /tmp/RtmpBUXYFG/file79bd455a2571.stdout
and the System Requirements
> readFileHeader(rawfile = "~/Downloads/ftp_local/NF2-2_75737.raw")
Error in .rawrrSystem2Source(rawfile, input = NULL, rawrrArgs = "headerR") :
  Rcode file to parse does not exist. '/home/user/.cache/R/rawrr/rawrrassembly/rawrr.exe' failed for an unknown reason.
Please check the debug files:
        /tmp/RtmpBUXYFG/file79bd1502c044.stderr
        /tmp/RtmpBUXYFG/file79bd14fdfa32.stdout
and the System Requirements
```
I'm unable to build `Rawrr.exe`. I don't have msbuild nor xbuild and these appear to be dependencies:
```
$ rm /home/user/.cache/R/rawrr/rawrrassembly/rawrr.exe
> library(rawrr)
Could not find 'msbuild' or 'xbuild' in the path. Therefore, it is not possible to build the 'rawrr.exe' assembly from source code.
Try to run rawrr::installRawrrExe().
Package 'rawrr' version 1.10.2 using
RawFileReader reading tool. Copyright © 2016 by Thermo Fisher Scientific, Inc. All rights reserved.
> sessionInfo()
R version 4.3.3 (2024-02-29)
Platform: x86_64-pc-linux-gnu (64-bit)
Running under: Ubuntu 24.04.1 LTS

Matrix products: default
BLAS:   /usr/lib/x86_64-linux-gnu/blas/libblas.so.3.12.0
LAPACK: /usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3.12.0

locale:
 [1] LC_CTYPE=C.UTF-8       LC_NUMERIC=C           LC_TIME=C.UTF-8
 [4] LC_COLLATE=C.UTF-8     LC_MONETARY=C.UTF-8    LC_MESSAGES=C.UTF-8
 [7] LC_PAPER=C.UTF-8       LC_NAME=C              LC_ADDRESS=C
[10] LC_TELEPHONE=C         LC_MEASUREMENT=C.UTF-8 LC_IDENTIFICATION=C

time zone: America/New_York
tzcode source: system (glibc)

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base

other attached packages:
[1] rawrr_1.10.2

loaded via a namespace (and not attached):
[1] compiler_4.3.3 tools_4.3.3
```
I do have mono in the `$PATH1`, which I understand is the xbuild compiler. 
```
$ which mono
/usr/bin/mono

$ ls -l /usr/lib/mono/
total 12
drwxr-xr-x  2 root root 4096 Oct 17 15:46 4.5
drwxr-xr-x  3 root root 4096 Oct 17 15:44 aot-cache
drwxr-xr-x 15 root root 4096 Oct 17 15:46 gac
```
Perhaps xbuild has been removed from the mono package? [The mono website lists the xbuild page as outdated](https://www.mono-project.com/archived/xbuild/). How do I get one/both of these?

A quick `apt install` for either `msbuild` or `xbuild` yields `E: Unable to locate package`. 

As an aside, the downloader appears to pull the `rawrr.exe` correctly when comparing to the [hash on github](https://github.com/fgcz/rawrr/releases). So likely not an issue there.
```
> rawrr::installRawrrExe()
trying URL 'https://github.com/fgcz/rawrr/releases/download/1.9.2/rawrr.1.9.2.exe'
Content type 'application/octet-stream' length 28672 bytes (28 KB)
==================================================
downloaded 28 KB

MD5 96e3a4cc1b7caaf92890d85ed4c72f77 /home/user/.cache/R/rawrr/rawrrassembly/rawrr.exe
> md5sum("~/.cache/R/rawrr/rawrrassembly/rawrr.exe") == "96e3a4cc1b7caaf92890d85ed4c72f77"
/home/hugh/.cache/R/rawrr/rawrrassembly/rawrr.exe
                                             TRUE
```

tobiasko, 2024-10-18:
Could you try

```
sudo apt-get install mono-mcs mono-xbuild
```

that was at least working for Ubuntu Linux at some point according to

https://bioconductor.org/packages/release/bioc/install/rawrr/INSTALL

tobiasko, 2024-10-18:
Which mono version is installed on your system?

The latest stable release should be 6.12.

https://www.mono-project.com/download/stable/#download-lin

Not sure if it matters, but maybe update. I am running

```
tobiaskockmann@Tobiass-MacBook-Pro ~ % mono --version
Mono JIT compiler version 6.12.0 ((no/0cbf0e290c3 Wed Apr 24 13:37:51 UTC 2024)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           
	SIGSEGV:       normal
	Notification:  kqueue
	Architecture:  arm64
	Disabled:      none
	Misc:          softdebug 
	Interpreter:   yes
	LLVM:          supported, not enabled.
	Suspend:       preemptive
	GC:            sgen
```

tobiasko, 2024-10-18:
According to your very first comment you only installed the mono runtime:

```
$ sudo apt-get install mono-runtime libmono-system-data4.0-cil -y
```

That is not sufficient to compile the C# code.

tobiasko, 2024-10-18:
But it is interesting that the package sample data works, while all other downloaded files fail. Have you inspected one of the debug files listed?

buzuddha, 2024-10-18:
Ah, I see... I guess I had figured precompiled would be quicker.

So, step 1: install `mono-mcs mono-xbuild`. 
```
xbuild --version

>>>> xbuild tool is deprecated and will be removed in future updates, use msbuild instead <<<<

XBuild Engine Version 14.0
Mono, Version 6.8.0.105
Copyright (C) 2005-2013 Various Mono authors
MSBUILD: error MSBUILD0000: Project file '--version' not found.
```
Step 2: `Build rawrr.exe`, which now works. Promising...
```
> buildRawrrExe()
Attempting to build 'rawrr.exe', one time setup ...
'rawrr.exe' successfully built in
'/home/hugh/.cache/R/rawrr/rawrrassembly/rawrr.exe'.
The build report should have been saved in
'/home/hugh/.cache/R/rawrr/rawrrassembly/rawrr_build.log.10a726f960c1'.
[1] TRUE
```
Step 3: Check that the `rawrr.exe` is in place. It is...so close...
```
> list.files('/home/hugh/.cache/R/rawrr/rawrrassembly/')
[1] "eula.txt"
[2] "rawrr_build.log.10a726f960c1"
[3] "rawrr.exe"
[4] "ThermoFisher.CommonCore.BackgroundSubtraction.dll"
[5] "ThermoFisher.CommonCore.Data.dll"
[6] "ThermoFisher.CommonCore.MassPrecisionEstimator.dll"
[7] "ThermoFisher.CommonCore.RawFileReader.dll"
```
Step 4: Attempt to read file headers. Maybeeee...
```
> readFileHeader(rawfile = "~/Downloads/ftp_local/1284_01.raw")
Error in .rawrrSystem2Source(rawfile, input = NULL, rawrrArgs = "headerR") :
  Rcode file to parse does not exist. '/home/hugh/.cache/R/rawrr/rawrrassembly/rawrr.exe' failed for an unknown reason.
Please check the debug files:
        /tmp/Rtmpq0trwd/file10a72790015ad.stderr
        /tmp/Rtmpq0trwd/file10a7231cc5f1d.stdout
and the System Requirements
```
Step 5: Agonize
Step 6: Check the debug logs. Nada.
```
> system('cat /tmp/Rtmpq0trwd/file10a72790015ad.stderr')
> system('cat /tmp/Rtmpq0trwd/file10a7231cc5f1d.stdout')
Unable to access the RAW file using the RawFileReader class!
```
So I'm still not sure where to go from here. I'll try getting mono 6.12 just in case.

For reference, the available mono version in the `apt` repository is:
```
$ sudo apt-cache policy mono-runtime
mono-runtime:
  Installed: 6.8.0.105+dfsg-3.6ubuntu2
  Candidate: 6.8.0.105+dfsg-3.6ubuntu2
  Version table:
 *** 6.8.0.105+dfsg-3.6ubuntu2 500
        500 http://archive.ubuntu.com/ubuntu noble/universe amd64 Packages
        100 /var/lib/dpkg/status
```
So I'm a bit behind, but hopefully this isn't the problem.

buzuddha, 2024-10-18:
Ruh roh...

So I removed everything `sudo apt remove libmono-corlib4.5-cil libmono-i18n4.0-cil
 libmono-system-configuration4.0-cil libmono-system-core4.0-cil libmono-system4.0-cil mono-4.0-gac mono-gac mono-mcs mono-runtime mono-runtime-common mono-runtime-sgen`

Then reinstalled the relevant packages `sudo apt install mono-mcs mono-xbuild mono-runtime libmono-system-data4.0-cil`

Confirmed I have the latest version of mono
```
mono --version
Mono JIT compiler version 6.12.0.200 (tarball Tue Jul 11 21:37:50 UTC 2023)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           __thread
        SIGSEGV:       altstack
        Notifications: epoll
        Architecture:  amd64
        Disabled:      none
        Misc:          softdebug
        Interpreter:   yes
        LLVM:          yes(610)
        Suspend:       hybrid
        GC:            sgen (concurrent by default)
```
Buuuut
```
> library(rawrr)
Attempting to build 'rawrr.exe', one time setup ...
Error: package or namespace load failed for ‘rawrr’:
 .onLoad failed in loadNamespace() for 'rawrr', details:
  call: buildRawrrExe()
  error: Building 'rawrr.exe' failed. For details see the build report, supposed to be saved in:
'/home/hugh/.cache/R/rawrr/rawrrassembly/rawrr_build.log.12236c638fb2'
Call 'rawrr::installRawrrExe()' to download and install a precompiled version
from a remote location. Note this requires internet connection.
In addition: Warning message:
In system2(cmd, cmdArgs, wait = TRUE, stderr = TRUE, stdout = TRUE) :
  running command ''xbuild' /p:OutputPath='/home/user/.cache/R/rawrr/rawrrassembly'/ /p:AdditionalLibPaths='/home/user/.cache/R/rawrr/rawrrassembly' /v:diagnostic /flp:LogFile='/home/user/.cache/R/rawrr/rawrrassembly/rawrr_build.log.12236c638fb2' rawrr.csproj 2>&1' had status 1

> system('cat /home/hugh/.cache/R/rawrr/rawrrassembly/rawrr_build.log.12236c638fb2')
﻿Loading default tasks for ToolsVersion: 14.0 from /usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.tasks

Build started 10/18/2024 17:04:42.
__________________________________________________
Project "/home/user/R/x86_64-pc-linux-gnu-library/4.3/rawrr/rawrrassembly/rawrr.csproj" (default target(s)):
Building target "Build" in project "/home/hugh/R/x86_64-pc-linux-gnu-library/4.3/rawrr/rawrrassembly/rawrr.csproj" ("/home/user/R/x86_64-pc-linux-gnu-library/4.3/rawrr/rawrrassembly/rawrr.csproj").
        Target Build:
        Task "Csc"
                Using task Csc from Microsoft.Build.Tasks.Csc, Microsoft.Build.Tasks.Core, Version=14.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a
        Task "Csc" execution -- FAILED
        Done building target "Build" in project "/home/hugh/R/x86_64-pc-linux-gnu-library/4.3/rawrr/rawrrassembly/rawrr.csproj".-- FAILED
Done building target "Build" in project "/home/user/R/x86_64-pc-linux-gnu-library/4.3/rawrr/rawrrassembly/rawrr.csproj" ("/home/user/R/x86_64-pc-linux-gnu-library/4.3/rawrr/rawrrassembly/rawrr.csproj").
Done building project "/home/user/R/x86_64-pc-linux-gnu-library/4.3/rawrr/rawrrassembly/rawrr.csproj".-- FAILED

Build FAILED.
         0 Warning(s)
         0 Error(s)

Time Elapsed 00:00:00.1546190```

Ack...maybe those packages were held back for a reason?

buzuddha, 2024-10-18:
Ok phew... I was able to 
1) remove the mono repository from `/etc/apt/sources.list.d/mono-official-stable.list` 
2) run the `apt update`
3) `apt remove` all the mono packages 
4) `apt install mono-mcs mono-xbuild mono-runtime libmono-system-data4.0-cil` 
5) confirm that the rawrr.exe build works (with `Mono JIT compiler version 6.8.0.105`)
```
> library(rawrr)
Attempting to build 'rawrr.exe', one time setup ...
'rawrr.exe' successfully built in
'/home/hugh/.cache/R/rawrr/rawrrassembly/rawrr.exe'.
The build report should have been saved in
'/home/hugh/.cache/R/rawrr/rawrrassembly/rawrr_build.log.12c8e6fd0a48a'.
Package 'rawrr' version 1.10.2 using
RawFileReader reading tool. Copyright © 2016 by Thermo Fisher Scientific, Inc. All rights reserved.
```
So I'm back stuck where I was before upgrading to 6.12 - can build `rawrr.exe` but still no ability to read in `.raw` files.

Any ideas? (Thanks for all your help by the way. Sorry to be a pain here)

tobiasko, 2024-10-20:
It is absolutely embarassing!

```
> readFileHeader(rawfile = "~/Downloads/20240925_C36374_006_S777857_MS2_mutpep_Mutant.raw")
Error in .rawrrSystem2Source(rawfile, input = NULL, rawrrArgs = "headerR") : 
  Rcode file to parse does not exist. '/Users/tobiaskockmann/Library/Caches/org.R-project.R/R/rawrr/rawrrassembly/rawrr.exe' failed for an unknown reason.
Please check the debug files:
	/var/folders/_l/psxm9vkj6m11z1vws9plwj6h0000gn/T//Rtmpd04CR2/file28cbee6948e.stderr
	/var/folders/_l/psxm9vkj6m11z1vws9plwj6h0000gn/T//Rtmpd04CR2/file28cb7ee779b1.stdout
and the System Requirements
> readFileHeader(rawfile = "/Users/tobiaskockmann/Downloads/20240925_C36374_006_S777857_MS2_mutpep_Mutant.raw")
$`RAW file`
[1] "20240925_C36374_006_S777857_MS2_mutpep_Mutant.raw"

$`RAW file version`
[1] "66"

$`Creation date`
[1] "9/25/2024 9:12:07 PM"

$Operator
[1] "Orbitrap Lumos"

$`Number of instruments`
[1] 2

$Description
[1] ""

$`Instrument model`
[1] "Orbitrap Fusion Lumos"

$`Instrument name`
[1] "Orbitrap Fusion Lumos"

$`Instrument method`
[1] "C:/Xcalibur/methods/__Service/ProteinInteraction/DIA_70x10Da_GT45min_RT65min.meth"

$`Serial number`
[1] "FSN20682"

$`Software version`
[1] "3.4.3072.18"

$`Firmware version`
[1] ""

$Units
[1] "None"

$`Mass resolution`
[1] "0.500"

$`Number of scans`
[1] 72846

$`Number of ms2 scans`
[1] 71819

$`Scan range`
[1]     1 72846

$`Time range`
[1]  0.00 65.04

$`Mass range`
[1]  145 1450

$`Scan filter (first scan)`
[1] "FTMS + c ESI Full ms [350.0000-1050.0000]"

$`Scan filter (last scan)`
[1] "FTMS + c ESI Full ms2 1045.0000@hcd30.00 [145.0000-1450.0000]"

$`Total number of filters`
[1] "71"

$`Sample name`
[1] "MS2_mutpep"

$`Sample id`
[1] "777857"

$`Sample type`
[1] "Unknown"

$`Sample comment`
[1] "on beads dig, inj 3/20ul"

$`Sample vial`
[1] "1:A,3"

$`Sample volume`
[1] "0"

$`Sample injection volume`
[1] "3"

$`Sample row number`
[1] "4"

$`Sample dilution factor`
[1] "0"

$`Sample barcode`
[1] ""

$`User text 0`
[1] ""

$`User text 1`
[1] ""

$`User text 2`
[1] "FGCZ"

$`User text 3`
[1] ""

$`User text 4`
[1] ""

```

tobiasko, 2024-10-20:
It looks like the function can't deal with the a simple relative path. OMG! Is this 2024 or the 90s... This is also the reason why the sample data worked, while all your downloaded files didn't.

```
sampleFilePath()
[1] "/Library/Frameworks/R.framework/Versions/4.4-arm64/Resources/library/rawrr/extdata/sample.raw"
```

I am really sorry for this!

buzuddha, 2024-10-20:
OMG! I'm so glad it was something so simple! I was beginning to worry I was going to have major issues getting this thing working!

Merci vielmal!

PR #78 (2024-10-22): Consistently normalize paths
Fixes #76

Input paths were normalized inconsistently before. This will be fixed by this PR and will also yield better error messages if a similar problem reoccurs.

As a result the simple Bash-style leading tilde expansion pattern is supported now (which was the problem in #76, not relative paths in general), more complex forms are not (e.g. `~user` or any other [more advanced](https://www.gnu.org/software/bash/manual/html_node/Tilde-Expansion.html) ones as these are very shell specific).

Note that these pattern features are widely unsupported or inconsistently s [... cut]
closed 2024-10-22 by cpanse