fgcz/rawrr #84: Error accessing RAWFileReader library! - The input string ... was not in a correct format.

rlumpkin, 2025-04-02:
When I try to read a scan using readSpectrum, I am getting the error "Error accessing RAWFileReader library! - The input string ... was not in a correct format." I looked at the tfo temporary .R file generated by .rawrrSystem2Source, and it is clearly generating unfinished code.

Notably - I haven't had any issues with data generated on Thermo instruments with Instrument Control 4.0, but on 4.2 I began seeing this issue.

Rawrr Versions:
On rawrr_1.10.2 and rawrr_1.14.0 couldn't read any scans from these files, regardless of the parameters set. 
On rawrr_1.15.10 I can sometimes read in the scans from the files, but only if I use mode = "" and not mode="barebone".

[image]

tobiasko, 2025-04-03:
Could you please provide the R code that led to the error incl. a call to `readFileHeader()` for the same raw files?

rlumpkin, 2025-04-03:
[file_1_header.csv](https://github.com/user-attachments/files/19592177/file_1_header.csv)

Here is the code I used just directly using the rawrr package. I attached the raw file header, and 

```
file = "path-to-file"

header <- rawrr::readFileHeader(file)
write.csv(header, "header.csv")

# Error accessing RAWFileReader library! - The input string '-7.87' was not in a correct format.
# Note - '-7.87' corresponds to Temperature Comp. (ppm)
rawrr::readSpectrum(rawfile = file, scan = 1, mode="barebone")

# Does Not Produce an Error
rawrr::readSpectrum(rawfile = file, scan = 1, mode="")

# Error accessing RAWFileReader library! - The input string '0.70' was not in a correct format.
rawrr::readSpectrum(rawfile = file, scan = 971, mode="barebone")

# Error accessing RAWFileReader library! - The input string '0.70' was not in a correct format.
rawrr::readSpectrum(rawfile = file, scan = 971, mode="")

# Error accessing RAWFileReader library! - The input string '55.2                     ' was not in a correct format.
# Note - '55.2' corresponds to HCD Energy V
rawrr::readSpectrum(rawfile = file, scan = 975, mode="barebone")

# No Error
rawrr::readSpectrum(rawfile = file, scan = 975, mode="")
```

Alternatively I tried running using source code from .rawrrSystem2Source:

```
input = 7
rawrrAwrgs = "scans" (or "cscans" for barebone)

exe <- .rawrrAssembly()

tfi <- tempfile(tmpdir=tmpdir, fileext = ".txt")
tfo <- tempfile(tmpdir=tmpdir, fileext = ".R")
tfstdout <- tempfile(tmpdir=tmpdir, fileext = ".stdout")
tfstderr <- tempfile(tmpdir=tmpdir, fileext = ".stderr" )
cat(input, file = tfi, sep="\n")
if(isFALSE(file.exists(tfi))){
        stop(paste0("No input file '", tfi, "' available!"))
}
rvs <- system2(exe, args = c( shQuote(rawfile), rawrrArgs, shQuote(tfi), shQuote(tfo)), )
```

If an error happens, it is produced by this system2 command. I saved each tfo file for the problematic scans listed above. I have attached the set of scans for scan 1 

[file_1_scan_1_barebone.txt](https://github.com/user-attachments/files/19592238/file_1_scan_1_barebone.txt)
[file_1_scan_1.txt](https://github.com/user-attachments/files/19592237/file_1_scan_1.txt)

tobiasko, 2025-04-04:
Dear @rlumpkin,

your way of describing the error makes it very hard to follow what happens on your system. I am not sure if you are new to `R` but please incl. the return of the code in your posts instead of replacing it with comments that are just an interpretation what is happening.

Example for the `readSpectrum()` function:

```
> rawrr::readSpectrum(rawrr::sampleFilePath(), scan = 1)
[[1]]
Total Ion Current:	 78725448
Scan Low Mass:	 350
Scan High Mass:	 1800
Scan Start Time (min):	 0.001619751
Scan Number:	 1
Base Peak Intensity:	 5979308
Base Peak Mass:	 445.1181
Scan Mode:	 FTMS + c NSI Full ms [350.0000-1800.0000]
======= Instrument data =====   :	NULL
Multiple Injection:	ii
Multi Inject Info:	IT=28;28 
AGC:	On
Micro Scan Count:	1
Scan Segment:	1
Scan Event:	1
Master Index:	0
Charge State:	0
Monoisotopic M/Z:	0.0000
Ion Injection Time (ms):	55.000
Max. Ion Time (ms):	55.00
FT Resolution:	120000
MS2 Isolation Width:	1450.00
MS2 Isolation Offset:	0.00
AGC Target:	3000000
HCD Energy:	
Analyzer Temperature:	27.06
=== Mass Calibration:	NULL
Conversion Parameter B:	211820589.2896
Conversion Parameter C:	-41163067.8349
Temperature Comp. (ppm):	-2.35
RF Comp. (ppm):	-0.09
Space Charge Comp. (ppm):	-2.30
Resolution Comp. (ppm):	0.57
Number of Lock Masses:	0
Lock Mass #1 (m/z):	0.0000
Lock Mass #2 (m/z):	0.0000
Lock Mass #3 (m/z):	0.0000
LM Search Window (ppm):	0.0
LM Search Window (mmu):	0.0
Number of LM Found:	0
Last Locking (sec):	0.0
LM m/z-Correction (ppm):	0.00
=== Ion Optics Settings:	NULL
S-Lens RF Level:	40.00
S-Lens Voltage (V):	21.00
Skimmer Voltage (V):	15.00
Inject Flatapole Offset (V):	5.00
Bent Flatapole DC (V):	2.00
MP2 and MP3 RF (V):	900.00
Gate Lens Voltage (V):	2.25
C-Trap RF (V):	2400.0
====  Diagnostic Data:	NULL
Dynamic RT Shift (min):	0.00
Intens Comp Factor:	1.4281
Res. Dep. Intens:	0.981
CTCD NumF:	0
CTCD Comp:	1.009
CTCD ScScr:	0.844
RawOvFtT:	2378645.5
LC FWHM parameter:	15.0
Rod:	0
[... 76 lines cut ...]
[124] R.huge_0.10.1           knitr_1.50              PSCBS_0.67.0           
[127] rstudioapi_0.17.1       tzdb_0.5.0              reshape2_1.4.4         
[130] checkmate_2.3.2         nlme_3.1-167            ggcorrplot_0.1.4.1     
[133] cachem_1.1.0            stringr_1.5.1           parallel_4.4.1         
[136] AnnotationDbi_1.68.0    pillar_1.10.1           grid_4.4.1             
[139] vctrs_0.6.5             promises_1.3.2          xtable_1.8-4           
[142] aroma.core_3.3.1        cluster_2.1.8.1         evaluate_1.0.3         
[145] isoband_0.2.7           magick_2.8.6            readr_2.1.5            
[148] mvtnorm_1.3-3           cli_3.6.4               locfit_1.5-9.12        
[151] compiler_4.4.1          rlang_1.1.5             crayon_1.5.3           
[154] R.devices_2.17.2        labeling_0.4.3          modelr_0.1.11          
[157] aroma.light_3.36.0      ps_1.9.0                affy_1.84.0            
[160] plyr_1.8.9              forcats_1.0.0           pander_0.6.6           
[163] stringi_1.8.4           viridisLite_0.4.2       BiocParallel_1.40.0    
[166] quarto_1.4.4            munsell_0.5.1           Biostrings_2.74.1      
[169] Matrix_1.7-3            rapportools_1.2         hms_1.1.3              
[172] patchwork_1.3.0         R.filesets_2.15.1       bit64_4.6.0-1          
[175] future_1.34.0           ggplot2_3.5.1           KEGGREST_1.46.0        
[178] statmod_1.5.0           FactoMineR_2.11         shiny_1.10.0           
[181] broom_1.0.7             memoise_2.0.1           affyio_1.76.0          
[184] bslib_0.9.0             bit_4.6.0               readxl_1.4.5  

```

And please also incl. the `sessionInfo()` command. It seems that your error is scan-specific and sensitive to the `mode=` attribute. Could you please make the scan mode visible for the critical scans? I assume that your Ascend files contain different scan types which might not incl. the data streams.

tobiasko, 2025-04-04:
Which `rawrr` package version are you using? It's not attached...

tobiasko, 2025-04-04:
When you check the rawrr assembly path in R using:

```
> rawrrAssemblyPath()
[1] "/Users/tobiaskockmann/Library/Caches/org.R-project.R/R/rawrr/rawrrassembly"
```

and list the content for instance by

```
staff-net-hg-0531:WU323388_Spectronaut_result tobiaskockmann$ ls "/Users/tobiaskockmann/Library/Caches/org.R-project.R/R/rawrr/rawrrassembly"
ThermoFisher.CommonCore.BackgroundSubtraction.dll
ThermoFisher.CommonCore.Data.dll
ThermoFisher.CommonCore.MassPrecisionEstimator.dll
ThermoFisher.CommonCore.RawFileReader.dll
eula.txt
rawrr.exe
rawrr_build.log.28cb22339cb1
```

What do you get? Did you build the rawrr executable from source?

rlumpkin, 2025-04-04:
I am using rawrr 1.15.10. 
I encountered this issue previously on 1.10.2 and 1.14.0, which did not retrieve any scans from this file, regardless of mode. Once I updated to 1.15.10, I was able to start getting some of the scans.

```
> rawrrAssemblyPath()
[1] "/home/rlumpkin/.cache/R/rawrr/rawrrassembly/linux-x64"
```

```
[rlumpkin@(ip-address) projects]$ ls /home/rlumpkin/.cache/R/rawrr/rawrrassembly/linux-x64/
eula.txt  rawrr
```

I installed the R package from source to get to 1.15.10 since I am on R 4.3 and Bioconductor 3.18 only has rawrr 1.10.2.
After installing the package I used installRawrrExe() to install the rawrr exe.

tobiasko, 2025-04-07:
Well, I am not surprised that you had to update `rawrr` in order to read data from Ascend files. The RawFileReader library is not static, Thermo constantly adds/changes functionality. Our `rawrr` executable is build using the dynamic link libraries from Thermo, so an old executable might very well fail reading data from a rather novel Thermo instrument generation.

rlumpkin, 2025-04-07:
Interestingly, I didn't have to update for the instrument. I have processed hundreds of Ascend raw Files on rawrr 1.10.2, and it was only once I updated the MS Tune software that I began seeing issues here.

I removed all packages from the rawrrassembly path and then downloaded the Nupkgs and built the rawrr exe. Notably - I get an error from the add package source, but I think this is simply because I had previously tried building it, so it already was in the path. 

I thought it was odd that the package built successfully, but then had an error about the nupkgs.

```
> rawrr:::.downloadNupkgs()
removing nupkgs files in directory '/home/rlumpkin/.cache/R/rawrr/rawrrassembly/linux-x64'
trying URL 'https://github.com/thermofisherlsms/RawFileReader/raw/refs/heads/main/Libs/NetCore/Net8//ThermoFisher.CommonCore.BackgroundSubtraction.8.0.6.nupkg'
Content type 'application/octet-stream' length 38126 bytes (37 KB)
==================================================
downloaded 37 KB

trying URL 'https://github.com/thermofisherlsms/RawFileReader/raw/refs/heads/main/Libs/NetCore/Net8//ThermoFisher.CommonCore.RandomAccessReaderPlugin.8.0.6.nupkg'
Content type 'application/octet-stream' length 10448 bytes (10 KB)
==================================================
downloaded 10 KB

trying URL 'https://github.com/thermofisherlsms/RawFileReader/raw/refs/heads/main/Libs/NetCore/Net8//ThermoFisher.CommonCore.Data.8.0.6.nupkg'
Content type 'application/octet-stream' length 377172 bytes (368 KB)
==================================================
downloaded 368 KB

trying URL 'https://github.com/thermofisherlsms/RawFileReader/raw/refs/heads/main/Libs/NetCore/Net8//ThermoFisher.CommonCore.RawfileReader.8.0.6.nupkg'
Content type 'application/octet-stream' length 461904 bytes (451 KB)
==================================================
downloaded 451 KB

trying URL 'https://github.com/thermofisherlsms/RawFileReader/raw/refs/heads/main/Libs/NetCore/Net8//ThermoFisher.CommonCore.MassPrecisionEstimator.8.0.6.nupkg'
Content type 'application/octet-stream' length 9323 bytes
==================================================
downloaded 9323 bytes

   ThermoFisher.CommonCore.BackgroundSubtraction.8.0.6.nupkg ThermoFisher.CommonCore.RandomAccessReaderPlugin.8.0.6.nupkg 
                                                           0                                                            0 
                    ThermoFisher.CommonCore.Data.8.0.6.nupkg            ThermoFisher.CommonCore.RawfileReader.8.0.6.nupkg 
                                                           0                                                            0 
  ThermoFisher.CommonCore.MassPrecisionEstimator.8.0.6.nupkg 
                                                           0 

> rawrr:::.addNupkgSource()
error: The source specified has already been added to the list of available package sources. Provide a unique source.

> rawrr:::buildRawrrExe()
Building rawrr assembly using .NET 8.0 ...
Copying rawrr.cs to /tmp/Rtmpzk9rwA
Copying rawrr.csproj to /tmp/Rtmpzk9rwA
Running build ...
Write stdout to/tmp/Rtmpzk9rwA/rawrr.build.stdout.1081121e7fd2.txt
Write stderr to/tmp/Rtmpzk9rwA/rawrr.build.stderr.108148cbf3c6.txt
Build succesfully done.
[1] TRUE
Warning message:
In rawrr:::buildRawrrExe() :
  Have you downloaded the 'thermofisherlsms/RawFileReader' NuGet packages? 
If not, please consider executing the methods:
  -> rawrr:::.downloadNupkgs()
  -> rawrr:::.addNupkgSource()
These steps should be performed once to ensure proper setup.
```

Unfortunately, I'm still getting the same errors

```
> library(rawrr)
> sessionInfo()
R version 4.3.0 (2023-04-21)
Platform: x86_64-pc-linux-gnu (64-bit)
Running under: Amazon Linux 2
[... 153 lines cut ...]
Scan Start Time (min):	 0.001471821
Scan Number:	 1
Base Peak Intensity:	 11799
Base Peak Mass:	 624.2291
Scan Mode:	 FTMS + c NSI cv=-40.00 Full ms [400.0000-1600.0000]
======= Instrument data =====   :	NULL
AGC:	On          
Micro Scan Count:	1
Ion Injection Time (ms):	50.000
=== Mass Calibration:	NULL
Conversion Parameter B:	169484641.678769
Conversion Parameter C:	-41127917.162129
Temperature Comp. (ppm):	-7.87
RF Comp. (ppm):	0.00
Space Charge Comp. (ppm):	-2.33
Resolution Comp. (ppm):	-0.39
Number of LM Found:	0
=== Ion Optics Settings:	NULL
====  Diagnostic Data:	NULL
RawOvFtT:	50987.0
Injection t0:	0.000

attr(,"class")
[1] "rawrrSpectrumSet"
```

tobiasko, 2025-04-07:
> Interestingly, I didn't have to update for the instrument. I have processed hundreds of Ascend raw Files on rawrr 1.10.2, and it was only once I updated the MS Tune software that I began seeing issues here.

Exactly, Tune updates usually happen in the context of Xcalibur/Foundation updates and this may also update the libraries used to write the raw files on the instrument PC.

rlumpkin, 2025-04-07:
I see. So do you think that the most recent libraries are just incompatible with the version of Tune I am running?

tobiasko, 2025-04-08:
no, there is something else here. @leoschwarz Could you help? I am also a bit surprise about these warnings

```
Warning message:
In rawrr:::buildRawrrExe() :
  Have you downloaded the 'thermofisherlsms/RawFileReader' NuGet packages? 
If not, please consider executing the methods:
  -> rawrr:::.downloadNupkgs()
  -> rawrr:::.addNupkgSource()
These steps should be performed once to ensure proper setup.
```

raised by `rawrr:::buildRawrrExe()`

But the big question is still: Why is the `mode="barebone` failing while the default mode is working. Looking at the return above it looks like the barone mode tries to parses the key-value pair `Temperature Comp. (ppm):	-7.87` although it should only target "mZ (centroidStream.Masses), intensity (centroidStream.Intensities), pepmass, StartTime and charge state.

tobiasko, 2025-04-08:
@rlumpkin what was written to

```
	/tmp/Rtmpzk9rwA/file1081fa50af.stderr
	/tmp/Rtmpzk9rwA/file10816f4141de.stdout
```

?

rlumpkin, 2025-04-08:
It does not write anything to those files - they don't exist at all. I see the .R and .txt files in the temp directory, but the readSpectrum command isn't ever generating the stderr or stdout.

Also regarding the mode='backbone' issues - it wasn't just "Temperature Comp. (ppm):	-7.87" causing issues. I noticed the same exact error for every MS1 scan. 

However MS2 scans seem to be hanging up on a different key-value pair, and MS3 scans have another unique error, which again seems to be catching on a specific key. The MS2s seem to be a problem for both "barebone" and regular modes, while MS1s and MS2s only seem to have a problem with "barebone". See below

```
# Scans 971 and 1082 are MS2 scans
# Scans 975 and 1085 are MS3 scans

> rawrr::readSpectrum(rawfile = file, scan = 971, mode="barebone")
Error accessing RAWFileReader library! - The input string '0.70' was not in a correct format.

Memory Usage:
   Before 89716 kb, After 312992 kb, Extra 223276 kb
Error in .rawrrSystem2Source(rawfile, input = scan, rawrrArgs = "cscans",  : 
  Parsing the output of '/home/rlumpkin/.cache/R/rawrr/rawrrassembly/linux-x64/rawrr' failed for an unknown reason.
Please check the debug files:
	/tmp/RtmplulFTq/file3c661d10bea4.stderr
	/tmp/RtmplulFTq/file3c6667a64212.stdout
and the System Requirements

> rawrr::readSpectrum(rawfile = file, scan = 971, mode="")
Error accessing RAWFileReader library! - The input string '0.70' was not in a correct format.

Memory Usage:
   Before 89716 kb, After 341992 kb, Extra 252276 kb
Error in .rawrrSystem2Source(rawfile, input = scan, rawrrArgs = "scans",  : 
  Parsing the output of '/home/rlumpkin/.cache/R/rawrr/rawrrassembly/linux-x64/rawrr' failed for an unknown reason.
Please check the debug files:
	/tmp/RtmplulFTq/file3c661382ee9e.stderr
	/tmp/RtmplulFTq/file3c665e3341f1.stdout
and the System Requirements

> rawrr::readSpectrum(rawfile = file, scan = 975, mode="barebone")
Error accessing RAWFileReader library! - The input string '55.2                     ' was not in a correct format.

Memory Usage:
   Before 89716 kb, After 341764 kb, Extra 252048 kb
Error in .rawrrSystem2Source(rawfile, input = scan, rawrrArgs = "cscans",  : 
  Parsing the output of '/home/rlumpkin/.cache/R/rawrr/rawrrassembly/linux-x64/rawrr' failed for an unknown reason.
Please check the debug files:
	/tmp/RtmplulFTq/file3c6648f0a4c9.stderr
	/tmp/RtmplulFTq/file3c661368c6d8.stdout
and the System Requirements

> rawrr::readSpectrum(rawfile = file, scan = 975, mode="")
[[1]]
Total Ion Current:	 78302.27
Scan Low Mass:	 100
Scan High Mass:	 500
Scan Start Time (min):	 6.541196
Scan Number:	 975
Base Peak Intensity:	 12912
Base Peak Mass:	 119.4514
Scan Mode:	 FTMS + c NSI cv=-70.00 d Full ms3 redacted-mass@cid35.00 redacted-mass@hcd50.00 [100.0000-500.0000]
======= Instrument data =====   :	NULL
AGC:	Predicted   
Micro Scan Count:	1
Charge State:	3
Monoisotopic M/Z:	0.0000
Ion Injection Time (ms):	91.000
MS2 Isolation Width:	1.20
MS2 Isolation Offset:	0.00
HCD Energy:	50.0                     
=== Mass Calibration:	NULL
Conversion Parameter B:	169485354.149782
Conversion Parameter C:	-41128262.943530
Temperature Comp. (ppm):	-7.88
RF Comp. (ppm):	0.00
[... 57 lines cut ...]
======= Instrument data =====   :	NULL
AGC:	Predicted   
Micro Scan Count:	1
Charge State:	3
Monoisotopic M/Z:	0.0000
Ion Injection Time (ms):	91.000
MS2 Isolation Width:	1.20
MS2 Isolation Offset:	0.00
HCD Energy:	50.0                     
=== Mass Calibration:	NULL
Conversion Parameter B:	169485354.107588
Conversion Parameter C:	-41128262.924965
Temperature Comp. (ppm):	-7.88
RF Comp. (ppm):	0.00
Space Charge Comp. (ppm):	-0.44
Resolution Comp. (ppm):	-0.17
Number of LM Found:	0
=== Ion Optics Settings:	NULL
====  Diagnostic Data:	NULL
RawOvFtT:	25185.9
Injection t0:	0.000

attr(,"class")
[1] "rawrrSpectrumSet"
```

rlumpkin, 2025-04-08:
Since noticing this, I looked into old files using the same method but on MS Tune 4.0. They don't have the input string issue, but they do sometimes have an issue extracting mZ from MS2 scans exclusive when using mode="barebone".

```
> scan <- rawrr::readSpectrum(rawfile = file, scan = 360, mode="barebone") # MS1 Scan - No Issues

> scan <- rawrr::readSpectrum(rawfile = file, scan = 362, mode="barebone") # MS2 Scan
Warning message:
length of mZ value vector is less than 1. 
> scan[[1]]$mZ
 [1]  1  2  3  4  5  6  7  8  9 10
> scan[[1]]$intensity
 [1]  1  2  3  4  5  6  7  8  9 10

> scan <- rawrr::readSpectrum(rawfile = file, scan = 362, mode="") # MS2 Scan
> scan[[1]]$mZ
 [1]  388.252  451.400  452.357  540.514  548.563  563.127  563.701  579.949  591.482  632.788  633.696  760.026  763.367  819.248
[15]  828.926  841.798  873.389  928.643  957.579  984.154 1023.698 1041.844 1048.745 1068.437 1108.857 1135.702 1162.323 1179.509
[29] 1565.707
> scan[[1]]$intensity
 [1] 14.214335 24.815260 26.949799 40.813793 26.686255 45.584423 39.242687 24.777418 35.083679 48.494141 49.619820 12.681935 70.720741
[14] 28.256338 38.041862 28.937572 16.263821 46.501919 29.899288 46.330956  9.376893 49.183643 30.107632 21.836058 52.057854 42.389629
[27] 68.382980 27.941280 23.086395

> scan <- rawrr::readSpectrum(rawfile = file, scan = 363, mode="barebone") # MS3 Scan - No Issues
```

tobiasko, 2025-04-09:
Interesting. You might want to correlate this with the raw file version of your files. The file header contains:

```
$`RAW file version`
[1] "66"
```
and

```
$`Software version`
[1] "4.2.4321"
```

The barebone mode needs scan data in a certain data stream. Not all scan events write this stream. Have a look at [the RawFileReader docu](https://github.com/thermofisherlsms/RawFileReader/blob/main/docs/UsingRawFileReader.docx) if you want to understand this better.

tobiasko, 2025-04-09:
@rlumpkin Could you send us a raw file for trouble shooting?

rlumpkin, 2025-04-09:
I'll look into it, I will have to consult my team to see if I am allowed to share a raw file.

rlumpkin, 2025-06-05:
@cpanse 

MS1 scans are read correctly now. 

I can read MS2 scans without barebone, but MS2 scans are not reading in properly with mode='barebone
I get the error "length of mZ value vector is less than 1." and the mZ and intensity vectors are just 1:10

`> library(rawrr)
> sessionInfo()
R version 4.4.0 (2024-04-24)
Platform: aarch64-apple-darwin20
Running under: macOS 15.5

Matrix products: default
BLAS:   /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib 
LAPACK: /Library/Frameworks/R.framework/Versions/4.4-arm64/Resources/lib/libRlapack.dylib;  LAPACK version 3.12.0

locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8

time zone: America/New_York
tzcode source: internal

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] rawrr_1.17.2

loaded via a namespace (and not attached):
[1] compiler_4.4.0 tools_4.4.0   
> file = "/Users/rlumpkin/Downloads/AD02893.raw"
> fileIndex <- rawrr::readIndex(file)
> 
> readSpectrum(rawfile = file, scan = unique(fileIndex[which(fileIndex$MSOrder=="Ms"),"scan"])[1:2], mode = "barebone")
[[1]]
Total Ion Current:	
Scan Low Mass:	
Scan High Mass:	
Scan Start Time (min):	 0.00105179
Scan Number:	 1
Base Peak Intensity:	
Base Peak Mass:	
Scan Mode:	
======= Instrument data =====   :	NULL
=== Mass Calibration:	NULL
=== Ion Optics Settings:	NULL
====  Diagnostic Data:	NULL

[[2]]
Total Ion Current:	
Scan Low Mass:	
Scan High Mass:	
Scan Start Time (min):	 0.007305123
Scan Number:	 2
Base Peak Intensity:	
Base Peak Mass:	
Scan Mode:	
======= Instrument data =====   :	NULL
=== Mass Calibration:	NULL
=== Ion Optics Settings:	NULL
====  Diagnostic Data:	NULL

attr(,"class")
[1] "rawrrSpectrumSet"
> readSpectrum(rawfile = file, scan = unique(fileIndex[which(fileIndex$MSOrder=="Ms"),"scan"])[1:2], mode = "")
[[1]]
Total Ion Current:	 15087928
Scan Low Mass:	 400
Scan High Mass:	 1600
Scan Start Time (min):	 0.00105179
Scan Number:	 1
Base Peak Intensity:	 2114552
Base Peak Mass:	 445.1187
Scan Mode:	 FTMS + c NSI Full ms [400.0000-1600.0000]
======= Instrument data =====   :	NULL
AGC:	On          
Micro Scan Count:	1
Ion Injection Time (ms):	19.516
=== Mass Calibration:	NULL
Conversion Parameter B:	169486306.731393
Conversion Parameter C:	-40686684.029560
Temperature Comp. (ppm):	-7.68
RF Comp. (ppm):	0.00
Space Charge Comp. (ppm):	-2.58
Resolution Comp. (ppm):	-0.20
Number of LM Found:	0
=== Ion Optics Settings:	NULL
====  Diagnostic Data:	NULL
RawOvFtT:	410423.4
Injection t0:	0.000

[[2]]
Total Ion Current:	 16357038
Scan Low Mass:	 400
Scan High Mass:	 1600
Scan Start Time (min):	 0.007305123
Scan Number:	 2
Base Peak Intensity:	 2206780
Base Peak Mass:	 445.1189
Scan Mode:	 FTMS + c NSI Full ms [400.0000-1600.0000]
======= Instrument data =====   :	NULL
AGC:	On          
Micro Scan Count:	1
Ion Injection Time (ms):	21.072
=== Mass Calibration:	NULL
Conversion Parameter B:	169486306.731393
Conversion Parameter C:	-40686684.029560
Temperature Comp. (ppm):	-7.68
RF Comp. (ppm):	0.00
Space Charge Comp. (ppm):	-2.58
Resolution Comp. (ppm):	-0.20
Number of LM Found:	0
=== Ion Optics Settings:	NULL
====  Diagnostic Data:	NULL
RawOvFtT:	466618.3
Injection t0:	0.000

attr(,"class")
[1] "rawrrSpectrumSet"
> 
> readSpectrum(rawfile = file, scan = unique(fileIndex[which(fileIndex$MSOrder=="Ms2"),"scan"])[1:2], mode = "barebone")
[[1]]
Total Ion Current:	
Scan Low Mass:	
Scan High Mass:	
Scan Start Time (min):	 0.007883633
Scan Number:	 3
Base Peak Intensity:	
Base Peak Mass:	
Scan Mode:	
======= Instrument data =====   :	NULL
=== Mass Calibration:	NULL
=== Ion Optics Settings:	NULL
====  Diagnostic Data:	NULL

[[2]]
Total Ion Current:	
Scan Low Mass:	
Scan High Mass:	
Scan Start Time (min):	 0.01022798
Scan Number:	 4
Base Peak Intensity:	
Base Peak Mass:	
Scan Mode:	
======= Instrument data =====   :	NULL
=== Mass Calibration:	NULL
=== Ion Optics Settings:	NULL
====  Diagnostic Data:	NULL

attr(,"class")
[1] "rawrrSpectrumSet"
**Warning messages:
1: length of mZ value vector is less than 1. 
2: length of mZ value vector is less than 1.** 

> readSpectrum(rawfile = file, scan = unique(fileIndex[which(fileIndex$MSOrder=="Ms2"),"scan"])[1:2], mode = "")
[[1]]
Total Ion Current:	 34679.51
Scan Low Mass:	 115
Scan High Mass:	 876
Scan Start Time (min):	 0.007883633
Scan Number:	 3
Base Peak Intensity:	 6777
Base Peak Mass:	 321.001
Scan Mode:	 ITMS + c NSI t d Full ms2 432.8263@cid35.00 [115.0000-876.0000]
======= Instrument data =====   :	NULL
AGC:	Predicted   
Micro Scan Count:	1
Charge State:	2
Monoisotopic M/Z:	432.8263
Ion Injection Time (ms):	35.000
MS2 Isolation Width:	0.70
MS2 Isolation Offset:	0.00
HCD Energy:	                         
=== Mass Calibration:	NULL
=== Ion Optics Settings:	NULL
====  Diagnostic Data:	NULL

[[2]]
Total Ion Current:	 15336.78
Scan Low Mass:	 121
Scan High Mass:	 1232
Scan Start Time (min):	 0.01022798
Scan Number:	 4
Base Peak Intensity:	 5345
Base Peak Mass:	 388.971
Scan Mode:	 ITMS + c NSI t d Full ms2 407.1340@cid35.00 [121.0000-1232.0000]
======= Instrument data =====   :	NULL
AGC:	Predicted   
Micro Scan Count:	1
Charge State:	3
Monoisotopic M/Z:	407.1340
Ion Injection Time (ms):	35.000
MS2 Isolation Width:	0.70
MS2 Isolation Offset:	0.00
HCD Energy:	                         
=== Mass Calibration:	NULL
=== Ion Optics Settings:	NULL
====  Diagnostic Data:	NULL

attr(,"class")
[1] "rawrrSpectrumSet"`

cpanse, 2025-06-05:
1. It is a warning. 
2. when ` mode = 'barebone'`  readSpectrum is implemented the way that it writes only if  `scanStatistics.IsCentroidScan is true`
See the C# code [line 340](https://github.com/fgcz/rawrr/blob/0cd4491b1bfcf93ebc8159ce60c3b6aff0534877/inst/rawrrassembly/rawrr.cs#L340). The library logic is also not 100% clear to me. Sorry about that.