fgcz/rawrr #85: Failed to read precursor charge and monoisotopicMz

Bohui2447, 2025-04-28:
Hi, the readIndex function can not extract precursor charge and monoisotopicMz.

library(rawrr)
raw_file = '../001_rawData/test.raw'
tmp = readIndex(rawfile = raw_file)

Here is the result of the readIndex
[image]

It works well two months before, but recent days, it can not extract these two columns.

tobiasko, 2025-04-29:
Hi @Bohui2447,

could you please run the example code for `readIndex` incl. a session info?

```
> Idx <- rawrr::sampleFilePath() |> rawrr::readIndex()
> head(Idx)
  scan                                                     scanType   StartTime precursorMass
1    1                    FTMS + c NSI Full ms [350.0000-1800.0000] 0.001619751     1075.0000
2    2 FTMS + c NSI Full ms2 487.2567@hcd28.00 [140.0000-1015.0000] 0.005839446      487.2567
3    3 FTMS + c NSI Full ms2 644.8226@hcd28.00 [140.0000-1335.0000] 0.006989346      644.8226
4    4 FTMS + c NSI Full ms2 683.8279@hcd28.00 [140.0000-1415.0000] 0.008149761      683.8279
5    5 FTMS + c NSI Full ms2 547.2980@hcd28.00 [140.0000-1135.0000] 0.009297695      547.2980
6    6 FTMS + c NSI Full ms2 669.8381@hcd28.00 [140.0000-1385.0000] 0.010449778      669.8381
  MSOrder charge masterScan dependencyType monoisotopicMz
1      Ms      0         NA             NA              0
2     Ms2      2         NA             NA              0
3     Ms2      2         NA             NA              0
4     Ms2      2         NA             NA              0
5     Ms2      2         NA             NA              0
6     Ms2      2         NA             NA              0
> sessionInfo()
R version 4.4.1 (2024-06-14)
Platform: aarch64-apple-darwin20
Running under: macOS Sonoma 14.6

Matrix products: default
BLAS:   /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib 
LAPACK: /Library/Frameworks/R.framework/Versions/4.4-arm64/Resources/lib/libRlapack.dylib;  LAPACK version 3.12.0

locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8

time zone: Europe/Zurich
tzcode source: internal

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] rawrr_1.14.0       SpectroPipeR_0.4.2

loaded via a namespace (and not attached):
  [1] splines_4.4.1           later_1.4.1             tibble_3.2.1           
  [4] R.oo_1.27.0             cellranger_1.1.0        preprocessCore_1.68.0  
  [7] XML_3.99-0.18           factoextra_1.0.7        PECA_1.42.0            
 [10] lifecycle_1.0.4         tcltk_4.4.1             edgeR_4.4.2            
 [13] doParallel_1.0.17       vroom_1.6.5             globals_0.16.3         
 [16] processx_3.8.6          lattice_0.22-6          MASS_7.3-65            
 [19] flashClust_1.01-2       backports_1.5.0         magrittr_2.0.3         
 [22] sass_0.4.9              openxlsx_4.2.8          limma_3.62.2           
 [25] rmarkdown_2.29          jquerylib_0.1.4         yaml_2.3.10            
 [28] httpuv_1.6.15           summarytools_1.1.3      zip_2.3.2              
 [31] askpass_1.2.1           reticulate_1.42.0       iq_1.10.1              
 [34] DBI_1.2.3               RColorBrewer_1.1-3      lubridate_1.9.4        
 [37] zlibbioc_1.52.0         R.cache_0.16.0          purrr_1.0.4            
 [40] R.utils_2.13.0          BiocGenerics_0.52.0     aroma.apd_0.7.0        
 [43] sva_3.54.0              GenomeInfoDbData_1.2.13 IRanges_2.40.1         
 [46] S4Vectors_0.44.0        ggrepel_0.9.6           listenv_0.9.1          
 [49] genefilter_1.88.0       aroma.affymetrix_3.2.2  umap_0.2.10.0          
 [52] RSpectra_0.16-2         annotate_1.84.0         R.rsp_0.46.0           
 [55] parallelly_1.43.0       svglite_2.1.3           codetools_0.2-20       
 [58] xml2_1.3.8              DT_0.33                 DNAcopy_1.80.0         
 [61] tidyselect_1.2.1        UCSC.utils_1.2.0        farver_2.1.2           
 [64] matrixStats_1.5.0       stats4_4.4.1            base64enc_0.1-3        
 [67] jsonlite_2.0.0          gghalves_0.1.4          survival_3.8-3         
 [70] iterators_1.0.14        emmeans_1.11.0          systemfonts_1.2.1      
 [73] foreach_1.5.2           tools_4.4.1             ragg_1.3.3             
 [76] pryr_0.1.6              snow_0.4-4              Rcpp_1.0.14            
 [79] glue_1.8.0              gridExtra_2.3           xfun_0.51              
[... 129 lines cut ...]
$`Sample row number`
[1] "1"

$`Sample dilution factor`
[1] "0"

$`Sample barcode`
[1] ""

$`User text 0`
[1] ""

$`User text 1`
[1] ""

$`User text 2`
[1] ""

$`User text 3`
[1] ""

$`User text 4`
[1] ""

```

tobiasko, 2025-04-29:
So it works for all files recorded using software version 4.0.4091, after upgrading to 4.2.4321 it always failed? Looks like all files were recorded with the very same method and on the very same hardware?

Bohui2447, 2025-04-29:
It's tricky, because it works for files generated by the software version 4.2.4321 but failed for files generated by version 4.0.4091.

tobiasko, 2025-04-29:
Now I am totally confused! :-) So your current rawrr.exe is from where? After installing the package there was no rawrr.exe in your assembly path and you were unable to download it using `installRawrrExe`?

cpanse, 2025-05-07:
@Bohui2447, thanks for reporting. Is it possible that you provide the raw file so that we can reproduce the problem?

Bohui2447, 2025-05-12:
@cpanse The raw file is too large to upload here. How can I send it to you privately?

cpanse, 2025-05-13:
@Bohui2447  A link to cp@fgcz.ethz.ch or a Google Drive or Dropbox would help. C

cpanse, 2025-05-30:
@Bohui2447, please verify if the new version resolves this issue.
```
install.packages('https://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr_1.17.2.tar.gz')
library(rawrr)
rawrr::installRawrrExe(force = T)
```