fgcz/rawrr #87: Reading "sample.raw" fails

breidan, 2025-06-05:
Hi,
after migration to R4.5 and updating my BioConcductor packages `rawrr` is not working anymore. Issueing  ` H <- rawrr::readFileHeader(rawfile = sampleFilePath())` throws the following error:

>GC heap initialization failed with error 0x8007000E
Failed to create CoreCLR, HRESULT: 0x8007000E
Error in if (rvs != "No RAW file specified!") { : 
  argument is of length zero
In addition: Warning message:
In system2(exe, stdout = TRUE) :
  running command ''/R/rawrr/rawrrassembly/linux-x64/rawrr'' had status 137

Mono is in the path:
> Sys.which("mono")
           mono 
"/usr/bin/mono" 

The rawrr assembly is installed:
> list.files(rawrrAssemblyPath())
[1] "rawrr"

And here is the sessionInfo:
> sessionInfo()
R version 4.5.0 (2025-04-11)
Platform: x86_64-pc-linux-gnu
Running under: Red Hat Enterprise Linux 9.6 (Plow)

> Matrix products: default
BLAS/LAPACK: /usr/lib64/libopenblas-r0.3.26.so;  LAPACK version 3.12.0

> locale:
 [1] LC_CTYPE=en_US.UTF-8       LC_NUMERIC=C               LC_TIME=en_US.UTF-8       
 [4] LC_COLLATE=en_US.UTF-8     LC_MONETARY=en_US.UTF-8    LC_MESSAGES=en_US.UTF-8   
 [7] LC_PAPER=en_US.UTF-8       LC_NAME=C                  LC_ADDRESS=C              
[10] LC_TELEPHONE=C             LC_MEASUREMENT=en_US.UTF-8 LC_IDENTIFICATION=C       

> time zone: Europe/Rome
tzcode source: system (glibc)

> attached base packages:
[1] stats4    stats     graphics  grDevices utils     datasets  methods   base     

> other attached packages:
 [1] MsBackendRawFileReader_1.14.0 Spectra_1.18.2               
 [3] S4Vectors_0.46.0              BiocGenerics_0.54.0          
 [5] generics_0.1.3                MsExperiment_1.10.0          
 [7] ProtGenerics_1.40.0           ABs.xcmsHelpers_1.8          
 [9] xcms_4.6.0                    BiocParallel_1.42.1          
[11] rawrr_1.16.0                  lubridate_1.9.4              
[13] forcats_1.0.0                 stringr_1.5.1                
[15] dplyr_1.1.4                   purrr_1.0.4                  
[17] readr_2.1.5                   tidyr_1.3.1                  
[19] tibble_3.2.1                  tidyverse_2.0.0              
[21] ggrepel_0.9.6                 ggplot2_3.5.2                
[23] magrittr_2.0.3                MUtools_1.4                  
[25] MASS_7.3-65                  

> loaded via a namespace (and not attached):
 [1] DBI_1.2.3                   rlang_1.1.6                 clue_0.3-66                
 [4] MassSpecWavelet_1.74.0      matrixStats_1.5.0           compiler_4.5.0             
 [7] vctrs_0.6.5                 reshape2_1.4.4              pkgconfig_2.0.3            
[10] MetaboCoreUtils_1.16.1      crayon_1.5.3                fastmap_1.2.0              
[13] XVector_0.48.0              rmarkdown_2.29              pracma_2.4.4               
[16] tzdb_0.5.0                  preprocessCore_1.70.0       UCSC.utils_1.4.0           
[19] xfun_0.52                   MultiAssayExperiment_1.34.0 GenomeInfoDb_1.44.0        
[22] jsonlite_2.0.0              progress_1.2.3              DelayedArray_0.34.1        
[25] prettyunits_1.2.0           parallel_4.5.0              cluster_2.1.8.1            
[28] R6_2.6.1                    stringi_1.8.7               RColorBrewer_1.1-3         
[31] limma_3.64.1                pkgload_1.4.0               GenomicRanges_1.60.0       
[34] iterators_1.0.14            Rcpp_1.0.14                 SummarizedExperiment_1.38.1
[37] knitr_1.50                  IRanges_2.42.0              BiocBaseUtils_1.10.0       
[40] Matrix_1.7-3                igraph_2.1.4                timechange_0.3.0           
[43] tidyselect_1.2.1            rstudioapi_0.17.1           abind_1.4-8                
[46] yaml_2.3.10                 doParallel_1.0.17           codetools_0.2-20           
[49] affy_1.86.0                 lattice_0.22-6              plyr_1.8.9                 
[52] Biobase_2.68.0              withr_3.0.2                 evaluate_1.0.3             
[55] pillar_1.10.2               affyio_1.78.0               BiocManager_1.30.25        
[58] MatrixGenerics_1.20.0       rsconnect_1.3.4             foreach_1.5.2              
[61] MSnbase_2.34.1              MALDIquant_1.22.3           ncdf4_1.24                 
[64] hms_1.1.3                   munsell_0.5.1               scales_1.3.0               
[67] glue_1.8.0                  MsFeatures_1.16.0           lazyeval_0.2.2             
[70] tools_4.5.0                 mzID_1.46.0                 QFeatures_1.18.0           
[73] vsn_3.76.0                  mzR_2.42.0                  XML_3.99-0.18              
[76] fs_1.6.6                    grid_4.5.0                  impute_1.82.0              
[79] MsCoreUtils_1.20.0          colorspace_2.1-1            GenomeInfoDbData_1.2.14    
[82] PSMatch_1.12.0              cli_3.6.4                   S4Arrays_1.8.1             
[85] AnnotationFilter_1.32.0     pcaMethods_2.0.0            gtable_0.3.6               
[88] digest_0.6.37               SparseArray_1.8.0           htmltools_0.5.8.1          
[91] lifecycle_1.0.4             httr_1.4.7                  statmod_1.5.0              

Any help is much appreciated.

cpanse, 2025-06-11:
@breidan, the snippet below should solve the issue.

```
install.packages('https://fgcz-ms.uzh.ch/~cpanse/rawrr/rawrr_1.17.2.tar.gz')
library(rawrr)
rawrr::installRawrrExe(force = T)
```

breidan, 2025-06-12:
@cpanse, thanks for the reply. But it did not solve the problem. Even with `rawrr_1.17.2` and its assembly I get the same error message.

cpanse, 2025-06-12:
@breidan can you send me the session info again? of note, rawrr 1.17 ships with a static compiled file.

breidan, 2025-06-12:
@cpanse ,

> Can you please run
> 
> ```
> ldd /delta/JRC/services/rstudio-home/breidan@delta.europa.eu/.cache/R/rawrr/rawrrassembly/linux-x64/rawrr

Here is the output:

>linux-vdso.so.1 (0x00007ffdb892a000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f6418e64000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007f6418e5f000)
        libz.so.1 => /lib64/libz.so.1 (0x00007f6418e45000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f6418d6a000)
        librt.so.1 => /lib64/librt.so.1 (0x00007f6418d65000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f6418d49000)
        libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007f6418a00000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f6418600000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f641995b000)

breidan, 2025-06-12:
> Hi [@breidan](https://github.com/breidan)
> 
> The error code at the beginning can indicate a lack of Memory or Storage. Can you verify neither is the problem on your system to rule out this cause?
> 
> https://www.hresult.info/FACILITY_WIN32/0x8007000E

@leoschwarz,

I can confirm that there is sufficient memory and disk space. When the above error appeared there was 8 % of the memory used and 14 % of the disk space.

breidan, 2025-06-12:
@cpanse, executing `rawrr` in the terminal throws the same error:

> [breidan@delta.europa.eu@siprapp01p-rstd` linux-x64]$ ./rawrr version

> GC heap initialization failed with error 0x8007000E#
Failed to create CoreCLR, HRESULT: 0x8007000E

cpanse, 2025-06-12:
@breidan, @leoschwarz  is on the right track.

At the moment, I have no idea how to help you.

```
ulimit -Ha
ulimit -Sa
```

On what kind of system are you working?

leoschwarz, 2025-06-12:
I found a related issue in dotnet/runtime: https://github.com/dotnet/runtime/issues/85556 

It does sound you may be meeting virtual memory constraints.

Can you check

```
cat /proc/meminfo | grep VmallocTotal
```

Potential workaround would be to lower the memory (e.g., but untested)

```
export DOTNET_GCHeapHardLimit=1C0000000
```

closed 2025-06-12 by breidan

leoschwarz, 2025-06-12:
Glad to hear that worked! I hope they will address the issue I've linked in the dotnet runtime eventually.

@cpanse 
Could we set a reasonable memory limit in the application *if the variable is not set already*? (The value is number of bytes in hexadecimal form.)